aaronwmorris/indi-allsky #1747: error: could not write to 'somewhere': No space left on device during build (fixed - comment on the workaround in the issue

richiejarvis, 2024-12-29:
Hi Aaron,

Filing this as a "bug" - it isn't really, but you should update the install and tuning sections with this info in case it bites someone else.

The root cause is that the build process is using /tmp for the build, and restricting the size as per the tuning section causes the build to fail with a confusing out of space message which doesn't mention that the problem is /tmp at all.

The solution is to `sudo umount /tmp` first, and re-enable (or reboot) later.  The build then works correctly as expected.

Cheers,

Richie
       
 ```
 
      copying build/lib.linux-aarch64-cpython-311/numcodecs/tests/test_zstd.py -> build/bdist.linux-aarch64/wheel/./numcodecs/tests
      creating build/bdist.linux-aarch64/wheel/numcodecs/tests/package_with_entrypoint
      copying build/lib.linux-aarch64-cpython-311/numcodecs/tests/package_with_entrypoint/__init__.py -> build/bdist.linux-aarch64/wheel/./numcodecs/tests/package_with_entrypoint
      creating build/bdist.linux-aarch64/wheel/numcodecs/tests/package_with_entrypoint-0.1.dist-info
      copying build/lib.linux-aarch64-cpython-311/numcodecs/tests/package_with_entrypoint-0.1.dist-info/entry_points.txt -> build/bdist.linux-aarch64/wheel/./numcodecs/tests/package_with_entrypoint-0.1.dist-info
      copying build/lib.linux-aarch64-cpython-311/numcodecs/blosc.cpython-311-aarch64-linux-gnu.so -> build/bdist.linux-aarch64/wheel/./numcodecs
      copying build/lib.linux-aarch64-cpython-311/numcodecs/zstd.cpython-311-aarch64-linux-gnu.so -> build/bdist.linux-aarch64/wheel/./numcodecs
      error: could not write to 'build/bdist.linux-aarch64/wheel/./numcodecs/zstd.cpython-311-aarch64-linux-gnu.so': No space left on device
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for numcodecs
Successfully built bottleneck dbus-python rpi-ws281x sysv-ipc RPi.GPIO
Failed to build numcodecs
WARNING: There was an error checking the latest version of pip.
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (numcodecs)

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...

richie@allsky:~/git/indi-allsky $ df -kh
Filesystem      Size  Used Avail Use% Mounted on
udev            3.6G     0  3.6G   0% /dev
tmpfs           781M  9.2M  772M   2% /run
/dev/mmcblk0p2  454G  6.1G  425G   2% /
tmpfs           3.9G     0  3.9G   0% /dev/shm
tmpfs           5.0M   16K  5.0M   1% /run/lock
/dev/mmcblk0p1  510M   55M  456M  11% /boot/firmware
tmpfs           512M     0  512M   0% /tmp
tmpfs           781M     0  781M   0% /run/user/1000

richie@allsky:~/git/indi-allsky $ umount /tmp
umount: /tmp: must be superuser to unmount.
richie@allsky:~/git/indi-allsky $ sudo !!
sudo umount /tmp
richie@allsky:~/git/indi-allsky $ df -kh
Filesystem      Size  Used Avail Use% Mounted on
udev            3.6G     0  3.6G   0% /dev
tmpfs           781M  9.2M  772M   2% /run
/dev/mmcblk0p2  454G  6.1G  425G   2% /
tmpfs           3.9G     0  3.9G   0% /dev/shm
tmpfs           5.0M   16K  5.0M   1% /run/lock
/dev/mmcblk0p1  510M   55M  456M  11% /boot/firmware
tmpfs           781M     0  781M   0% /run/user/1000
```

richiejarvis, 2024-12-29:
Update:  

Doing the buld a second time then gave this error:

```
**** Starting gunicorn-indi-allsky.socket
**** Update config camera interface ****
jq: error: Could not open file /tmp/tmp.lPyTTkN5a5.json: No such file or directory

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...
```
So I checked /tmp, and noted that the json file has a different name.  

```
richie@allsky:/ $ cd tmp
richie@allsky:/tmp $ ls
systemd-private-a61d6c38390f44318c5c80d29c7920e4-apache2.service-Ge7XRt  tmp.LAuHKopQBv.json
```
So I copied it to the second name:

```
richie@allsky:/tmp $ cp tmp.LAuHKopQBv.json tmp.lPyTTkN5a5.json
```

This solved the issue.

Cheers,

Richie

aaronwmorris, 2025-01-01:
Merged #1755 to warn if there is less than 512MB on the /tmp filesystem.  This is tough situation because some systems may need more than 1GB of space depending on the number of modules that have to be compiled from scratch.

richiejarvis, 2025-01-02:
Hi Aaron,

I would suggest the best way to handle this would be to either:

1. Auto umount /tmp prior to compiling, and mount afterwards
2. Update the docs to remind users to umount /tmp before running the build

I expect 1 would be preferable, as unmounting/remounting /tmp before and
after the build using sudo would fix the issue, and do nothing if /tmp was
not a separate filesystem.

Just my thoughts really.

Cheers,

Richie

---
Mx Richie Jarvis (Xe/Xem/Xyr)
***@***.***
***@***.***
***@***.***
South Common Observatory, Sussex, UK
Website (OLD!): https://deepsky.org.uk
Website (NEW!): https://deep-sky.uk

aaronwmorris, 2025-01-03:
In this case, I can only update the docs with the recommendation.  There is a reasonable chance that attempting to unmount the /tmp filesystem  will fail due to open files (on the filesystem).

closed 2025-02-20 by aaronwmorris