aaronwmorris/indi-allsky #1748: Restore sqlite3 database as per "backup and recovery" pages doesn't work on a newly installed indi-allsky instance

richiejarvis, 2024-12-29:
Hi Aaron,

I am working on restoring the sqlite3 DB, and ran into a problem because the existing database already existed.  So I had to remove the existing DB first in order to reload my backup version.

```
richie@allsky:~ $ rm /var/lib/indi-allsky/indi-allsky.sqlite
richie@allsky:~ $ gunzip -c backup_indi-allsky_sqlite_20241229_141124.sql.gz | sqlite3 /var/lib/indi-allsky/indi-allsky.sqlite
```

Ah... Just found this page: https://github.com/aaronwmorris/indi-allsky/wiki/Rebuilding-the-database  

I could've used that instead to restore my settings I guess?

Anyway, my kludge worked and indi-allsky is running :)  It seems to have missed my copied images though.  Weird, but only 2 days worth so not a disaster.

Cheers,

Richie

aaronwmorris, 2025-01-01:
For just restoring your settings, there is an alternative where you can just extract the config JSON from the database and just force import those settings to a different database.  The instructions for that are in the wiki page you found.

As for the images, you should be able to just move those.  The images are stored with the relative path to the IMAGE_FOLDER from the config.

richiejarvis, 2025-01-02:
Thanks Aaron.  Can you put a link in the backup/restore for others who
might want to do the same thing please?

Interestingly, I did move the images from the old system to the new one,
but the system didn't "detect" them.  I am not sure if that is because I
restored the DB or not maybe.

Anyway, it doesn't matter in my case - the old images I mean, because it
was cloudy those days anyway!  Thought it might be good for you to know
that it might not be working quite as intended when you move just the
images.

Cheers,

Richie

aaronwmorris, 2025-08-24:
I have created a new section in the wiki to highlight all of the backup and recovery pages that are available.

closed 2025-08-24 by aaronwmorris