aaronwmorris/indi-allsky #1750: Appearing to not generate videos, startrails, etc at EoN.

jshall77, 2024-12-31:
Aaron,

Happy New Year!  I hope all is well!  I did an upgrade to one of my systems before going on vacation.  After that, it appears that the system is no longer generating star trail, keogram, videos, etc.  Even when I try to generate them manually, they don't ever seem to appear.  Where do I need to be looking in the logs for issues with EoN?

Thanks!

aaronwmorris, 2025-01-01:
Check the log at `/var/log/indi-allsky/indi-allsky.log` for any exceptions that might be generated.

Can you post your Support Info bundle here?

jshall77, 2025-01-01:
See below.  Happy New Year!

/home/hallnetadmin/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3882912 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux hallsky02 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
Hostname
hallsky02
hallsky02.hallnet.local 

Time
Wed  1 Jan 10:23:00 EST 2025

System timezone
America/New_York

Uptime
 10:23:00 up 22:34,  0 user,  load average: 0.08, 0.17, 0.17

Memory
               total        used        free      shared  buff/cache   available
Mem:         3882912     1175032      413324        1888     2374068     2707880
Swap:        2097148           0     2097148

Filesystems
Filesystem     1K-blocks      Used Available Use% Mounted on
udev             1671108         0   1671108   0% /dev
tmpfs             388292      1204    387088   1% /run
/dev/mmcblk0p2 238215724 144276488  81826752  64% /
tmpfs            1941456       184   1941272   1% /dev/shm
tmpfs               5120        16      5104   1% /run/lock
/dev/mmcblk0p1    522232     65566    456666  13% /boot/firmware
tmpfs             388288         0    388288   0% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
16544

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: end0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether d8:3a:dd:6b:23:61 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.72/24 brd 192.168.1.255 scope global dynamic noprefixroute end0
       valid_lft 91571sec preferred_lft 91571sec
    inet6 fd4f:2e8a:4b11:1cf2:50d4:befe:cdf8:a4c8/64 scope global dynamic noprefixroute 
[... 1160 lines cut ...]
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": true,
    "DAYTIME_TRACK": true,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:5,110 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2025-01-01:
I do not see anything unusual in the support info.  Do you think you could compress your logs and make them available through google drive or something like that?

```
tar -I "gzip -9" -cvf indi-allsky_logs_$(date +%Y%m%d_%H%M%S).tar.gz /var/log/indi-allsky/indi-allsky.log*
```

jshall77, 2025-01-01:
[Uploading indi-allsky_logs_20250101_140613.tar.gz…]()

jshall77, 2025-01-01:
Here you go:

https://1drv.ms/f/c/19e7298afc80497b/EmIV5jf_gKdJpqA5T40YebMBVMcFUM-PNnjnNrMbmIb_zQ?e=VSz1Tr

czimm74, 2025-01-01:
Same thing for me aaron.  I did an update on 12/24 and have not had a generation of files since then.  I updated again yesterday and still nothing being generated, and not seeing anything in errors either.

Even if I try forcing a generation of a missing timelapse/star trail/etc, I can't find any errors and nothing is created.  I am trying a mini-timelapse to see if that will be created.

Could it be something related to external USB file storage of the images again?

aaronwmorris, 2025-01-01:
@jshall77 @czimm74 I think was caused by my recent upgrade of the python numpy module.  Numpy 2.0 was released mid-2024 and a bunch of modules that relied on numpy had to be upgraded for compatibility.  I held off for 6 months and I thought I caught all of the modules that needed specific versions, but I missed the `shapely` module.

Merged #1754 to fix.

jshall77, 2025-01-01:
Got it! Thanks Aaron!!

czimm74, 2025-01-01:
Looks like it is all fixed.  Always that last location you forget to look at. ;)

kf6spf, 2025-01-02:
I'm having the same issue with timelapses ... starting after 24 Dec 2024.

In the error loge I get:

2025-01-02T13:01:25.982106-08:00 allsky [ERROR] Video-1-1066887/MainThread timelapse.generate() [160]: FFMPEG failed to generate timelapse, return code: 1
2025-01-02T13:01:25.982176-08:00 allsky [ERROR] Video-1-1066887/MainThread timelapse.generate() [161]: FFMPEG output: b"[swscaler @ 0x55949557f0] [swscaler @ 0x5594962780] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949557f0] [swscaler @ 0x55949f6730] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949557f0] [swscaler @ 0x5594a8a6e0] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949557f0] [swscaler @ 0x5594b1e690] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949557f0] [swscaler @ 0x5594bb2640] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x55949be460] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x55949f8e70] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x5594a8ce20] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x5594b20d90] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x5594bb7210] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x55949be460] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x55949f8e70] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x5594a8ce20] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x5594b20d90] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x5594bb7210] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x55949be460] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x55949f8e70] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x5594a8ce20] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x5594b20d90] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x55949b14d0] [swscaler @ 0x5594bb7210] [warning] deprecated pixel format used, make sure you did set range correctly\n[h264_v4l2m2m @ 0x5594675e20] [warning] check_size: Size check failed: asked for 2281x3043, got: 1920x1920\n[h264_v4l2m2m @ 0x5594675e20] [warning] check_size: Size check failed: asked for 2281x3043, got: 1920x1920\n[h264_v4l2m2m @ 0x5594675e20] [warning] check_size: Size check failed: asked for 2281x3043, got: 1920x1920\n[h264_v4l2m2m @ 0x5594675e20] [warning] check_size: Size check failed: asked for 2281x3043, got: 1920x1920\n[h264_v4l2m2m @ 0x5594675e20] [error] Could not find a valid device\n[h264_v4l2m2m @ 0x5594675e20] [error] can't configure encoder\n[error] Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height\n"
2025-01-02T13:01:25.982399-08:00 allsky [ERROR] Video-1-1066887/MainThread timelapse.generate() [165]: FFMPEG created broken video file, cleaning up
2025-01-02T13:01:37.926446-08:00 allsky [ERROR] Sensor-1-1066888/MainThread sensor.update_sensors() [477]: SensorReadException: Checksum did not validate. Try again.
2025-01-02T13:01:53.286853-08:00 allsky [ERROR] Sensor-1-1066888/MainThread sensor.update_sensors() [477]: SensorReadException: A full buffer was not returned. Try again.
2025-01-02T13:02:08.640948-08:00 allsky [ERROR] Sensor-1-1066888/MainThread sensor.update_sensors() [477]: SensorReadException: A full buffer was not returned. Try again.
2025-01-02T13:02:24.006413-08:00 allsky [ERROR] Sensor-1-1066888/MainThread sensor.update_sensors() [477]: SensorReadException: A full buffer was not returned. Try again.
2025-01-02T13:02:39.362801-08:00 allsky [ERROR] Sensor-1-1066888/MainThread sensor.update_sensors() [477]: SensorReadException: A full buffer was not returned. Try again.
2025-01-02T13:02:54.716441-08:00 allsky [ERROR] Sensor-1-1066888/MainThread sensor.update_sensors() [477]: SensorReadException: A full buffer was not returned. Try again.
2025-01-02T13:03:10.070974-08:00 allsky [ERROR] Sensor-1-1066888/MainThread sensor.update_sensors() [477]: SensorReadException: A full buffer was not returned. Try again.
2025-01-02T13:03:25.425218-08:00 allsky [ERROR] Sensor-1-1066888/MainThread sensor.update_sensors() [477]: SensorReadException: A full buffer was not returned. Try again.
2025-01-02T13:04:26.841176-08:00 allsky [ERROR] Sensor-1-1066888/MainThread sensor.update_sensors() [477]: SensorReadException: A full buffer was not returned. Try again.
[... cut ...]
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": false,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2025-01-02:
@kf6spf It looks like you are trying to use the Raspberry Pi h264 acceleration for ffmpeg.  I believe this codec is limited to a maximum video size of 1920x1080, so you will need to probably set the ffmpeg scaling option to `Height 1080px - [-2:1080]`

kf6spf, 2025-01-02:
Hmm ... I reset FFMPEG Codec to x264 and the scaling as suggested.

I now can generate the Timelapse videos, but they are much smaller ...

-rw-r--r-- 1 mwallis 78966018 Dec 25 07:36 timelapse/20241224/allsky-timelapse_ccd1_20241224_night.mp4
-rw-r--r-- 1 mwallis 17408998 Jan  2 15:29 timelapse/20241229/allsky-timelapse_ccd1_20241229_night.mp4

Michael

----------------------------------------------------------------------
Michael Wallis     kf6spf     (408) 396-9037      ***@***.***

aaronwmorris, 2025-01-03:
The 1080p version of your video is ~25% of the data of the original videos (4k vs 1k).  The smaller file size is in line with this new resolution.

kf6spf, 2025-01-03:
Actually, I think the smaller size reflects intermittent operation of the camera on those days.

-rw-r--r-- 1 mwallis 79073313 Dec 24 07:39 timelapse/20241223/allsky-timelapse_ccd1_20241223_night.mp4
-rw-r--r-- 1 mwallis 78966018 Dec 25 07:36 timelapse/20241224/allsky-timelapse_ccd1_20241224_night.mp4
-rw-r--r-- 1 mwallis 17408998 Jan  2 15:29 timelapse/20241229/allsky-timelapse_ccd1_20241229_night.mp4
-rw-r--r-- 1 mwallis 35676758 Jan  2 15:37 timelapse/20241230/allsky-timelapse_ccd1_20241230_night.mp4
-rw-r--r-- 1 mwallis 36220732 Jan  2 15:42 timelapse/20241231/allsky-timelapse_ccd1_20241231_night.mp4
-rw-r--r-- 1 mwallis 73180286 Jan  2 15:51 timelapse/20250101/allsky-timelapse_ccd1_20250101_night.mp4

Last night is back up to 73Mb in size  close to the 79Mb size previously.

I'll keep an eye on things for the next few days just to verify, but I think it's running properly now.

Net-facing URL is:	https://soso.wallis.com/

Michael

----------------------------------------------------------------------
Michael Wallis     kf6spf     (408) 396-9037      ***@***.***

closed 2025-02-20 by aaronwmorris

morganjrees, 2025-04-11:
which scaling is the best option, I see h264 is meant for RPI but is this better than x264?

aaronwmorris, 2025-04-11:
@morganjrees x264 is just the software ffmpeg option and this has the best compatibility.  The other hardware encoders can be limited in resolution and I have not done as much testing with them.