aaronwmorris/indi-allsky #1758: After indiserver disconnect, capture process stops but never restarts

USA-RedDragon, 2025-01-06:
Hello,

I noticed this behavior recently. 

When the code reaches here ( https://github.com/aaronwmorris/indi-allsky/blob/7b52e1974509fd2ad076ffa8a11ceb4dce0ce950/indi_allsky/capture.py#L238-L240) due to the indiserver connection being disrupted at some point, the capture process is exited and never recovers, however the main thread is still running, so the net effect is that capturing gets stuck permanently until a restart of the container/`allsky.py` can be performed. In Docker I would either expect the capture process to be restarted inside the container or the container to outright exit so a new one can be started.

aaronwmorris, 2025-01-06:
What camera type are you using?

USA-RedDragon, 2025-01-06:
ASI678MC, though the system setup is probably a little undervolted and is the root cause of the disconnection itself. 

My only issue is that the `CaptureWorker` process thread (https://github.com/aaronwmorris/indi-allsky/blob/7b52e1974509fd2ad076ffa8a11ceb4dce0ce950/indi_allsky/capture.py#L53) in indi-allsky quietly stops execution after an indiserver disconnection without the main thread either stopping execution or restarting the `CaptureWorker` process, and because of that the main thread functionally hangs and never attempts to reconnect to the indiserver and start capturing again.

aaronwmorris, 2025-01-06:
That is odd.  I guess we need to track down where the main thread is hanging.  The main thread has no interaction with the indiserver, so it would not directly deal with it.  Are any messages still being logged to the capture process when it hangs?

aaronwmorris, 2025-02-20:
Closing this for now.  Let me know if there is anything else I can do.

closed 2025-02-20 by aaronwmorris