aaronwmorris/indi-allsky #1760: Issue upgrading and running setup.sh

Vertikar, 2025-01-07:
Just ran an update on one of my allsky RPis and tried to update the packages. Looks like it had an issue on rpi.lgpio

git pull relevant output:

```
From https://github.com/aaronwmorris/indi-allsky
 * branch              main       -> FETCH_HEAD
   546d5a48..7b52e197  main       -> origin/main
Updating 546d5a48..7b52e197                           
```

Output from setup.sh

```
Requirement already satisfied: Adafruit-Circuitpython-ConnectionManager in ./virtualenv/indi-allsky/lib/python3.9/site-packages (from adafruit-circuitpython-requests->adafruit-circuitpython-typing->Adafruit-Blinka->-r /opt/indi-allsky/requirements/requirements_gpio.txt (line 1)) (3.1.0)
Installing collected packages: RPi.GPIO
Successfully installed RPi.GPIO-0.7.1
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Found existing installation: RPi.GPIO 0.7.1
Uninstalling RPi.GPIO-0.7.1:
  Successfully uninstalled RPi.GPIO-0.7.1
WARNING: Skipping rpi.lgpio as it is not installed.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting rpi.lgpio
  Using cached https://www.piwheels.org/simple/rpi-lgpio/rpi_lgpio-0.6-py3-none-any.whl (11 kB)
Collecting lgpio>=0.1.0.1 (from rpi.lgpio)
  Using cached lgpio-0.2.2.0.tar.gz (90 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: lgpio
  Building wheel for lgpio (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [14 lines of output]
      running bdist_wheel
      running build
      running build_py
      running build_ext
      building '_lgpio' extension
      swigging lgpio.i to lgpio_wrap.c
      swig -python -o lgpio_wrap.c lgpio.i
      creating build/temp.linux-aarch64-cpython-39
      aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -Isrc -I/opt/indi-allsky/virtualenv/indi-allsky/include -I/usr/include/python3.9 -c lgpio_wrap.c -o build/temp.linux-aarch64-cpython-39/lgpio_wrap.o
      creating build/lib.linux-aarch64-cpython-39
      aarch64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 build/temp.linux-aarch64-cpython-39/lgpio_wrap.o -L/usr/lib -llgpio -o build/lib.linux-aarch64-cpython-39/_lgpio.cpython-39-aarch64-linux-gnu.so
      /usr/bin/ld: cannot find -llgpio
      collect2: error: ld returned 1 exit status
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for lgpio
  Running setup.py clean for lgpio
Failed to build lgpio
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (lgpio)

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...
```

output from support info here: https://gist.github.com/Vertikar/8c82223a0e311a4b39e67554a610e2a7

Vertikar, 2025-01-07:
Worked around it for now by re-running setup.sh but skipped installing the modules.
I'll see if i can re-create it again tomorrow morning re-running setup.sh

Vertikar, 2025-01-08:
Selected no to the first Python module option this time, and selected yes to install Python GPIO modules and was able to recreate it again

[image]

```
Installing collected packages: RPi.GPIO
Successfully installed RPi.GPIO-0.7.1
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Found existing installation: RPi.GPIO 0.7.1
Uninstalling RPi.GPIO-0.7.1:
  Successfully uninstalled RPi.GPIO-0.7.1
WARNING: Skipping rpi.lgpio as it is not installed.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting rpi.lgpio
  Using cached https://www.piwheels.org/simple/rpi-lgpio/rpi_lgpio-0.6-py3-none-any.whl (11 kB)
Collecting lgpio>=0.1.0.1 (from rpi.lgpio)
  Using cached lgpio-0.2.2.0.tar.gz (90 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: lgpio
  Building wheel for lgpio (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [14 lines of output]
      running bdist_wheel
      running build
      running build_py
      running build_ext
      building '_lgpio' extension
      swigging lgpio.i to lgpio_wrap.c
      swig -python -o lgpio_wrap.c lgpio.i
      creating build/temp.linux-aarch64-cpython-39
      aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -Isrc -I/opt/indi-allsky/virtualenv/indi-allsky/include -I/usr/include/python3.9 -c lgpio_wrap.c -o build/temp.linux-aarch64-cpython-39/lgpio_wrap.o
      creating build/lib.linux-aarch64-cpython-39
      aarch64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 build/temp.linux-aarch64-cpython-39/lgpio_wrap.o -L/usr/lib -llgpio -o build/lib.linux-aarch64-cpython-39/_lgpio.cpython-39-aarch64-linux-gnu.so
      /usr/bin/ld: cannot find -llgpio
      collect2: error: ld returned 1 exit status
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for lgpio
  Running setup.py clean for lgpio
Failed to build lgpio
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (lgpio)

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...

karl@allsky-mobile:/opt/indi-allsky $
```

aaronwmorris, 2025-01-08:
This appears to be a problem on Debian 11.  The rpi.lgpio module will not build.  I merged #1761 with some more conditions on when to install some of the GPIO modules.

Vertikar, 2025-01-08:
That sorted it, thanks as always!

closed 2025-01-08 by Vertikar