aaronwmorris/indi-allsky #1776: ./misc/build_indi.sh Warnings

rmarles, 2025-01-14:
Good day:

On my RPi4b4gb, I have a newly imaged high-endurance SD card with Bookworm x64 installed.  Ran `./misc/build_indi.sh` and it was 'warning' laiden.

For example: 
```
[ 11%] Building C object libfli/CMakeFiles/fli.dir/libfli-camera-usb.c.o
/home/admin/Projects/src/indi_3rdparty/libfli/libfli-camera-usb.c: In function fli_camera_usb_open:
/home/admin/Projects/src/indi_3rdparty/libfli/libfli-camera-usb.c:201:25: warning: __builtin_strncpy output may be truncated copying 30 bytes from a string of length 63 [-Wstringop-truncation]
  201 |                         strncpy(DEVICE->devinfo.devnam, (char *) buf, 30);
      |                         ^
/home/admin/Projects/src/indi_3rdparty/libfli/libfli-camera-usb.c:202:25: warning: __builtin_strncpy output may be truncated copying 30 bytes from a string of length 63 [-Wstringop-truncation]
  202 |                         strncpy(DEVICE->devinfo.model, (char *) buf, 30);
      |                         ^
[ 11%] Building CXX object libapogee/CMakeFiles/apogee.dir/AltaModeFsm.cpp.o
[ 12%] Building CXX object libapogee/CMakeFiles/apogee.dir/AltaUsbIo.cpp.o
/home/admin/Projects/src/indi_3rdparty/libfli/libfli-camera-usb.c: In function fli_camera_usb_get_camera_mode_string:
/home/admin/Projects/src/indi_3rdparty/libfli/libfli-camera-usb.c:2736:33: warning: __builtin_strncpy output may be truncated copying between 0 and 31 bytes from a string of length 63 [-Wstringop-truncation]
 2736 |                                 strncpy((char *) dest, (char *) buf, MIN(siz - 1, 31));
      |                                 ^
```

I am still scouring the logs to confirm if I am using any impacted modules and attaching them here in hope it helps.

[Setup.log](https://github.com/user-attachments/files/18411670/Setup.log)

aaronwmorris, 2025-01-14:
These warning are normal messages that occur during the compile of indilib.  You can just ignore them.

rmarles, 2025-01-14:
Excellent.  Thanks for confirming.

closed 2025-01-14 by rmarles