aaronwmorris/indi-allsky #1778: Cache Memory Filling Up

spikked27, 2025-01-15:
I'm running the latest indi-allsky release on a raspberry pi 5 8GB, over the course of a day the cache memory keeps filling up until eventually it starts spilling over into swap memory, any idea what could be causing this?  Let me know if any additional information is needed to diagnose.

Thank you!

[image]

aaronwmorris, 2025-01-15:
I would suggest setting the vm.swappiness on your device to 1 so the system will steal pages from cache before swapping to disk.

```
echo "vm.swappiness = 1" | sudo tee /etc/sysctl.d/90-swappiness.conf

sudo sysctl --system
```

spikked27, 2025-01-15:
> I would suggest setting the vm.swappiness on your device to 1 so the system will steal pages from cache before swapping to disk.
> 
> ```
> echo "vm.swappiness = 1" | sudo tee /etc/sysctl.d/90-swappiness.conf
> 
> sudo sysctl --system
> ```

Applied, will keep an eye on it today to see if cache usage stabilizes

spikked27, 2025-01-15:
> > I would suggest setting the vm.swappiness on your device to 1 so the system will steal pages from cache before swapping to disk.
> > ```
> > echo "vm.swappiness = 1" | sudo tee /etc/sysctl.d/90-swappiness.conf
> > 
> > sudo sysctl --system
> > ```
> 
> Applied, will keep an eye on it today to see if cache usage stabilizes

So unfortunately the cache memory has been creeping up again, increasing from 15% to 40% in the last 3 hours, what's the best way to determine what is causing the potential memory leak?

[image]

aaronwmorris, 2025-01-15:
This is normal behavior.  The cache memory is just caching file system contents in memory.  The `vm.swappiness` setting will cause the kernel to use cache memory instead of swapping when it needs to allocate additional memory.

closed 2025-01-16 by spikked27