aaronwmorris/indi-allsky #1783: 7 blinking lights after restart?

OceanStateAstro, 2025-01-18:
Hello, 

I've tried to install at least 5 times, all success. But when I restart or shutdown my pi 4b I get 7 flashing green lights.

Is it from changing the ARM 64bit to 0? Or getting a bad recent update from Pi OS?

I've tried multiple SD cards all formated and updated with the latest 64bit bullseye but end up getting 7 flashing green lights eventually after reboot/ shutdown.

aaronwmorris, 2025-01-18:
I had to find the documentation for the Raspberry Pi.  7 flashes means the kernel image was not found:

https://www.raspberrypi.com/documentation/computers/configuration.html#led-warning-flash-codes

Maybe you have a bad SDcard?

OceanStateAstro, 2025-01-19:
I have tried multiple SD cards, 32gb, and 512gb. Same issue all times. 🥴

OceanStateAstro, 2025-01-19:
> I have tried multiple SD cards, 32gb, and 512gb. Same issue all times. 🥴

I have purchased a new 64gb SanDisk extreme and 128gb Samsung pro endurance. I will try with new SD cards and post an update later today.

OceanStateAstro, 2025-01-19:
Problem still exists and I don't know what I'm doing wrong. Reformed 64gb SanDisk extreme to fat32 using pi imager. Loaded 64bit Pi OS on Micro SD. 

Booted fine and ran sudo apt-get updates/ upgrade to get updates. Rebooted fine. Changed the ARM 64bit=0. Reboot with no issue. 

Installed indi Allsky and ran sequence. Views in web viewer and everything looked good.

Ended service, shut down pi. (Waited till act lights went off and unplugged USB. 

Replugged in USB to reboot and got 7 green lights again with no OS boot at all.

aaronwmorris, 2025-01-19:
I see what is going on.  You are loading the 64-bit OS, so there is no reason to set arm_64bit=0.  If you set it to 0, your system will no longer function.

This setting is only necessary for the 32-bit OS.

OceanStateAstro, 2025-01-19:
> I see what is going on. You are loading the 64-bit OS, so there is no reason to set arm_64bit=0. If you set it to 0, your system will no longer function.
> 
> This setting is only necessary for the 32-bit OS.

Wow this makes total sense. 🙈 I'm currently installing again. I will not change anything in the /boot/firmware/config.txt and post update. Pretty sure this will fix. 

Thank you much

OceanStateAstro, 2025-01-19:
Issue Resolved! 

I probably misread the instructions about the 64bit ARM. Thank you.

closed 2025-01-19 by OceanStateAstro