aaronwmorris/indi-allsky #1788: Home assistant auto discovery script adds base_topic which prevents auto discovery

1im1, 2025-01-21:
Hello,

I noticed that the Home Assistant auto discovery script did not work for me. It seems that when running `misc//home_assistant_auto_discovery.py` line 773:
https://github.com/aaronwmorris/indi-allsky/blob/07268ee5c8138ec74b08e9aded41864cf040f0f8/misc/home_assistant_auto_discovery.py#L773
adds the “base_topic” to the auto discovery topic. This prevents the auto discovery from working. 
when removing the “base_topic”:
`'topic'    : '/'.join((self.discovery_base_topic, sensor['component'], sensor['object_id'], 'config')),`
everything works as expected.

Let me know if there was a reason to include the base_topic or if you want to change it or if I should do a PR.

Best wishes

Tim

aaronwmorris, 2025-01-21:
The `base_topic` is whatever the base topic is defined within the indi-allsky configuration.  I just ran a test in my lab and the auto discovery appears to be working.

What version of Home Assistant are you using?

1im1, 2025-01-21:
My HA core version is 2025.1.2.

I checked the auto discovery topic that is published and the base_topic is included in the auto discovery topic. That seems to not work since then you get too many levels in the topic. 

per HA documentation the auto discovery topic should be:
`<discovery_prefix>/<component>/[<node_id>/]<object_id>/config`
the script does:
`‘topic'    : '/'.join((self.discovery_base_topic, sensor['component'], base_topic, sensor['object_id'], 'config')),`

- self.discovery_base_topic. :  “<discovery_prefix>”
- sensor['component'] :” <‘component’>/[<node_id>/”
-  base_topic : unclear to me
-  sensor['object_id']  :” <object_id>”

Maybe the problem is that my base_topic and discovery_base_topic are not the same?

In any case it works for me without the base_topic

aaronwmorris, 2025-01-21:
I can work on updating the name of the `base_topic` variable, but this is the `MQTT Base Topic` that is in the indi-allsky config.  It is effectively being used as the `<node _id>` according to the info you posted.

1im1, 2025-01-21:
Then I guess the problem is if (like in my case) that the base_topic has multiple topic levels (e.g.: base_topic = “myHome/myHAinstallation“) which is not allowed for a node id. Maybe omitting the node_id as HA suggests for entities with a unique_id  would be better than using the base_topic

> <node_id>: (Optional): ID of the node providing the topic, this is not used by Home Assistant but may be used to structure the MQTT topic. The ID of the node must only consist of characters from the character class [a-zA-Z0-9_-] (alphanumerics, underscore and hyphen).
> The <node_id> level can be used by clients to only subscribe to their own (command) topics by using one wildcard topic like <discovery_prefix>/+/<node_id>/+/set.
> Best practice for entities with a unique_id is to set <object_id> to unique_id and omit the <node_id>.

aaronwmorris, 2025-01-21:
Merged #1794 with the empty value for unit_of_measurement.

1im1, 2025-01-23:
I saw the changes in merge #1789 for renaming the base topic. THat doesn’t solve it for me, but I might be the exotic one here with a more complex base_topic. This can be closed.

closed 2025-01-27 by 1im1