aaronwmorris/indi-allsky #1789: Home assistant sensors need empty unit_of_measurement to be plotted.

1im1, 2025-01-21:
Hello,

Home assistant needs a unit_of_measurement specified to interpret numbers as numerical. See e.g.: https://github.com/home-assistant/core/issues/45130

In the `misc/home_assistant_auto_discovery.py` no unit_of_measurements are specified for values like SQM.

for e.g. SQM:
https://github.com/aaronwmorris/indi-allsky/blob/07268ee5c8138ec74b08e9aded41864cf040f0f8/misc/home_assistant_auto_discovery.py#L424-L438

This obviously makes sense but makes HA treat this as a string so adding an empty unit_of_measurement (`'unit_of_measurement' : '',`)solves this:

```python 
{
                'component' : 'sensor',
                'object_id' : 'indi_allsky_sqm',
                'config' : {
                    'name' : 'SQM',
                    'unique_id' : 'indi_allsky_sqm_{0}'.format(self.unique_id_base),
                    'state_topic' : '/'.join((base_topic, 'sqm')),
                    'unit_of_measurement' : '',
                    'device'   : {
                        'name' : self.device_name,
                        'identifiers' : [
                            self.device_name,
                        ],
                    },
                },
            },
```
This also applies to others sensors such as stars, cpu_total, K-P index and to a certain degree camera gain.

This also seems to be backward compatible. So once you run the auto discovery script again the sensors will be treated as numerical. 

Best wishes

Tim

aaronwmorris, 2025-01-21:
That is great info.  I will work on this.

aaronwmorris, 2025-01-23:
Merged #1794 to set the unit_of_measurement on additional values.