aaronwmorris/indi-allsky #1790: Pink hue in some images

fferrann, 2025-01-21:
## Issue description
At random times of the day (night or day, doesn't matter) some of the captured images have a pinkish hue to them. Is there an explanation or has somebody experience something similar?

## Hardware
Raspberry PI 4B + Raspberry PI HQ Camera

## Settings
White Balance: disabled (although I've observed the same behavior when enabled)
Bayer Pattern: BGGR (although I've observed the same behavior when set to Auto)
SCNR: Average Neutral

[image]

aaronwmorris, 2025-01-21:
Is the pink hue always a band/bar in the image like that?  I just wanted to make sure this was not the combination of two images.

fferrann, 2025-01-21:
Yes I have only noticed the image being partially pink. Here's another example:

[image]

aaronwmorris, 2025-01-21:
That is really bizarre.  Do you use dark frames?

My only suggestion I can think of is to reseat the ribbon cable on both ends for the camera.

fferrann, 2025-01-21:
I’m not using dark frames at the moment. Seems indeed very odd. I don’t have direct access to the camera right now so I won’t be a me to readjust the ribbon cable but I’ll make sure to test that out when I get a chance.
Meanwhile I’ll have to somehow eliminate the pink frames so that they don’t pollute the startrails or time lapses 🙂.

aaronwmorris, 2025-01-21:
For the white balance, did you mean the libcamera AWB settings?  (There is a different AWB setting that is entirely software based)

aaronwmorris, 2025-01-21:
I found a couple of references to pink/purple images, but not exactly the same behavior.

https://forums.raspberrypi.com/viewtopic.php?t=131161  seems to reference interference, but the pattern is a little different.

fferrann, 2025-01-22:
> I found a couple of references to pink/purple images, but not exactly the same behavior.
> 
> https://forums.raspberrypi.com/viewtopic.php?t=131161 seems to reference interference, but the pattern is a little different.

Indeed that one looks slightly different, almost like some level of noise is being introduced externally. 
The one I'm facing has the images so perfectly split into two tones, that made me wonder about some software processing misbehaving. 
Swapping the ribbon cable or readjusting it won't be possible for some time as the camera is in a remote location, so I'll just live with it.

fferrann, 2025-01-22:
Oddly enough last night I got three frames with grey in them too:

[image]
[image]

fferrann, 2025-01-24:
I was able to get somebody to help me with readjusting the ribbon cable. It turns out the end that connected to the camera had come out loose (or wasn't put properly in the first place). So far I haven't seen any pink hue anymore, so I think this might have solved the issue. Closing this for now.

closed 2025-01-24 by fferrann