aaronwmorris/indi-allsky #1793: Indi-Allsky Crashes - Drops off Network

spikked27, 2025-01-21:
Hello, I placed my all-sky camera outside yesterday for the first time and noticed that the web ui became unresponsive and would drop off of my network, restarting would bring it back up for a bit but eventually (especially while trying to view the latest image) the server would crash again.  I assumed it was a wifi issue so I switched to my 2.5GHz network but the same thing was happening.  I have pretty good signal strength so I'm thinking that this isn't the issue and I see nothing in the logs prior to the crash happening.  When I bring the camera inside everything works great again.  Any idea what could be going on?

I do have many of the below log entries repeating over and over again but MQTT is working fine, I also disabled image transfer over MQTT and turned off the file transfer service but these messages still persist.
`2025-01-20T23:19:15.306243-05:00 allsky python3[1144]: Exception in thread Upload-115:
2025-01-20T23:19:15.306347-05:00 allsky python3[1144]: Traceback (most recent call last):
2025-01-20T23:19:15.306492-05:00 allsky python3[1144]:   File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
2025-01-20T23:19:15.306868-05:00 allsky python3[1144]:     self.run()
2025-01-20T23:19:15.306918-05:00 allsky python3[1144]:   File "/home/pi/indi-allsky/indi_allsky/uploader.py", line 112, in run
2025-01-20T23:19:15.307038-05:00 allsky python3[1144]:     raise e
2025-01-20T23:19:15.307087-05:00 allsky python3[1144]:   File "/home/pi/indi-allsky/indi_allsky/uploader.py", line 108, in run
2025-01-20T23:19:15.307220-05:00 allsky python3[1144]:     self.saferun()
2025-01-20T23:19:15.307294-05:00 allsky python3[1144]:   File "/home/pi/indi-allsky/indi_allsky/uploader.py", line 139, in saferun
2025-01-20T23:19:15.307346-05:00 allsky python3[1144]:     self.processUpload(u_dict)
2025-01-20T23:19:15.307377-05:00 allsky python3[1144]:   File "/home/pi/indi-allsky/indi_allsky/uploader.py", line 457, in processUpload
2025-01-20T23:19:15.307469-05:00 allsky python3[1144]:     response = client.put(**put_kwargs)
2025-01-20T23:19:15.307592-05:00 allsky python3[1144]:                ^^^^^^^^^^^^^^^^^^^^^^^^
2025-01-20T23:19:15.307691-05:00 allsky python3[1144]:   File "/home/pi/indi-allsky/indi_allsky/filetransfer/paho_mqtt.py", line 108, in put
2025-01-20T23:19:15.307722-05:00 allsky python3[1144]:     publish.multiple(
2025-01-20T23:19:15.307743-05:00 allsky python3[1144]:   File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/paho/mqtt/publish.py", line 185, in multiple
2025-01-20T23:19:15.307767-05:00 allsky python3[1144]:     raise ValueError('msgs is empty')
2025-01-20T23:19:15.307789-05:00 allsky python3[1144]: ValueError: msgs is empty
2025-01-20T23:19:23.275023-05:00 allsky kernel: [ 7027.037160] rp1-cfe 1f00110000.csi: Using a link rate of 900 Mbps
2025-01-20T23:19:53.247021-05:00 allsky kernel: [ 7057.010250] rp1-cfe 1f00110000.csi: Using a link rate of 900 Mbps
`

spikked27, 2025-01-21:
Support Info:
`/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 8245488 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux allsky 6.6.62+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
Hostname
allsky

Time
Tue 21 Jan 15:39:41 EST 2025

System timezone
US/Eastern

Uptime
 15:39:41 up  2:54,  1 user,  load average: 0.17, 0.17, 0.28

Memory
               total        used        free      shared  buff/cache   available
Mem:         8245488     1114608     6130896        7952     1107168     7130880
Swap:         524272           0      524272

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             4080976        0   4080976   0% /dev
tmpfs             824560     5920    818640   1% /run
/dev/mmcblk0p2 492308272 13822984 453460876   3% /
tmpfs            4122736      704   4122032   1% /dev/shm
tmpfs               5120       48      5072   1% /run/lock
/dev/mmcblk0p1    522230    68386    453844  14% /boot/firmware
tmpfs             824544      176    824368   1% /run/user/1000

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
46300

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether d8:3a:dd:e8:13:74 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether d8:3a:dd:e8:13:75 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.142/24 brd 192.168.0.255 scope global dynamic noprefixroute wlan0
       valid_lft 6830sec preferred_lft 6830sec
    inet6 fdf8:982b:a51f:cbc4:585f:282:db5e:2eb2/64 scope global dynamic noprefixroute 
[... 1285 lines cut ...]
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": true,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-30,350 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################
`

spikked27, 2025-02-03:
New raspberry pi and SD card, brand new minimal install of indi-allsky (no MQTT or file transfer set up) and it dropped off the network again this morning, it appears to happen around the time that timelapse generation occurs which makes me wonder if it's power related (pulling too much current causing some voltage drop and the pi doesn't like it) but that didn't happen with TJ's allsky so I'm not sure, I'll pull logs when I get home and can power cycle, anything else besides webapp-indi-allsky.log I should look at?

Thanks again for your help!

aaronwmorris, 2025-02-03:
It could be power related.  The Pi5 is a bit more power hungry than the Pi4.  It definitely requires greater than 15W (>3A @ 5v) when generating the timelapse.

The log you should probably check is `/var/log/indi-allsky/indi-allsky.log`.

The other thing you can try is setting up the watchdog so the system reboots if it locks up.
https://github.com/aaronwmorris/indi-allsky/wiki/Raspberry-Pi-Watchdog

spikked27, 2025-02-03:
> It could be power related. The Pi5 is a bit more power hungry than the Pi4. It definitely requires greater than 15W (>3A @ 5v) when generating the timelapse.
>
> The log you should probably check is [...]

I thought I had solved any potential power issues with a 12V 60W power supply and a 12V to 5V 10A (50W) buck converter plus changes to config.txt to tell the pi that the power supply is 25W capable with PSU_MAX_CURRENT=5000.  Will power issues produce any log outputs?  When this first happened I didn't see anything in the log prior to the crash.

aaronwmorris, 2025-02-03:
I would not expect power issues to log much.  They Raspi has no idea how much current is being used.  There might be messages issue if the voltage sags (it can detect voltage), but not for current.  If there is not enough current (for whatever reason), the system is just not able to function.

This may not be the problem, anyway.

I am still not sure what would cause the original errors you posted...

spikked27, 2025-02-04:
[indi-allsky.txt](https://github.com/user-attachments/files/18658296/indi-allsky.txt)

[syslog.txt](https://github.com/user-attachments/files/18658327/syslog.txt)

[webapp-indi-allsky.txt](https://github.com/user-attachments/files/18658338/webapp-indi-allsky.txt)

Log attached, the strange thing is the camera continues to capture images, it's just not connected to the network anymore, startrails and keogram are produced but the timelapse is corrupt and won't play

spikked27, 2025-02-04:
> [indi-allsky.txt](https://github.com/user-attachments/files/18658296/indi-allsky.txt)
>
> [syslog.txt](https://github.com/user-attachments/files/18658327/syslog.txt) [...]

Also worth noting that when I go to generate a timelapse I start to get the undervoltage errors in the syslog, startrail and keogram generation doesn't cause the errors.  When I was running TJ's allsky I wasn't running into any undervoltage errors, not sure if the processing power required by indi is higher or not.

aaronwmorris, 2025-02-05:
I would not expect indi-allsky to cause significantly more load than allsky.  Nothing that would cause undervoltage errors.

It sounds to me like you are just losing the wifi connection.  It is hard to decipher based on the log.  Do you recall how you configured the wifi connection?

spikked27, 2025-02-05:
> I would not expect indi-allsky to cause significantly more load than allsky. Nothing that would cause undervoltage errors.
> 
> It sounds to me like you are just losing the wifi connection. It is hard to decipher based on the log. Do you recall how you configured the wifi connection?

Wifi was set up when imaging the SD card

aaronwmorris, 2025-02-05:
That makes sense.  I think your wifi connection is just dropping.

spikked27, 2025-02-08:
So I went a bit crazy solving the problem, stepped up my power supply to 60V to help with voltage drop with a dedicated 60V-5V 10A buck converter for the pi 5, no more under voltage complaints in the log.  I also added an Intel AX210 wifi card with a E-Key m.2 hat, wifi is much faster now which is an all around improvement.  Not sure if either on their own would've done the trick but it stayed online throughout the night so it's a success regardless.

closed 2025-04-24 by aaronwmorris