aaronwmorris/indi-allsky #1801: indi_webcam_ccd, B and R channel interchanged

wildi, 2025-01-24:
ello

first of all thank you for your great work.

I recently installed IndiAlsky on a RP4B+ (Linux 6.6.62+rpt-rpi-v8) together with an Logitech C925. The webcan is driven by indi_webcam_ccd and INDI's version is 2.2.1. It seems so, that the R and B channel are interchanged.

I found that the drivers fits output has 3 axes, meaning that the RGB values are already debayered. Hence, there is no BAYERPAT entry in the fits header.

In processing.py on lines 636, 642:

```python
        if not len(data.shape) == 2:
            # data is already RGB(fits)  //
``` 
this is recognized. In image.py on lines 1347, 1349:

```python
        if self.config['IMAGE_FILE_TYPE'] in ('jpg', 'jpeg'):
            img_rgb = Image.fromarray(cv2.cvtColor(data, cv2.COLOR_BGR2RGB))
            img_rgb.save(str(tmpfile_name), quality=self.config['IMAGE_FILE_COMPRESSION']['jpg'], exif=jpeg_exif)
``` 
the data is converted and written. I changed the code to:
```python
@@ -1345,7 +1345,8 @@ class ImageWorker(Process):
 
         # write to temporary file
         if self.config['IMAGE_FILE_TYPE'] in ('jpg', 'jpeg'):
-            img_rgb = Image.fromarray(cv2.cvtColor(data, cv2.COLOR_BGR2RGB))
+            #img_rgb = Image.fromarray(cv2.cvtColor(data, cv2.COLOR_BGR2RGB))
+            img_rgb = Image.fromarray(data)
             img_rgb.save(str(tmpfile_name), quality=self.config['IMAGE_FILE_COMPRESSION']['jpg'], exif=jpeg_exif)
         elif self.config['IMAGE_FILE_TYPE'] in ('png',):
             # exif does not appear to work with png

``` 
and the colors are correct. Of course, the color of the cardinal points are now blue instead of read.

The reason why I'm posting at first here is, if I let indi_webcam_ccd write the fits file and open it with gimp, then the colors are correct.

Kind regards, bye, wildi

aaronwmorris, 2025-01-24:
Thank you for reporting this.  The problem is likely around processing.py 636.  When the data is imported from the FITS, it needs to be converted to BGR for the processing pipeline.

I will work on this.

aaronwmorris, 2025-01-24:
Merged #1803 which should fix this issue.

wildi, 2025-01-24:
Hello

after

```shell
git pull && ./setup.sh  &&  sudo shutdown -r now
``` 

red is red and blue blue.  My first bug report after retirement :-) Thanks a lot.

Kind regards, bye wildi

closed 2025-01-29 by wildi