aaronwmorris/indi-allsky #1802: ASI 224mc flickers

newgecko, 2025-01-24:
Hello quick question, 

I have a problem with my Asi224mc, the picture always 

 somehow. Most of the time the picture is normal and sometimes it looks like a colored bayer pattern. that looks very strange. 

I want to test the asi camera 16bit fix, but in which file does the code have to go in exactly (path / filename / which position).

Sorry for the stupid question, you have probably already given the answer x times. I just can't find it.

Greedings Nico 

[image]
[image]

aaronwmorris, 2025-01-24:
I think this is an issue with the ASI224.  There have been 2-3 other instances of this behavior reported.  What I suspect is happening is the sensor is randomly reading the image sensor using a different offset.  Image sensors generally have an overscan area, additional rows and columns that are not normally used.

If the camera suddenly reads the image sensor with one column or row offset by one, it would require a different bayer pattern to have the correct colors.  It is possible that the offset is being correctly reported in the FITS headers, but I do not have a good test case to ensure indi-allsky does the correct process.

aaronwmorris, 2025-02-20:
Let me know if there is anything else I can do!