aaronwmorris/indi-allsky #1807: Remote server not using AWS S3 bucket to serve images

timchurches, 2025-01-28:
The Local RPi4B indi-allsky installation is successfully uploading images to an S3 bucket, before SyncAPI sync to a remote instance running on AWS Lightsail (VPS). The bucket and the objects it contains are public-read and accessible from anywhere, have verified this. The remote server is set to use non-local images. However, it continues to serve images from its local filesystem, not S3. I can't work out why.

Remote server details:

#################################
### indi-allsky support info  ###
#################################
```

Distribution: ubuntu
Release: 24.04
Arch: x86_64
Bits: 64

CPUs: 2
Memory: 936092 kB

System: Generic PC

Linux ip-172-26-15-40 6.8.0-1021-aws #23-Ubuntu SMP Mon Dec  9 23:59:34 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
Hostname
ip-172-26-15-40.ap-southeast-2.compute.internal
ip-172-26-15-40.ap-southeast-2.compute.internal ip-172-26-15-40 

Time
Tue Jan 28 02:35:01 UTC 2025

System timezone
Etc/UTC

Uptime
 02:35:01 up 22:14,  0 user,  load average: 0.06, 0.01, 0.00

Memory
               total        used        free      shared  buff/cache   available
Mem:          936092      532156      125240        3632      457140      403936
Swap:              0           0           0

Filesystems
Filesystem      1K-blocks     Used Available Use% Mounted on
/dev/root        39535100 31430056   8088660  80% /
tmpfs              468044        0    468044   0% /dev/shm
tmpfs              187220      892    186328   1% /run
tmpfs                5120        0      5120   0% /run/lock
efivarfs              128        4       120   4% /sys/firmware/efi/efivars
/dev/nvme0n1p16    901520   135324    703068  17% /boot
/dev/nvme0n1p15    106832     6246    100586   6% /boot/efi
tmpfs               93608       16     93592   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info

system python: Python 3.12.3

indiserver: not found

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: ens5: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 9001 qdisc mq state UP group default qlen 1000
    link/ether 02:fd:c9:1d:09:53 brd ff:ff:ff:ff:ff:ff
    inet 172.26.15.40/20 metric 100 brd 172.26.15.255 scope global dynamic ens5
       valid_lft 2759sec preferred_lft 2759sec
    inet6 2406:da1c:27c:4600:af2d:bf8c:36d1:9407/128 scope global dynamic noprefixroute 
[... 889 lines cut ...]
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20.0,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": false,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20.0,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

Local RPi4B system details in a comment to follow.

timchurches, 2025-01-28:
Local RPi4B system details:

/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 7997316 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux allskycam1 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
Hostname
allskycam1

Time
Tue 28 Jan 13:44:27 AEDT 2025

System timezone
Australia/Sydney

Uptime
 13:44:27 up 1 day,  7:07,  2 users,  load average: 0.88, 0.26, 0.11

Memory
               total        used        free      shared  buff/cache   available
Mem:         7997316     1724936      326884       88956     6216500     6272380
Swap:         204796        7168      197628

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             3725364        0   3725364   0% /dev
tmpfs             799732     1352    798380   1% /run
/dev/mmcblk0p2 122353988 34480240  81642196  30% /
tmpfs            3998656    22376   3976280   1% /dev/shm
tmpfs               5120       16      5104   1% /run/lock
/dev/mmcblk0p1    522230    78262    443968  15% /boot/firmware
tmpfs             799728       44    799684   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
60861

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether d8:3a:dd:a8:78:a8 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.234/24 brd 192.168.0.255 scope global dynamic noprefixroute eth0
       valid_lft 4370sec preferred_lft 4370sec
    inet6 fd4d:8302:fd9b:9645:fc7:3789:2d3:812b/64 scope global dynamic noprefixroute 
       valid_lft 1791sec preferred_lft 1791sec
[... 1138 lines cut ...]
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}Â° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": false,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}Â° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

timchurches, 2025-01-28:
Example image object in target bucket: 

https://ocsn-allsky.s3.ap-southeast-2.amazonaws.com/ccd_8a29f8fb-b626-4189-96f8-6bcf9114a64c/exposures/20250128/day/28_13/ccd1_20250128_130001.jpg

aaronwmorris, 2025-01-28:
Thanks for this info.  Is there a public link to your remote indi-allsky site?

aaronwmorris, 2025-01-28:
And just for giggles, can you restart the flask application on the remote side?  Maybe it is not reading the new config.

```
systemctl --user restart gunicorn-indi-allsky
```

timchurches, 2025-01-28:
Oh, sorry, I should have provided this: https://www.ocsn-allsky.net/indi-allsky/

timchurches, 2025-01-28:
> systemctl --user restart gunicorn-indi-allsky

Have done this. No apparent effect, alas. Which module has the code hat decides where to server the images from?

aaronwmorris, 2025-01-28:
Okay, this is entirely my fault.  I completely forgot I made this change to the logic.  I will work on updating the wiki with this info.

The setting on the remote side has to be manually updated in the database.  The reason for this is because you can have multiple cameras on the remote side and there is no way to manage settings for cameras individually.

```
sqlite> SELECT id, name, friendlyName, web_nonlocal_images, web_local_images_admin FROM camera;
+----+------------------+--------------+---------------------+------------------------+
| id |       name       | friendlyName | web_nonlocal_images | web_local_images_admin |
+----+------------------+--------------+---------------------+------------------------+
| 1  | CCD Simulator    |              | 0                   | 0                      |
| 2  | ZWO CCD ASI290MM |              | 0                   | 0                      |
+----+------------------+--------------+---------------------+------------------------+

sqlite> UPDATE camera SET web_nonlocal_images=1 WHERE id=1;
```

timchurches, 2025-01-28:
No worries! I changed the setting in the database and yep, that works, now serving images from S3. Many thanks for your prompt and thorough support for a kick-arse bit of software.

closed 2025-01-28 by timchurches