aaronwmorris/indi-allsky #1808: Broken package? libssl-dev : Depends: libssl3 (= 3.0.15-1~deb12u1) but 3.0.15-1~deb12u1+rpt1 is to be installed

kf6spf, 2025-01-28:
Hi ...

Trying to install indi-allsky on an RPi 5 (4GB) using an NVMe drive (much faster than mmcblk).

I run ./setup.sh ... but get an abort:

```
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libssl-dev : Depends: libssl3 (= 3.0.15-1~deb12u1) but 3.0.15-1~deb12u1+rpt1 is to be installed
E: Unable to correct problems, you have held broken packages.

```

[indi-allsky-fail.txt](https://github.com/user-attachments/files/18568802/indi-allsky-fail.txt)

I've tried several approaches to fix this, but noting has been successful.

I have indi-allsky running fine on an RPi 4b (4GB) just fine and am not sure where the problem is.

Any help would be appreciated!

aaronwmorris, 2025-01-28:
Let's try a full package update to see if the issue resolves.

```
sudo apt-get update

sudo apt-get dist-upgrade
```

Then rerun setup.sh.

kf6spf, 2025-01-28:
Tried and got a different error ...

`Selecting previously unselected package libgfortran-12-dev:arm64.
Preparing to unpack .../040-libgfortran-12-dev_12.2.0-14_arm64.deb ...
dpkg: unrecoverable fatal error, aborting:
 unable to fsync updated status of 'libgfortran-12-dev:arm64': Input/output error
E: Sub-process /usr/bin/dpkg returned an error code (2)
W: Problem unlinking the file /var/cache/apt/pkgcache.bin - pkgDPkgPM::Go (5: Input/output error)

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...

[allsky-setup.sh.txt](https://github.com/user-attachments/files/18578932/allsky-setup.sh.txt)

`

aaronwmorris, 2025-01-28:
This looks like a disk I/O error.  You might be dealing with a bad SDcard.

What is the output of `sudo dmesg` ?

kf6spf, 2025-01-29:
So ... I shut the system down, moved it to where it could use a wired connection to the net rather than a wifi, and booted it from the SD drive rather than the NVMe. I then rebuilt the Indi and reran setup.sh ... and it seems to have worked.

There were lots of WARNINGS and some DEPRECATED errors during the compiles, so I've added a text file with all the messages.

[allsky-fullRebuild.txt](https://github.com/user-attachments/files/18581264/allsky-fullRebuild.txt)

closed 2025-01-29 by kf6spf

aaronwmorris, 2025-01-29:
Excellent!  All of those messages are normal.