aaronwmorris/indi-allsky #1813: capture.saferun() [235]: Exposure state: BUSY while driver is in state IDLE or OK

wildi, 2025-01-29:
Hello

my setup ist IndiAllsky on a RP4B+ (Linux 6.6.62+rpt-rpi-v8) together with an Logitech C925. The webcam is driven by indi_webcam_ccd and INDI's version is 2.2.1.

From time to time capture.saferun() logs that the camera's state is BUSY while KStars client says it is OK or even IDLE. Under these circumstances no images are taken. If I expose at the same time through KStars I get a normal image and I can repeat the exposure as often as I want. I observed the same behavior  if the driver is indi_libcamera_ccd using am IMX378 camera (on a separate host).

Kind regards, bye, wildi

[image]
[image]

aaronwmorris, 2025-01-29:
Are you connecting to the same indiserver between Kstars and indi-allsky?

wildi, 2025-01-29:
The indiserver/indi_webcam_ccd is running on the same host as indi-allsky. I connect from my laptop, via KStars, Tools, Devices, Client over the LAN to the host in question. There is only one indiserver instance running. The time stamps in the indi-alllsky log file are CET (UTC+1) while  the one's of  KStars client are UTC.

wildi, 2025-01-29:
... here it is:

[si.txt](https://github.com/user-attachments/files/18591930/si.txt)

aaronwmorris, 2025-01-30:
This might be a bug in the indi_webcam_ccd driver.  You might try the indi_v4l2_ccd driver instead.

At a high level, this is the way indilib works (which is what indi-allsky uses to communicate with the camera).  Everything in indilib is asynchronous.  When you request an exposure, the command immediately returns (it does not wait for the exposure to complete).  Eventually, a second method will be automatically started as a callback when the exposure is received.  The time between these two actions, the only way to know if the exposure is taking place is by querying the status of the exposure "control".  Querying the exposure control is what returns "BUSY" or "IDLE".  This is the only way to tell if the exposure command was successful.

There is limited control of an exposure.  While the exposure control is in "BUSY" state, you cannot start another exposure.  In the log messages in your post above, you can see near the top the exposure was aborted.  indi-allsky does track how much time has elapsed for an exposure and will eventually try to abort the exposure (some drivers support this, some do not).  The abort MIGHT work.  If it does, the exposure control will return to "IDLE" state and another exposure can start.  If the control stays in "BUSY" or "ERROR" state, there is nothing indi-allsky can do to resolve this.

wildi, 2025-01-30:
In the meantime I came to same conclusion. I added to capture.py 
```shell
git diff camera/indi.py 
diff --git a/indi_allsky/camera/indi.py b/indi_allsky/camera/indi.py
index 29f7921d..64007c36 100644
--- a/indi_allsky/camera/indi.py
+++ b/indi_allsky/camera/indi.py
@@ -982,7 +982,7 @@ class IndiClient(PyIndi.BaseClient):
 
     def getCcdExposureStatus(self):
         camera_ready, exposure_state = self.ctl_ready(self._ctl_ccd_exposure)
-
+        logger.warning("getCcdExposureStatus: camera_ready: {}, exposure_state: {}".format(camera_ready, exposure_state))
         return camera_ready, exposure_state
 
 
``` 
in order to see what indi-allsky receives, in addition I slowed down 
```shell

@@ -430,6 +429 git diff capture.py
diff --git a/indi_allsky/capture.py b/indi_allsky/capture.py
index a99da07a..5a7904ea 100644
--- a/indi_allsky/capture.py
+++ b/indi_allsky/capture.py,7 @@ class CaptureWorker(Process):
 
                 while True:
                     time.sleep(0.05)
+                    time.sleep(.5) # wildi

```
I assume that it'll be really BUSY. I'm trying to ensure, that indi-allsky really isn't  the cause, before I post. 

FYI: There is a related case. Today indi-allsky took approx. every 300 seconds an image after _Capture-3-2886/MainThread capture._periodic_tasks() [1067]: Periodic tasks triggered_:
```shell
2025-01-30T16:44:03.955697+01:00 pbfls [INFO] Capture-3-2886/MainThread capture.saferun() [234]: Camera last ready: 305.8s
2025-01-30T16:44:03.956097+01:00 pbfls [INFO] Capture-3-2886/MainThread capture.saferun() [235]: Exposure state: BUSY
2025-01-30T16:44:03.956332+01:00 pbfls [INFO] Capture-3-2886/MainThread capture.detectNight() [1387]: Sun altitude: 6.2
2025-01-30T16:44:03.956491+01:00 pbfls [INFO] Capture-3-2886/MainThread capture.detectNight() [1394]: Moon altitude: 15.0, phase 1.7%
2025-01-30T16:44:04.156502+01:00 pbfls [WARNING] Capture-3-2886/MainThread indi.getCcdTemperature() [891]: Camera temperature not supported
2025-01-30T16:44:04.257378+01:00 pbfls [INFO] Capture-3-2886/MainThread indi.getTelescopeRaDec() [881]: Telescope Coord: RA 3.59, Dec 47.10
2025-01-30T16:44:04.265282+01:00 pbfls [WARNING] Capture-3-2886/MainThread miscDb.addNotification() [990]: Not adding existing notification
2025-01-30T16:44:04.265650+01:00 pbfls [WARNING] Capture-3-2886/MainThread indi.abortCcdExposure() [990]: Aborting exposure
2025-01-30T16:44:06.252944+01:00 pbfls [INFO] MainProcess-811/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks
2025-01-30T16:44:06.273067+01:00 pbfls [WARNING] Capture-3-2886/MainThread indi.abortCcdExposure() [995]: Abort not supported
2025-01-30T16:44:06.323492+01:00 pbfls [WARNING] Capture-3-2886/MainThread capture._periodic_tasks() [1067]: Periodic tasks triggered
2025-01-30T16:44:06.329908+01:00 pbfls [INFO] Capture-3-2886/MainThread capture.shoot() [1588]: Taking 0.00328846 s exposure (gain -1)
2025-01-30T16:44:06.430634+01:00 pbfls [INFO] Capture-3-2886/MainThread indi.set_number() [1363]: Setting Duration (s) = 0.0032884632237255573
2025-01-30T16:44:06.431405+01:00 pbfls [INFO] Capture-3-2886/MainThread capture.saferun() [553]: Image queue depth: 0
2025-01-30T16:44:06.432216+01:00 pbfls [INFO] Capture-3-2886/MainThread capture.saferun() [584]: Total time since last exposure 308.1879 s
2025-01-30T16:44:06.473828+01:00 pbfls [INFO] Capture-3-2886/Dummy-3 indi.newMessage() [353]: new Message 2025-01-30T15:44:06: [INFO] Setting number CCD_EXPOSURE
2025-01-30T16:44:06.720831+01:00 pbfls [INFO] Capture-3-2886/Dummy-3 indi.newMessage() [353]: new Message 2025-01-30T15:44:06: [INFO] Buffer Cleared of 6 stale frames.
2025-01-30T16:44:06.899961+01:00 pbfls [INFO] Capture-3-2886/Dummy-3 indi.newMessage() [353]: new Message 2025-01-30T15:44:06: [INFO] Download complete.
2025-01-30T16:44:06.900874+01:00 pbfls [INFO] Capture-3-2886/Dummy-3 indi.newMessage() [353]: new Message 2025-01-30T15:44:06: [DEBUG] Exposure complete
2025-01-30T16:44:06.901691+01:00 pbfls [INFO] Capture-3-2886/Dummy-3 indi.newMessage() [353]: new Message 2025-01-30T15:44:06: [WARNING] Telescope focal length is missing.
2025-01-30T16:44:06.902403+01:00 pbfls [INFO] Capture-3-2886/Dummy-3 indi.newMessage() [353]: new Message 2025-01-30T15:44:06: [WARNING] Telescope aperture is missing.
2025-01-30T16:44:06.914010+01:00 pbfls [INFO] Capture-3-2886/Dummy-3 indi.newMessage() [353]: new Message 2025-01-30T15:44:06: [DEBUG] Uploading file. Ext: fits, Size: 4305600, sendImage? Yes, \
saveImage? No
2025-01-30T16:44:07.121408+01:00 pbfls [INFO] Capture-3-2886/Dummy-3 indi.newMessage() [353]: new Message 2025-01-30T15:44:06: [DEBUG] BLOB transfer took 0.000541141 seconds
2025-01-30T16:44:07.122034+01:00 pbfls [INFO] Capture-3-2886/Dummy-3 indi.newMessage() [353]: new Message 2025-01-30T15:44:06: [DEBUG] Upload complete
2025-01-30T16:44:07.254782+01:00 pbfls [INFO] Image-2-2888/MainThread processing.add() [521]: Image bits: 8, cfa: None
```
If necessary I'll find out what happens here. 

I'll close this issue as soon I convinced myself, thanks, bye, wildi

wildi, 2025-01-30:
P.S: I'll double check it with  indi_v4l2_ccd.

wildi, 2025-01-31:
The kernel killed process indi_webcam_ccd with signal 9 after it filled the memory, the restart of driver was successful but there was no property  CCD_EXPOSURE defined. That appears strange to me since if I connect with KStars client I see the field ```Expose``` and I can take an images.

Why indi-allsky can take images every 300 sec after manually taking an image (KStars client ) should be looked at. At the moment it does not bother me. Soon I'll post my findings on the INDI forum.

Thanks for your help, bye wildi 

Images caught until Jan 30 23:15:07:
```shell
sudo journalctl -k
Jan 31 01:01:53 pbfls kernel: oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=indi_webcam_ccd,pid=28373,uid=1000
Jan 31 01:01:53 pbfls kernel: Out of memory: Killed process 28373 (indi_webcam_ccd) total-vm:4678628kB, anon-rss:2727752kB, file-rss:89628kB, shmem-rss:3996kB, UID:1000 pgtables:8532kB oom_sco>
Jan 31 01:02:00 pbfls kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
Jan 31 01:02:00 pbfls kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
Jan 31 01:02:00 pbfls kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
Jan 31 01:02:00 pbfls kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
Jan 31 01:02:02 pbfls kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd090 fail, reason -52
Jan 31 01:02:02 pbfls kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd095 fail, reason -52
Jan 31 01:02:02 pbfls kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd099 fail, reason -52
Jan 31 01:02:02 pbfls kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd09d fail, reason -52
Jan 31 01:02:02 pbfls kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd0a1 fail, reason -52
Jan 31 01:02:02 pbfls kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd0a5 fail, reason -5
```
At 2025-01-31T05:07:17 I exposed an image with KStars client:
```shell
sudo journalctl /usr/local/bin/indiserver
Jan 30 23:15:07 pbfls indiserver[2225]: 2025-01-30T22:15:07: Driver indi_webcam_ccd: [rawvideo @ 0x55a7d97e10] PACKET SIZE: 2867200, STRIDE: 3200
Jan 31 01:01:56 pbfls indiserver[2225]: 2025-01-31T00:01:49: Driver indi_webcam_ccd: process 28373 killed with signal 9 - Killed
Jan 31 01:01:56 pbfls indiserver[2225]: <delProperty device="INDI Webcam"/>
Jan 31 01:01:56 pbfls indiserver[2225]: 2025-01-31T00:01:49: Driver indi_webcam_ccd: restart #2
Jan 31 01:01:56 pbfls indiserver[2225]: 2025-01-31T00:01:49: Client 8: read: No route to host
Jan 31 01:01:56 pbfls indiserver[2225]: 2025-01-31T00:01:49: Client 13: read: No route to host
Jan 31 01:02:00 pbfls indiserver[2225]: 2025-01-31T00:02:00: Driver indi_webcam_ccd: [video4linux2,v4l2 @ 0x5575eb5480] Not a video capture device.
...
Jan 31 01:07:08 pbfls indiserver[2225]: 2025-01-31T00:07:08: Driver indi_webcam_ccd: indi_webcam_ccd dispatch error: Property CCD_EXPOSURE is not defined in INDI Webcam.
Jan 31 01:12:16 pbfls indiserver[2225]: 2025-01-31T00:12:16: Driver indi_webcam_ccd: indi_webcam_ccd dispatch error: Property CCD_EXPOSURE is not defined in INDI Webcam.
...
Jan 31 06:04:33 pbfls indiserver[2225]: 2025-01-31T05:04:33: Driver indi_webcam_ccd: indi_webcam_ccd dispatch error: Property CCD_EXPOSURE is not defined in INDI Webcam.
Jan 31 06:07:17 pbfls indiserver[2225]: 2025-01-31T05:07:17: Driver indi_webcam_ccd: [video4linux2,v4l2 @ 0x5575e8f460] The V4L2 driver changed the video from 1600x1200 to 1600x896
Jan 31 06:07:17 pbfls indiserver[2225]: 2025-01-31T05:07:17: Driver indi_webcam_ccd: [video4linux2,v4l2 @ 0x5575e8f460] The driver changed the time per frame from 1/30 to 2/15
Jan 31 06:14:49 pbfls indiserver[2225]: 2025-01-31T05:14:49: Driver indi_webcam_ccd: [rawvideo @ 0x5575e97fa0] PACKET SIZE: 2867200, STRIDE: 3200
```
After the manual exposure I got evry 300 sec an image.

wildi, 2025-02-03:
After I convinced myself that indi_webcam_ccd has a memory leak I posted a message on [libindi.org](https://www.indilib.org/forum/development/15279-out-of-memory-killed-process-7900-indi-webcam-ccd.html).

closed 2025-02-03 by wildi