aaronwmorris/indi-allsky #1815: No publishing in MQTT

denisb88, 2025-01-29:
Hi, I have setup the MQTT broker in indi-allsky
I have use 
source virtualenv/indi-allsky/bin/activate
./misc/home_assistant_auto_discovery.py
Topics are created 
In home assistant, sensors are created with unknown status but NEVER NEVER topics are updated.

How can I solve this issue ?
Thanks

aaronwmorris, 2025-01-29:
Can you post your Support Info here?

aaronwmorris, 2025-01-29:
Can you reboot your Raspberry Pi?  It is possible the program just needs to be restarted once the MQTT settings were setup.

denisb88, 2025-01-29:
Reboot done. Same issue. No MQTT publishing

aaronwmorris, 2025-01-29:
I would check the logs at `/var/log/indi-allsky/indi-allsky.log` to see what errors might be reported.

denisb88, 2025-01-29:
Can the non-publication of topics in mqtt explorer be explained by the non-installation of my camera?
The topics are created and visible in mqtt explorer but they are never updated. How often should they update?

aaronwmorris, 2025-01-29:
Okay, I understand now.  All of the MQTT publish activity is based on receiving a valid image from the camera.  Since there is no camera, there is no data to publish.

If you want to test while you wait for your camera, you can re-run setup.sh and use INDI with indi_simulator_ccd.

denisb88, 2025-01-29:
It doesn't work. No change with CCD Simulator. After 10min,  no topics are updated

aaronwmorris, 2025-01-29:
Make sure you restart the indiserver if you switch to the CCD Simulator.  Review the log above to see what is going on.

denisb88, 2025-01-29:
I restarted the raspberry but I no longer have an image produced by CCD Simulator with this message: No Image for 15 minutes. Here is the log of the last startup

> 2025-01-29T21:40:57.144938+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._startup() [279]: indi-allsky release: indi_v2024.09.1
2025-01-29T21:40:57.145100+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._startup() [280]: indi-allsky config level: 20250101.0
2025-01-29T21:40:57.145134+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._startup() [282]: Python version: 3.11.2
2025-01-29T21:40:57.145162+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._startup() [283]: Platform: aarch64
2025-01-29T21:40:57.145189+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._startup() [284]: System Type: Raspberry Pi 5 Model B Rev 1.0#000
2025-01-29T21:40:57.145217+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._startup() [286]: System CPUs: 4
2025-01-29T21:40:57.145242+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._startup() [291]: System memory: 8064 MB
2025-01-29T21:40:57.145401+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._startup() [294]: System uptime: 169s
2025-01-29T21:40:57.318274+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._startCaptureWorker() [341]: Starting Capture-1 worker
2025-01-29T21:40:57.328603+01:00 allsky [INFO] Capture-1-2097/MainThread indi.__init__() [129]: creating an instance of IndiClient
2025-01-29T21:40:57.333640+01:00 allsky [INFO] Capture-1-2097/MainThread indi.__init__() [136]: PyIndi version: 2.1.1
2025-01-29T21:40:57.333739+01:00 allsky [INFO] Capture-1-2097/MainThread capture._initialize() [612]: Connecting to indiserver
2025-01-29T21:40:57.333765+01:00 allsky [INFO] Capture-1-2097/MainThread indi.serverConnected() [356]: Server connected (localhost:7624)
2025-01-29T21:40:57.333792+01:00 allsky [INFO] Capture-1-2097/Dummy-1 indi.newDevice() [224]: new device CCD Simulator
2025-01-29T21:40:57.333818+01:00 allsky [INFO] Capture-1-2097/Dummy-1 indi.newDevice() [224]: new device Telescope Simulator
2025-01-29T21:40:57.379135+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._startImageWorker() [396]: Starting Image-1 worker
2025-01-29T21:40:57.379304+01:00 allsky [WARNING] MainProcess-2093/MainThread processing._load_detection_mask() [2992]: No detection mask defined
2025-01-29T21:40:57.465421+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._startVideoWorker() [461]: Starting Video-1 worker
2025-01-29T21:40:57.465565+01:00 allsky [WARNING] MainProcess-2093/MainThread video._load_detection_mask() [1942]: No detection mask defined
2025-01-29T21:40:57.496507+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._startSensorWorker() [519]: Starting Sensor-1 worker
2025-01-29T21:40:57.570253+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._fileUploadWorkerStart() [579]: Starting Upload-1 worker
2025-01-29T21:40:57.570843+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._fileUploadWorkerStart() [579]: Starting Upload-2 worker
2025-01-29T21:40:57.571054+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks
2025-01-29T21:41:00.501732+01:00 allsky [WARNING] Sensor-1-2101/MainThread sensor.night_day_change() [204]: Day/Night change
2025-01-29T21:41:02.331917+01:00 allsky [INFO] Capture-1-2097/MainThread indi._findTelescopes() [657]: Searching for available telescopes/mounts
2025-01-29T21:41:02.332045+01:00 allsky [INFO] Capture-1-2097/MainThread indi._findTelescopes() [668]:  Detected Telescope Simulator
2025-01-29T21:41:02.346077+01:00 allsky [INFO] Capture-1-2097/MainThread indi.findTelescope() [677]: Found 1 Telescopess
2025-01-29T21:41:02.346117+01:00 allsky [INFO] Capture-1-2097/MainThread indi._findGpss() [690]: Searching for available GPS interfaces
2025-01-29T21:41:02.346204+01:00 allsky [INFO] Capture-1-2097/MainThread indi.findGps() [710]: Found 0 GPSs
2025-01-29T21:41:02.346262+01:00 allsky [WARNING] Capture-1-2097/MainThread capture._initialize() [656]: Connecting to CCD device libcamera_imx477
2025-01-29T21:41:02.346315+01:00 allsky [WARNING] Capture-1-2097/MainThread capture._initialize() [660]: Connecting to Telescope device Telescope Simulator
2025-01-29T21:41:02.355844+01:00 allsky [INFO] Capture-1-2097/MainThread indi.configureDevice() [738]: Setting switch TELESCOPE_TRACK_STATE
2025-01-29T21:41:02.388152+01:00 allsky [INFO] Capture-1-2097/Dummy-1 indi.newMessage() [353]: new Message 2025-01-29T20:41:02: [INFO] Telescope simulator is online. 
2025-01-29T21:41:02.388893+01:00 allsky [INFO] Capture-1-2097/Dummy-1 indi.newMessage() [353]: new Message 2025-01-29T20:41:02: [INFO] Mount is parked. 
2025-01-29T21:41:02.396973+01:00 allsky [INFO] Capture-1-2097/Dummy-1 indi.newMessage() [353]: new Message 2025-01-29T20:41:02: [INFO] Sync is successful. 
2025-01-29T21:41:02.556210+01:00 allsky [INFO] Capture-1-2097/MainThread indi.set_switch() [1391]: Enabling Off (TRACK_OFF)
2025-01-29T21:41:02.556282+01:00 allsky [INFO] Capture-1-2097/MainThread indi.configureDevice() [743]: Setting property (number) GEOGRAPHIC_COORD
2025-01-29T21:41:02.656470+01:00 allsky [INFO] Capture-1-2097/MainThread indi.set_number() [1363]: Setting Lat (dd:mm:ss.s) = 48.069698333740234
2025-01-29T21:41:02.656529+01:00 allsky [INFO] Capture-1-2097/MainThread indi.set_number() [1363]: Setting Lon (dd:mm:ss.s) = 6.517099857330322
[... cut ...]
2025-01-29T21:41:17.264229+01:00 allsky [ERROR] Capture-1-2097/MainThread libcamera._processMetadata() [436]: libcamera sensor black level key not found
2025-01-29T21:41:17.264449+01:00 allsky [INFO] Capture-1-2097/MainThread capture.saferun() [467]: Exposure received in 0.8070 s (0.7970)
2025-01-29T21:41:17.265611+01:00 allsky [ERROR] Image-1-2099/MainThread image.processImage() [327]: Frame not found: /tmp/tmp6q87r3cd.dng
2025-01-29T21:41:23.580247+01:00 allsky [INFO] MainProcess-2093/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks

Nothing about MQTT

aaronwmorris, 2025-01-29:
Its still trying to use use the IMX477 Raspberry Pi camera.  You may want to try a full reboot.

denisb88, 2025-01-29:
What do you mean : full reboot ? Restart my Raspberry ? 
I have already done this twice

denisb88, 2025-01-29:
How to delete my rpi hd camera ?

aaronwmorris, 2025-01-29:
You probably need to go into the Config -> Camera Tab and change the camera interface to INDI and restart.

You can hide unused cameras using information from this link:
https://github.com/aaronwmorris/indi-allsky/wiki/SQL-queries

Although, if indi-allsky tries to activate this camera again, it will automatically unhide it.

denisb88, 2025-01-29:
- removing the raspberry camera
- change interface to INDI
- restart
- discovery commands for home assistant (no errors so mqtt broker correct)
- log without errors
---> no update of sensors with mqtt explorer. I don't understand what's wrong.

aaronwmorris, 2025-01-29:
If you can post more of the log, I think we can figure it out.

denisb88, 2025-01-30:
[log.txt](https://github.com/user-attachments/files/18598703/log.txt)
here is the log

aaronwmorris, 2025-01-30:
Data is definitely being published to MQTT now.  If you are not seeing anything now in home assistant, try running the home assistant auto discovery setup script again.  Sometimes the order between setting up auto discovery and subscribing is important.

denisb88, 2025-01-30:
It's not even in home assistant that the data is not published, I don't see anything published in MQTT publisher so the data is not sent to the broker at all. As of yesterday, I have no history. I restarted the script but it doesn't change anything. Just an update of the topics but they are never updated. For the moment, the problem is not with home assistant but with the publication on the broker. The only updates on the come from running the script. I will receive my camera tomorrow, we will see if that changes anything but in any case, with CCD simulator the data is not published on the broker.

[image]

denisb88, 2025-01-30:
YES. It works but there are two things happening that I don't understand. I deleted the MQTT entries in Home Assistant that were unknown and re-ran the discovery script. I deleted the MQTT configuration in SENSORS. 
I also saw a new series called Allsky appear. I saw a new data series appear in MQTT Explorer. (see attached images). 
The third series updates well (but not the other two). THANKS FOR ALL.

[image]
[image]

aaronwmorris, 2025-01-30:
Fantastic!  I am glad it is working now.

closed 2025-01-30 by denisb88