aaronwmorris/indi-allsky #1823: Running allsky and indi-allsky on same rapberry freeze image

denisb88, 2025-01-31:
I have installed allsky and indi-allsky with this command:
INDIALLSKY_HTTP_PORT=81 INDIALLSKY_HTTPS_PORT=444 ./setup.sh

This worked on both systems all afternoon but this evening I restarted the raspberry and now I have a fixed image on indi-allsky. I stopped allsky thinking it was due to camera sharing but that doesn't change anything. I stopped and restarted the indi-allsky service
The image is not updated.

[image]

aaronwmorris, 2025-01-31:
It sounds like allsky might still be running.  During the day, the exposures are so short, it is conceivable that both packages will run concurrently because they never take images simultaneously.  However, at night, with longer exposures, they start stepping on each other.

The logs would have more info.  `/var/log/indi-allsky/indi-allsky.log`

denisb88, 2025-01-31:
Okay thank you. Is there a way to make them work at the same time at night (perhaps by adjusting the exposure times so that the camera is alternately shared?)

aaronwmorris, 2025-01-31:
No.  There would be no way to synchronize them.

denisb88, 2025-01-31:
Ok I understand. Thanks

closed 2025-01-31 by denisb88