aaronwmorris/indi-allsky #1828: Allsky running, no pictures being taken

wwagonman, 2025-02-01:
Fresh install of Bookworm on a 4B 4G, HQ camera.

I can log into the webserver, but no pictures appear to being taken.  Status is "RUNNING".   Camera works fine outside of indi-allsky. Thanks for helping a newbie out.

Lee

./misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3882912 kB

System: Raspberry Pi 4 Model B Rev 1.2

Linux raspberrypi 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
Hostname
raspberrypi
raspberrypi.home

Time
Sat Feb  1 12:44:48 PM CST 2025

System timezone
US/Central

Uptime
 12:44:48 up 20:00,  3 users,  load average: 0.27, 0.32, 0.29

Memory
               total        used        free      shared  buff/cache   available
Mem:         3882912      920244     1867500       25404     1198056     2962668
Swap:         524284           0      524284

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             1670384        0   1670384   0% /dev
tmpfs             388292     1424    386868   1% /run
/dev/mmcblk0p2 122322964 10989788 105099856  10% /
tmpfs            1941456      440   1941016   1% /dev/shm
tmpfs               5120       16      5104   1% /run/lock
/dev/mmcblk0p1    522230    68458    453772  14% /boot/firmware
tmpfs             388288       48    388240   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
54530

system python: Python 3.11.2

indiserver: not found

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
[... 1217 lines cut ...]
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": false,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################
pi@raspberrypi:~/indi-allsky $

aaronwmorris, 2025-02-02:
Something is wrong with libcamera:

```
rpicam-hello: symbol lookup error: /lib/aarch64-linux-gnu/libcamera.so.0.3: undefined symbol: _ZN7libpisp22compute_optimal_strideER24pisp_image_format_config
```

Let's do a full system upgrade and remove packages that are no longer needed and then reboot.
```
sudo apt-get update

sudo apt-get dist-upgrade

sudo apt-get autoremove

# reboot
```

wwagonman, 2025-02-02:
Worked like a charm!  Thanks.

FYI - This was a repeatable problem. Initially I had installed indi-allsky on a fresh Bookworm Lite SD card.  When it didn't work, I installed it onto a separate fresh Bookworm w/h Desktop SD card.  Both had the same issue.  Your recommended actions fixed both installations.

Thanks aging,

Lee

aaronwmorris, 2025-02-03:
I am glad it fixed the problem.

I have tried to recreate this issue, but I cannot.  I have tried a couple of existing installs and a fresh install, but I cannot reproduce the error.

closed 2025-04-24 by aaronwmorris