aaronwmorris/indi-allsky #1840: View shows lots of overlaping discs

Spaceknighti, 2025-02-04:
Was working great then changed camera to a Altair 178 Colour camera, but the issue was the image showed four overlaped distorted discs, any ideas?

[image]

aaronwmorris, 2025-02-05:
Altair and ToupTek cameras are notorious for not being in INDI_RAW mode by default.

Try setting the following in the INDI config and restart.

```json
{
    "PROPERTIES": {},
    "SWITCHES": {
        "CCD_RESOLUTION": {
            "on": [
                "3040 x 2048"
            ]
        },
        "CCD_CAPTURE_FORMAT": {
            "on": [
                "INDI_RAW"
            ]
        }
    }
}
```

Spaceknighti, 2025-02-05:
Thanks Aaron
  
By the way if there is any help we can give you let me know as I am partner of Altairastro and we are developing an Allsky housing and showing it in use using your program
  

  
Ian

aaronwmorris, 2025-02-05:
Very interesting.  I have not run into any odd problems lately with any Altair cameras lately.  The only problems I have run into were specific to the Altair indi server.

For reference: https://github.com/indilib/indi-3rdparty/issues/681

The hardware has been rock solid.

Spaceknighti, 2025-02-05:
> For reference:  indilib/indi-3rdparty#681

Ok so it should run all Altair cameras ok?
  

  
Ian

aaronwmorris, 2025-02-05:
It should run fine.  This is the full config I have noted for the 178C which include a bandwidth setting that might help with stability when using longer USB3 cables.

Reference:  https://github.com/aaronwmorris/indi-allsky/wiki/INDI-custom-config#altair-hypercam-178c

```json
{
    "PROPERTIES": {
        "CCD_CONTROLS": {
            "BandWidth": 40
        }
    },
    "SWITCHES": {
        "CCD_RESOLUTION": {
            "on": [
                "3040 x 2048"
            ]
        },
        "CCD_CAPTURE_FORMAT": {
            "on": [
                "INDI_RAW"
            ]
        }
    }
}
```

Issue Reference:  https://github.com/aaronwmorris/indi-allsky/issues/378

Spaceknighti, 2025-02-05:
Ok cool thanks, where do I add that? Sorry just getting my head around how the raspberry pi works.
  

  
Attached is my Allsky housing btw
  

  
Ian

aaronwmorris, 2025-02-05:
You would add that ind the `INDI Camera Config (Default)` on the Camera tab and restart/reload.

Unfortunately, image attachments on emails are not posted to GitHub tickets, but I would love to see the housing.

aaronwmorris, 2025-02-06:
You can email me at aaron@aarmor.net

aaronwmorris, 2025-02-07:
I pulled out my Altair GPCAM3 224C and it exhibits the same behavior as your 178C.  As soon as I switch the mode to `INDI_RAW` the image is properly rendered so your camera should work fine.

I have opened the above issue with the indilib development team to investigate.

Spaceknighti, 2025-02-07:
Thanks for that, would be good to get this solved as we have a good range of cameras for Allsky.
  

  
I did try the script you recommended and that solved the 178c issue so thanks for that.

vastmeridian, 2025-02-14:
Hi.  

I've been using the Altair 178C fan-cooled camera for two years now and it works fine, once the config was sorted out. I also used the 224C too, although that's now used elsewhere.

My 178C setup is as follows (extracted from the **INDI Camera Config (Default)** window)

```
{
    "PROPERTIES": {
        "CCD_CONTROLS": {
            "BandWidth": 40
        }
    },
    "SWITCHES": {
        "CCD_RESOLUTION": {
            "on": [
                "3040 x 2048"
            ]
        },
        "CCD_CAPTURE_FORMAT": {
            "on": [
                "INDI_RAW"
            ]
        }
    }
}
```
I am the one who posted the 4" black plastic pipe camera picture on your Facebook page, in case you were wondering. Ive been using the software running on an Intel Atom based SBC for these past two years.

vastmeridian, 2025-02-14:
> You would add that ind the `INDI Camera Config (Default)` on the Camera tab and restart/reload.
> 
> Unfortunately, image attachments on emails are not posted to GitHub tickets, but I would love to see the housing.

It's an impressive housing, Aaron. I'm ditching my existing housing and going to get one of these for my camera.

closed 2025-08-24 by aaronwmorris