aaronwmorris/indi-allsky #1853: SD size?

kf6spf, 2025-02-07:
Didn't see any recommendation in the docs for SD size.

I started out with a 64GB SD card and find that I run out of space if I don't manually trim the number of days that I keep.

I suspect I'll have to go to a 51GB SD and NVMe card for backup.

aaronwmorris, 2025-02-07:
It really depends on the size of the images from your camera.  You can save a significant amount of space by reducing the retention from the default of 10 days down to 2 or 3.

If your camera takes 1920x1080 images, a 128GB card could probably comfortably store 6 months of timelapses and 10 days of images.  If your camera takes 4056x3040 images, you would need 512GB to store a similar amount.

You also do not want to completely fill up your card.  You realistically want at least 50% of the space available so that the card can probably perform wear leveling.

kf6spf, 2025-02-07:
Q: Do I need to keep the Projects directory after I've compiled the software? 

Current camera directory is:

11G	exposures/20250204
16G	exposures/20250205
2.6G	exposures/20250206
85M	timelapse/20250202
83M	timelapse/20250203
4.0K	timelapse/20250205

Ran out of space so it didn't make a timelapse for 20250104. :-(

Michael

----------------------------------------------------------------------
Michael Wallis     kf6spf     (408) 396-9037      ***@***.***

aaronwmorris, 2025-02-07:
You can absolutely remove the Projects directory.

kf6spf, 2025-02-07:
Thanks ...

Here's what I have this morning:

> $ dSpace
> @ 09:25:23 up 15:02,  1 user,  load average: 0.79, 0.88, 0.81
> *** allsky.mrcrer.com ***
> Drive              Size    Used FreeSpace % |----2----4----6----8----0|
> /dev/mmcblk0p2      57G     51G   3.7G   6% |#######################  | /
> tmpfs              512M    6.4M   506M  98% |                         | /tmp
> tmpfs              1.9G    176K   1.9G  99% |                         | /dev/shm
> /dev/mmcblk0p1     510M     56M   455M  89% |##                       | /boot/firmware

And here's the key directories:

> $ dush exposures/*; dush images/*; dush timelapse/*
> 11G	exposures/20250204
> 16G	exposures/20250205
> 13G	exposures/20250206
> 4.0K	images/20250202
> 83M	images/20250203
> 79M	images/20250206
> 4.0K	timelapse/20250202
> 83M	timelapse/20250203
> 79M	timelapse/20250206

So clearly I don't have room for another nights imaging. 

I thought PNG format was smaller files, but I'm changing back to JPEG to see how much of a difference that makes.

What size SD are you recommending people use? And could you put that in the Requirements doc, please?

Michael

----------------------------------------------------------------------
Michael Wallis     kf6spf     (408) 396-9037      ***@***.***

aaronwmorris, 2025-02-07:
I will update the docs with a recommendation.

PNGs will definitely take more space.  I just ran a quick test with my simulator with 4056x3040 images.  JPEGs were ~1MB each.  PNGs were ~10MB each (10x the storage).