aaronwmorris/indi-allsky #1857: Cannot clear file transfer once it has been set up

vastmeridian, 2025-02-07:
This is probably a User Error, but I cannot remove a File Transfer that I no longer need. I’ve removed the host and port number in the config, but I still get pop-up warnings like this from time to time.

[image]

The config options do not allow me to select ‘no protocol’.

[image]

Any ideas?

vastmeridian, 2025-02-07:
Sure. here it is, attached.

[support.info.txt](https://github.com/user-attachments/files/18714388/support.info.txt)

aaronwmorris, 2025-02-07:
You still have image transfers enabled (set to 1) but no hostname set.  You would logically assume the file transfer would be disabled with no hostname, but indi-allsky still tries it.

I have included a small change in #1860 that will stop the attempted transfer without the notifications.

vastmeridian, 2025-02-08:
I did try setting the file transfer to zero, but AllSky wouldn't accept it. 
That might have been while I still had a hostname listed, so I'll try again and confirm.

aaronwmorris, 2025-02-08:
You just need to set the `Transfer Images` value to 0.

vastmeridian, 2025-02-14:
Thank you. This has worked.  I'll close this now.

closed 2025-02-14 by vastmeridian