aaronwmorris/indi-allsky #1871: CROP feature is not working.

JeyJeyC, 2025-02-11:
Hi,

I installed INDI during the last summer. The CROP was not working.
I had time 2 days ago to update INDI.

Then, it's still not working.
I tried to set 200 in the first setting box, and notthing is made on the picture.

I tried to set, for example, 200 in each box, and not any image is any-more taken.

I had this popup fe minutes later : 
Date: 2025-02-11 05:39:24
WARNING: Image worker was restarted more than 10 times

Then back to 0 in each box. Then... now I can't anymore set any number, even, of course, if I save then reload.

If failed to upload the TXT file, so I paste it : 

`/home/Jey/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 8257680 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux skycam 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
Hostname
skycam

Time
Tue 11 Feb 09:29:55 EET 2025

System timezone
Europe/Helsinki

Uptime
 09:29:55 up 1 day, 17:11,  0 user,  load average: 0.72, 0.75, 0.80

Memory
               total        used        free      shared  buff/cache   available
Mem:         8257680     1282512     2346480       35488     4819712     6975168
Swap:         204784           0      204784

Filesystems
Filesystem     1K-blocks      Used Available Use% Mounted on
udev             4087744         0   4087744   0% /dev
tmpfs             825776      5664    820112   1% /run
/dev/nvme0n1p2 961200880  74416040 847414944   9% /
tmpfs            4128832      1952   4126880   1% /dev/shm
tmpfs               5120        48      5072   1% /run/lock
/dev/nvme0n1p1    522230     65620    456610  13% /boot/firmware
/dev/sda1      983378668 109833288 823518816  12% /mnt/usb-drive
tmpfs             825760         0    825760   0% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
43000

[... 1147 lines cut ...]
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": true,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################
`

aaronwmorris, 2025-02-11:
I just tested the crop function and it works.

The crop is specified by giving the `x1, y1` coordinates of the upper-left boundary of the new image, and then the `x2, y2` lower-right boundary.  Think of it as cropping in an image editor.

For example, if you had a 1920x1080 image and you wanted to crop 100 pixels on the left side, you would specify the coordinates like this:

```
x1 = 100, y1 = 0
x2 = 1920, y2 = 1080
```

JeyJeyC, 2025-02-11:
I confirm, it's working.

I'm sorry, I didn't understood that it's working like that. Thinking I have to write the number of pixels to remove on each side.

Thanks a lot !

closed 2025-03-07 by aaronwmorris