aaronwmorris/indi-allsky #1872: FEATURE REQUEST about file transfer (on FTP)

JeyJeyC, 2025-02-11:
Hi :)

INDI is set to upload many files on my FTP.
I access to images trough internet, like other people on my web server.

1 - If we access to the sky picture (remote image) during the file upload, the image is not fully displayed (because the upload is not finished).
Please, could you upload the "lastest" with a temporary filename, then rename it. Like that, we always have the full image.

2 - Please, could you add a LASTEST file for the KEOGRAM? In addition of the DATE-HOur-file

3 - Please, could you add a LASTEST file for the STARTRAIL?

For 2 and 3, like that we can always use the same URL to display the keogram and startrail.

Thank you :)

aaronwmorris, 2025-02-12:
I just merged #1874 which includes atomic file transfers for a few of the transfer methods.  I have updated the wiki.

As for the others, there views for this, but there are no direct links to them.

https://github.com/aaronwmorris/indi-allsky/wiki/Redirect-Views

For instance, to view the latest keogram, you can use:

https://hostname.local/indi-allsky/latestkeogramview

JeyJeyC, 2025-02-24:
> I just merged https://github.com/aaronwmorris/indi-allsky/pull/1874 which includes atomic file transfers for a few of the transfer methods. I have updated the wiki.
Thanks !
When do you plan to push the new update of Indi allsky ?

> As for the others, there views for this, but there are no direct links to them. I have not come up with a good way of showing the links without overcrowding the menu.
> https://github.com/aaronwmorris/indi-allsky/wiki/Redirect-Views
> For instance, to view the latest keogram, you can use:

This is for local, but I use only images sent on the FTP, to my own website.
Currently, the keogram sent on the FTP is named like that : allsky-keogram_ccd1_20250215_night.jpg
I need to send on the FTP "lastest-keogram.jpg" and "lastest-startrail.jpg", like you already manage to send the last taken image "skycam.jpg", to always use the same link on websites.

JeyJeyC, 2025-03-04:
I updated INDI ALLSKY few weeks ago.
My version is indi_v2024.09.1
On Hithub I see that I have the last version https://github.com/aaronwmorris/indi-allsky/releases

What is meaning "merged" ? I don't see any new version.

JeyJeyC, 2025-03-04:
Ok. I succeded to update.
Now I see the possibility to set a specific name.

But, if I change "allsky-keogram_ccd{camera_id:d}_{ts:%Y%m%d_%H%M%S}_{timeofday:s}.{ext}" to "keogram.jpg", we will not anymore have the history of each day?

"In my request", I spoke about the feature to add a specific file ( lastest keogram ) in addition of the current system.

Is it possible to save both at the same time ?

like

allsky-keogram_ccd{camera_id:d}_{ts:%Y%m%d_%H%M%S}_{timeofday:s}.{ext} + lastest.jpg

by using a + or ; between each file?

aaronwmorris, 2025-03-04:
I will think about it.

JeyJeyC, 2025-03-04:
Maybe let the possibility to save like today / choose the name, and by default add automatially a "lastest.jpg".

Thank you :)

JeyJeyC, 2025-04-25:
Did you have the time to shink / look about it ? :)

aaronwmorris, 2025-04-27:
Merged #2097 

I had to go about this in a slightly different way than I originally intended.  There is a new remote folder option for latest images and videos.  All of the latest images and videos will be transferred to this one folder.  There is an option at the bottom of the File Transfer tab to enable uploading the latest images, panoramas, RAW images, and videos.  I grouped all of the Startrails, Timelapse videos, keograms, etc into the "Timelapse" option.

JeyJeyC, 2025-05-13:
Thanks a lot ! Perfect :)

Now here my Skycam is off for 4 months, it's the midnight sun period... :)

aaronwmorris, 2025-06-03:
Closing this for now.  We can re-open if there are any issues when night returns for you!

closed 2025-06-03 by aaronwmorris