aaronwmorris/indi-allsky #1879: No chart data in history range

aaronwmorris, 2025-02-13:
> Hello Aaron. I'm not sure if I am having a similar problem. The heading at the top of the "Charts" page says "No chart data in history range" and then the sensor readings (temp ++) dissappear from the "Device" page. I have experienced this for some weeks and am able to reset it sometimes by reloading the configuration and sometime restarting the rPI5. Every now and then I have to power off and restart. I have the latest software from Github since last week. Any thoughts?
> 
> [image] 

 _Originally posted by @hedrot in [#1227](https://github.com/aaronwmorris/indi-allsky/issues/1227#issuecomment-2656912771)_

aaronwmorris, 2025-02-13:
I would assume this is happening because images are not being captured.  All of the sensor data is stored with the image entry in the database, if there are no images, there is no data for the chart.

The one exception to this is if you have `Daytime Save Images` disabled.  In this case, images are captured, but corresponding entries are not created in the database (the images are just ephemeral).  Since database entries are not created for the images, no sensor data is saved.

hedrot, 2025-02-13:
Thank you very much for the explanation. I switched off "daytime save" as I was trying to save space on the SD card during this test phase here at home before I move the camera to the mountains. Really appreciate the great support.

closed 2025-04-24 by aaronwmorris