aaronwmorris/indi-allsky #1882: cannot install

marco050969, 2025-02-14:
hi,

I wanted to install a new camera with a new PI5 and HQ camera. But I immediately fail the installation with an error. Bookworm 64 bit is used

[image]

aaronwmorris, 2025-02-14:
Did you use any command line arguments when you ran this?

marco050969, 2025-02-14:
I have copy and paste from your GitHub Wiki for new Installation….

marco050969, 2025-02-16:
nope, new system, same error....after the the window to set up drivers for cam...

Klone nach '/home/pi5/Projects/src/indi_core'...
Warnung: Konnte zu klonenden Remote-Branch before nicht finden.
Schwerwiegend: Remote-Branch before nicht im Upstream-Repository origin gefunden
pi5@Testi:~/indi-allsky $ 

after this pic the error came...ever and ever...

[image]

marco050969, 2025-02-16:
i have yet use ./misc/build_indi.sh and this one works...

aaronwmorris, 2025-02-17:
Can you update to the latest code and re-run the build_indi.sh script and let me know what the error is?

marco050969, 2025-02-17:
oki...i test it...

marco050969, 2025-02-17:
New sd card, new system, one install and all works, only the setup from Cam is horrible, but this is another Thing.

[image]

I test tomorrow Zwo Asi 676MC .
Must i use indi under Cam select??

aaronwmorris, 2025-02-17:
No, you do need to use indi for the Rpi HQ camera.  This is kind of confusing, but there are now multiple ways to utilize the HQ Camera (and other CSI connected cameras).

When you select the camera interface in setup.sh, you should use "libcamera" with "libcamera_imx477".  This method uses the rpicam-still command to capture images.  This is a stable interface.

You may also use indi with the indi_libcamera_ccd driver in indi, but the indi interface for the HQ camera should still be considered beta.

marco050969, 2025-02-17:
> No, you do need to use indi for the Rpi HQ camera. This is kind of confusing, but there are now multiple ways to utilize the HQ Camera (and other CSI connected cameras).
>
> When you select the camera [...]

i have use libcamera" with "libcamera_imx477 with the HQ Cam. My question is, must i select Indi for new ASI 676 MC Cam??

aaronwmorris, 2025-02-17:
I am sorry, I misunderstood.  Yes, you have to select indi for the ASI676 camera.

closed 2025-04-24 by aaronwmorris