aaronwmorris/indi-allsky #1885: Enhancement for desktop/landscape orientation playback of timalapse/star trail videos.

vastmeridian, 2025-02-14:
I've noticed in more recent versions of the software, that when viewing timelapses on a laptop or desktop PC (typically in landscape screen orientation) I only get the top part of the timelapse or star trail video visible in the browser window.
Inspecting the HTML code associated with the display of the video (extract below) I see that the video is programmed to display at 100% width. As my video is in portrait format (my camera has a resolution of 2048x3040 or screen format of 2:3) this explains is why the video only shows the top half. If I make my browser window narrower (i.e. a more 'portrait' format) I can see the whole of the video on the screen.
```
<video width="100%" controls="">
    <source src="images/ccd_47ea2a32-841a-41ae-ad0b-b5d756f72184/timelapse/20250205/allsky-startrail_timelapse_ccd1_20250205_night.mp4">
</video>
```
 Is there a way to render the video so that it is all displayed in the browser window when it is played - no matter what the size of the browser window is? I appreciate that best practice is to render the video as you have here - `width="100%"` but I wonder if there is a way of detecting the orientation of the broswer window and rendering appropriately?
If this isn't easily possible, then don't worry. I'll just have to get used to re-sizing my browser window to display the video properly.

aaronwmorris, 2025-02-15:
I have noticed the behavior.  It is even worse if you have a square sensor.  I have not quite figured out what to do about it.

TheRealSunJao, 2025-02-15:
How about removing the fixed width="100%" from the <video> element and replace it with a CSS rule that constrains both width and height while preserving the aspect ratio?

Something like this...?

<div class="row">
  <div class="col-sm-12">
    <video
      style="max-width: 100%; max-height: 100vh; width: auto; height: auto; object-fit: contain;"
      controls
    >
      <source src="images/ccd_47ea2a32-841a-41ae-ad0b-b5d756f72184/timelapse/20250205/allsky-startrail_timelapse_ccd1_20250205_night.mp4">
    </video>
  </div>
</div>

TheRealSunJao, 2025-02-15:
How about removing the fixed width="100%" from the <video> element and replace it with a CSS rule that constrains both width and height while preserving the aspect ratio?

Something like this...?

```
<div class="row">
  <div class="col-sm-12">
    <video
      style="max-width: 100%; max-height: 100vh; width: auto; height: auto; object-fit: contain;"
      controls
    >
      <source src="images/ccd_47ea2a32-841a-41ae-ad0b-b5d756f72184/timelapse/20250205/allsky-startrail_timelapse_ccd1_20250205_night.mp4">
    </video>
  </div>
</div>

```

aaronwmorris, 2025-02-15:
Thank you @TheRealSunJao .  I will be honest, I struggle with CSS.  I will add this to my todo list to investigate.

TheRealSunJao, 2025-02-15:
Sounds good, you are rock'n everything else!

I mod'd the code on the fly on my page and looks okay. Hopefully, it is robust enough for you as well.

Here are before and after (mod'd) shots:

[image]

 
[image]

aaronwmorris, 2025-02-21:
Merged #1907 

I settled on `80vh` so that the text above and below the video are included.

TheRealSunJao, 2025-02-21:
Right, I updated my watch_video.html and tested same 80vh to 89vh. 80vh
appears good conservative value.

closed 2025-03-07 by aaronwmorris