aaronwmorris/indi-allsky #1887: interface to web server

Alias634, 2025-02-16:
Hello, 

I recently install indi allsky everything works great.
I juste need one more things i would like to put interface on a web server like DNS noip or something else.

I not familiar to that skill of config and i asking for help like a tuto step by step for up allsky interface to webui i read wiki but i did'nt understand  how to.

Thanks

Alias634, 2025-02-16:
[image]
[image]
[image]

I have of course configured my home router with port opening 80-443 and add the address and my logs for the dyndns

aaronwmorris, 2025-02-16:
Do you have a remote instance of indi-allsky setup somewhere else?

Alias634, 2025-02-16:
No I just have a local instance

Alias634, 2025-02-16:
i thinks i have wrong config in SyncAPI but i don't know what

Alias634, 2025-02-17:
Forget to add DMZ and some config noip in pi, indi-allsky is available from outside local

closed 2025-02-17 by Alias634

Alias634, 2025-02-17:
That what I learned. I search to much for a simple thing