aaronwmorris/indi-allsky #1888: IMX477 HQ Camera Configuration

OceanStateAstro, 2025-02-16:
Hello, 

I'm having trouble getting the green cast out of my HQ cam image. I've used a few different configuration settings found on here but the image stays the same. I've tried with and without SCNR. Here are examples...

[image]
[image]

Here are my setting as of now. Any help for color correction and optimal camera setting would be greatly appreciated 

Camera tab:

[image]
[image]

Image Tab:

[image]
[image]
[image]

OceanStateAstro, 2025-02-16:
I've changed the file type to DNG(raw) and it fixes the white balance issue, AWB is still off but am concerned about file size. Will changing the file type be okay? 

[image]

aaronwmorris, 2025-02-16:
The problem you are having is the image is turning blue when you enable SCNR?

OceanStateAstro, 2025-02-16:
> The problem you are having is the image is turning blue when you enable SCNR?

Yes it's green when off, then blue when enabled

aaronwmorris, 2025-02-17:
It is fine if you switch to DNG.

I have seen the issue you are experiencing.  It is similar to this issue below, but it affects every image.  I am still tracking this down.

https://github.com/raspberrypi/rpicam-apps/issues/715

OceanStateAstro, 2025-02-17:
> It is fine if you switch to DNG.
> 
> I have seen the issue you are experiencing. It is similar to this issue below, but it affects every image. I am still tracking this down.
> 
> [raspberrypi/rpicam-apps#715](https://github.com/raspberrypi/rpicam-apps/issues/715)

Okay good to know, it seems stable so I'll continue testing and report back. Thank you

aaronwmorris, 2025-03-04:
Opened https://github.com/raspberrypi/rpicam-apps/issues/784 with rpicam-apps for the color balance issue

gmke, 2025-09-25:
with the --ccm option now available for rpicamera-still will this be used by indi-allsky and mqtt-libcamera?

CS, Markus

aaronwmorris, 2025-09-25:
Yes, and the option is already available as of #2370 (and #2383) .  I just missed updating this ticket.