aaronwmorris/indi-allsky #1891: Enhancement: Deleting images/timelapses etc. for a given day

vastmeridian, 2025-02-17:
I've run out of disk space on my AllSky instance and AllSky os DOWN according to the Status: marker. No problem, I can delete some cloudy days' worth of data to free up room.

So I head to Tools --> Generate and select the action to delete all images for a given day: 
[image]

I click on the tick box and then click on Execute and the job is accepted - but there are no deleted files when the job has finished.

I wonder if this notice at the top iof the screen is the reason and I don't currently have a working AllSky.
[image]

If it is, may I suggest an enhancement? On the System --> Utilities screen you have a DANGER section. 
[image]

What about putting the controls there to:
- select a date
- double-confirm that you are sure
- Delete all image files assoicated with that day

aaronwmorris, 2025-02-17:
I will take a look at this.

The "note" about being in a running state is only valid when **creating** assets because the assets are generated in the capture process.  Deleting assets works without indi-allsky running.

aaronwmorris, 2025-08-24:
I have run many tests for this and deleting the images works as I would expect it to.  I am going to leave the interface as is for deleting files for a given day.

closed 2025-08-24 by aaronwmorris