aaronwmorris/indi-allsky #1892: Timelapse not Generating after switching to RAW

HeavensGlare, 2025-02-17:
My timelapses arent generating since I switched to RAW. I switched to RAW because my images stayed very dark on JPEG. I dont know if its due to changing the image type or if I have a setting wrong somewhere. Nor do I have any idea why JPEG doesnt work well for me.

HeavensGlare, 2025-02-17:
/home/pyleallsky/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3882916 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux raspberrypi 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
Hostname
raspberrypi
raspberrypi.attlocal.net raspberrypi.attlocal.net raspberrypi.attlocal.net 

Time
Mon Feb 17 02:56:32 PM CST 2025

System timezone
America/Chicago

Uptime
 14:56:32 up 28 min,  2 users,  load average: 0.98, 1.12, 1.07

Memory
               total        used        free      shared  buff/cache   available
Mem:         3882916     1193832     1747108       30416     1048440     2689084
Swap:         524284           0      524284

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             1670400        0   1670400   0% /dev
tmpfs             388292     8040    380252   3% /run
/dev/mmcblk0p2 961463432 15763408 896847928   2% /
tmpfs            1941456      384   1941072   1% /dev/shm
tmpfs               5120       16      5104   1% /run/lock
/dev/mmcblk0p1    522230    68458    453772  14% /boot/firmware
tmpfs             388288       48    388240   1% /run/user/1000

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
47712

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether d8:3a:dd:ee:b2:e6 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether d8:3a:dd:ee:b2:e7 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.191/24 brd 192.168.1.255 scope global dynamic noprefixroute wlan0
       valid_lft 84734sec preferred_lft 84734sec
    inet6 2600:1700:c1f0:dc80::26/128 scope global dynamic noprefixroute 
[... 1371 lines cut ...]
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": false,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

HeavensGlare, 2025-02-17:
[image]

[image]

HeavensGlare, 2025-02-17:
[image]

aaronwmorris, 2025-02-17:
I cannot think of a reason why the timelapses would fail after switching to RAW.  Can you check the logs?

Timelapses are generated under the Video process, so you may be able to filter through the messages with this.
```
grep Video /var/log/indi-allsky/indi-allsky.log | less
```

HeavensGlare, 2025-02-18:
So I ran your attached command and got the data, but it won't let me copy all of it to paste it here.

HeavensGlare, 2025-02-18:
2025-02-18T07:30:52.797176-06:00 raspberrypi [WARNING] Video-1-1863/MainThread keogram.finalize() [334]: Creating keogram: /var/www/html/allsky/images/ccd_f82b7fb5-615a-4e0a-99dd-5720231eae63/timelapse/20250217/allsky-keogram_ccd1_2025021
7_night.jpg
2025-02-18T07:30:52.901521-06:00 raspberrypi [INFO] Video-1-1863/MainThread keogram.finalize() [354]: Image compressed in 0.1043 s
2025-02-18T07:30:52.922708-06:00 raspberrypi [INFO] Video-1-1863/MainThread miscDb.addThumbnail() [1148]: Adding thumbnail to DB: /var/www/html/allsky/images/ccd_f82b7fb5-615a-4e0a-99dd-5720231eae63/timelapse/20250217/thumbnails/d9c098c1-6383-4833-b999-edc1d2ae7dbc.jpg
2025-02-18T07:30:53.190952-06:00 raspberrypi [WARNING] Video-1-1863/MainThread starTrails.finalize() [412]: Star trails images processed in 118.0 s
2025-02-18T07:30:53.191888-06:00 raspberrypi [WARNING] Video-1-1863/MainThread starTrails.finalize() [413]: Excluded 1656 images
2025-02-18T07:30:53.199303-06:00 raspberrypi [WARNING] Video-1-1863/MainThread starTrails.finalize() [491]: Creating star trail: /var/www/html/allsky/images/ccd_f82b7fb5-615a-4e0a-99dd-5720231eae63/timelapse/20250217/allsky-startrail_ccd1_20250217_night.jpg
2025-02-18T07:30:54.368617-06:00 raspberrypi [INFO] Video-1-1863/MainThread starTrails.finalize() [511]: Image compressed in 1.1680 s
2025-02-18T07:30:54.425668-06:00 raspberrypi [INFO] Video-1-1863/MainThread miscDb.addThumbnail() [1148]: Adding thumbnail to DB: /var/www/html/allsky/images/ccd_f82b7fb5-615a-4e0a-99dd-5720231eae63/timelapse/20250217/thumbnails/fbc0fa7a-8deb-47e6-b9fe-049a360da492.jpg
2025-02-18T07:30:56.254908-06:00 raspberrypi [ERROR] Video-1-1863/MainThread video.generateKeogramStarTrails() [1477]: Not enough frames to generate star trails timelapse: 100
2025-02-18T07:30:56.255391-06:00 raspberrypi [WARNING] Video-1-1863/MainThread video.generateKeogramStarTrails() [1483]: Total keogram/star trail processing in 3166.1 s
2025-02-18T07:30:56.255597-06:00 raspberrypi [WARNING] Video-1-1863/MainThread miscUpload.upload_keogram() [260]: Keogram uploading disabled
2025-02-18T07:30:56.255778-06:00 raspberrypi [WARNING] Video-1-1863/MainThread miscUpload.upload_startrail() [314]: Star trail uploading disabled
2025-02-18T07:30:56.681224-06:00 raspberrypi [INFO] Video-1-1863/MainThread video.generateVideo() [315]: Found 1760 images for timelapse
2025-02-18T07:30:56.737372-06:00 raspberrypi [INFO] Video-1-1863/MainThread video.generateVideo() [361]: Max kpindex: 3.67, ovation: 0, smoke rating: Clear
2025-02-18T07:30:57.962472-06:00 raspberrypi [INFO] Video-1-1863/MainThread miscDb.addVideo() [428]: Adding video ccd_f82b7fb5-615a-4e0a-99dd-5720231eae63/timelapse/20250217/allsky-timelapse_ccd1_20250217_night.mp4 to DB
2025-02-18T07:30:58.097534-06:00 raspberrypi [WARNING] Video-1-1863/MainThread timelapse.generate() [94]: Skipping 4 frames for timelapse
2025-02-18T07:30:58.669359-06:00 raspberrypi [INFO] Video-1-1863/MainThread timelapse.generate() [142]: FFmpeg command: ffmpeg -y -loglevel level+warning -r 25.00 -f image2 -i /tmp/tmpeu8zn0if_timelapse/%05d.jpg -vcodec libx264 -b:v 5000k -pix_fmt yuv420p -movflags +faststart -level 3.1 /var/www/html/allsky/images/ccd_f82b7fb5-615a-4e0a-99dd-5720231eae63/timelapse/20250217/allsky-timelapse_ccd1_20250217_night.mp4
2025-02-18T07:32:48.700046-06:00 raspberrypi [INFO] Video-1-1863/MainThread timelapse.generate() [159]: FFMPEG ran for 110.0077 s
2025-02-18T07:32:48.701842-06:00 raspberrypi [ERROR] Video-1-1863/MainThread timelapse.generate() [160]: FFMPEG failed to generate timelapse, return code: -9
2025-02-18T07:32:48.702319-06:00 raspberrypi [ERROR] Video-1-1863/MainThread timelapse.generate() [161]: FFMPEG output: b'[swscaler @ 0x5596f5aa40] [swscaler @ 0x5596f679d0] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x5596f5aa40] [swscaler @ 0x5596ff8f80] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x5596f5aa40] [swscaler @ 0x559708a530] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x5596f5aa40] [swscaler @ 0x559711bae0] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x5596f5aa40] [swscaler @ 0x55971ad090] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x5596f5aa40] [swscaler @ 0x5596f679d0] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x5596f5aa40] [swscaler @ 0x5596ff9e90] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x5596f5aa40] [swscaler @ 0x55971ad830] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x5596f5aa40] [swscaler @ 0x559708b260] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x5596f5aa40] [swscaler @ 0x559711c810] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x559701eb50] [swscaler @ 0x559702bae0] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x559701eb50] [swscaler @ 0x5596fddfa0] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x559701eb50] [swscaler @ 0x55971b4730] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x559701eb50] [swscaler @ 0x559708e1e0] [warning] deprecated pixel format used, mak [... cut ...]
2025-02-18T07:32:49.185571-06:00 raspberrypi [WARNING] Video-1-1863/MainThread aurora.download_json() [135]: Downloading https://services.swpc.noaa.gov/json/ovation_aurora_latest.json
2025-02-18T07:32:51.134459-06:00 raspberrypi [WARNING] Video-1-1863/MainThread aurora.download_json() [135]: Downloading https://services.swpc.noaa.gov/products/noaa-planetary-k-index.json
2025-02-18T07:32:51.248114-06:00 raspberrypi [WARNING] Video-1-1863/MainThread aurora.processOvationLocationData() [152]: Looking up data for 27.0, -97.0

HeavensGlare, 2025-02-18:
This is a snippet that looked relevant

aaronwmorris, 2025-02-19:
```
FFMPEG failed to generate timelapse, return code: -9
```

The ffmpeg documentation is very lacking when it comes to exit code info

aaronwmorris, 2025-02-19:
Merged #1895 with a new option to generate FFMPEG debug reports.  This might help to track down the problem.

aaronwmorris, 2025-02-19:
Upgrade to the latest code using this guide:  https://github.com/aaronwmorris/indi-allsky/wiki/Updating-indi-allsky

Once you do, you will have a new option in the Config -> Timelapse view to generate a FFMPEG debug report when timelapses are created.  The files will be in the /tmp folder.

HeavensGlare, 2025-02-25:
Just updating that everything is working correctly as of the last few nights. Timelapses are generating in RAW. I did the update you suggested and was waiting for a time lapse to fail so I could generate the report and it was successful.

aaronwmorris, 2025-02-25:
There are certainly worse outcomes :-)  I am glad things are working out.

closed 2025-04-24 by aaronwmorris