aaronwmorris/indi-allsky #1902: Rotate Image causes unexpected scaling

omeganon, 2025-02-20:
I had a need to use Image -> Rotate Image to perform an arbitrary angle rotations and when doing so, I observed that the image scaled by an amount related to the rotation angle. In this [mini timelapse](https://youtube.com/shorts/lIQjg9AsclU?si=FVPLjgAMmy4MY072), the image rotation was changed from 0, to 20, to 15. You can see that 20 introduced a significant shrink scaling of the various components of the image (seemingly independent for each), and 15 less so. I have reverted to a rotation of 0 and the scaling reverted as well. No other parameters were changed. 

Bug or some unexpected relationship to another setting?

omeganon, 2025-02-20:
Hmmm. I think I understand.  It’s the crop [here](https://github.com/aaronwmorris/indi-allsky/blob/fa83b02079bf30a33b0cdccfeb7350c2645a287e/indi_allsky/processing.py#L1156). The camera image is _actually_ smaller after that operation. 

Would an option to rescale back to the original image height/width restore it back to the original size, but with the expected orientation?

aaronwmorris, 2025-02-20:
The crop you found would only remove a single column/row from the image to ensure the resolution is divisible by 2 for the timelapse.

When you rotate an image, the resulting image is larger.  See the following examples:

[image]

[image]

omeganon, 2025-02-20:
I see. But it’s still constrained by the original image dimensions somehow? That would explain why the actual camera image reduces based on the rotation angle. Is there a scaling option to offset that? I see that for Kenovea’s, but not for the base images.

aaronwmorris, 2025-02-20:
Yes, I believe ffmpeg will accept different resolution images in a series of images when converting to video.  The final video resolution is (I am assuming) the resolution of the first image of the series.  Larger images will be scaled down, so the image circle will probably appear smaller.

If you change the resolution in the middle of the night, the video will not be the larger size until the following night when all of the images are the new resolutions.

omeganon, 2025-02-21:
It doesn’t appear to be an ffmpeg issue. The raw images themselves show the change.  Seems to indicate to me that the rotate itself is changing the size of the image. 

[image]

omeganon, 2025-02-21:
It doesn’t appear to be an ffmpeg issue. The base images themselves show the change.  Seems to indicate to me that the rotate itself is changing the visual size of the image per your example.  What I’m trying to achieve is the rotation, but without the reduction in size of the camera image itself.  I expect that the portion of the image that is now rotated out of the original horizontal/vertical image ratio should just be cropped out with no change to the visual size of the camera image. Can this be achieved? Using your example --

[image]

omeganon, 2025-02-21:
I had some time to look at this. I see what needs to be done and it needs a really trivial change to _rotate_angle() to support a new flag for this behavior; a simple if/else to not use the 2 rot[] operations, and use the original image height/width for the WarpAffine for this new behavior, else use the existing logic.

aaronwmorris, 2025-02-21:
I am actually working on this now.  I am going to add two options.

1. Rotate image but keep the same dimensions
2. Allow rotation with an offset

I have the option to maintain the aspect ration added and working, but the offsets are not working like I want them too yet.  It sounds like you might have a better understanding of the rotation.

If you want to add a PR off my dev branch, please do that

omeganon, 2025-02-21:
Sounds good. Thanks!

omeganon, 2025-02-22:
I’m flying back to the States today and have been doing some reading. Pretty sure I know how to handle the offset, and maybe even simplify some of the code a bit. I’ll be able to do some testing with it tomorrow to validate. Thanks for adding the code for the new feature toggles.

aaronwmorris, 2025-02-22:
Sounds good.  If you just want to just submit a different branch of code, I can backport all of my changes to it.  I have not invested much time in what I did.

omeganon, 2025-02-25:
I've been wracking my brain against this for a couple days now and I've finally gone back to basics and think that maybe the code you have currently is exactly what's needed, and that there is no need to specifically take into account the offsets. It seems to do that already, somewhere else in the code that I can't find.

Every time I try to incorporate the offset logic into _rotate_angle (and it seems straightforward where to do it), the image has always been offset too much. Here you can see that the image circle is offset in X by 100, and the image itself is offset in X by 200 --

[image]

If I complete ignore any offset logic in _rotate_angle, the image is properly aligned with the offset circle --

[image]

I think this is the desired behavior, isn't it?

My current diff is really a no-op here --

```
@@ -1134,18 +1134,15 @@ class ImageProcessor(object):
         height, width = self.image.shape[:2]
         center_x = int(width / 2)
         center_y = int(height / 2)
+        x_offset = self.config.get('LENS_OFFSET_X', 0)
+        y_offset = self.config.get('LENS_OFFSET_Y', 0)

-        # not sure what to do here (or if its in the right place)
         if use_offset:
-            #x_offset = self.config.get('LENS_OFFSET_X', 0)
-            #y_offset = self.config.get('LENS_OFFSET_Y', 0)
-            pass
-        else:
-            pass
-
+            # set center of rotation to the lens offset
+            center_x += x_offset
+            center_y += y_offset

-        # consider rotating at center offset
         rot = cv2.getRotationMatrix2D((center_x, center_y), int(angle), 1.0)

@@ -1153,7 +1150,6 @@ class ImageProcessor(object):
             bound_w = width
             bound_h = height
         else:
-            # rotating will change the size of the resulting image
             abs_cos = abs(rot[0, 0])
             abs_sin = abs(rot[0, 1])
```

aaronwmorris, 2025-02-25:
Hah, that is exactly what I was going through.  I cannot figure out what logic OpenCV is applying when changing the center point.

omeganon, 2025-02-26:
Looks like this might have introduced an issue with the histogram chart display? Chart's are not populating, and this error is in the log. Does it mean anything to you?

`2025-02-26T14:45:20.449097-06:00 allsky [ERROR] MainProcess-2228 app.log_exception() [875]: Exception on /indi-allsky/js/charts [GET]#012Traceback (most recent call last):#012  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 1511, in wsgi_app#012    response = self.full_dispatch_request()#012               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 919, in full_dispatch_request#012    rv = self.handle_user_exception(e)#012         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 917, in full_dispatch_request#012    rv = self.dispatch_request()#012         ^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 902, in dispatch_request#012    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/admin/indi-allsky/indi_allsky/flask/misc.py", line 22, in decorated_view#012    return app.ensure_sync(func)(*args, **kwargs)#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/views.py", line 110, in view#012    return current_app.ensure_sync(self.dispatch_request)(**kwargs)  # type: ignore[no-any-return]#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/admin/indi-allsky/indi_allsky/flask/base_views.py", line 1140, in dispatch_request#012    json_data = self.get_objects()#012                ^^^^^^^^^^^^^^^^^^#012  File "/home/admin/indi-allsky/indi_allsky/flask/views.py", line 1262, in get_objects#012    'chart_data' : self.getChartData(camera_id, ts_dt, history_seconds),#012                   ^^^^^^^^^`^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/admin/indi-allsky/indi_allsky/flask/views.py", line 1597, in getChartData#012    col_ma = numpy.ma.masked_array(image_data[:, :, i], mask=numpy_mask)#012             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/numpy/ma/core.py", line 2983, in __new__#012    raise MaskError(msg % (nd, nm))#012numpy.ma.core.MaskError: Mask and data not compatible: data size is 12616704, mask size is 18524416.`

omeganon, 2025-02-27:
This introduced a new issue in flask/views.py for the histogram chart generation. PR to fix it is https://github.com/aaronwmorris/indi-allsky/pull/1936

```
MainProcess-2203 app.log_exception() [875]: Exception on /indi-allsky/js/charts [GET]
Traceback (most recent call last):
  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/indi_allsky/flask/misc.py", line 22, in decorated_view
    return app.ensure_sync(func)(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/views.py", line 110, in view
    return current_app.ensure_sync(self.dispatch_request)(**kwargs)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/indi_allsky/flask/base_views.py", line 1140, in dispatch_request
    json_data = self.get_objects()
                ^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/indi_allsky/flask/views.py", line 1262, in get_objects
    'chart_data' : self.getChartData(camera_id, ts_dt, history_seconds),
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/indi_allsky/flask/views.py", line 1597, in getChartData
    col_ma = numpy.ma.masked_array(image_data[:, :, i], mask=numpy_mask)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/numpy/ma/core.py", line 2983, in __new__
    raise MaskError(msg % (nd, nm))
numpy.ma.core.MaskError: Mask and data not compatible: data size is 12616704, mask size is 18524416.
```

aaronwmorris, 2025-02-27:
I think I am going to go ahead and remove/disable the offset options for now and just get the new "maintain size" deployed.

omeganon, 2025-02-27:
Yes, agreed. That PR I just submitted didn't fully fix the issue. I got an initial successful chart generation, but not after. Chasing this error now and I'll retract that PR until it really works.

```
Exception on /indi-allsky/js/charts [GET]Traceback (most recent call last):
  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/indi_allsky/flask/misc.py", line 22, in decorated_view
    return app.ensure_sync(func)(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/views.py", line 110, in view
    return current_app.ensure_sync(self.dispatch_request)(**kwargs)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/indi_allsky/flask/base_views.py", line 1140, in dispatch_request
    json_data = self.get_objects()
                ^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/indi_allsky/flask/views.py", line 1262, in get_objects
    'chart_data' : self.getChartData(camera_id, ts_dt, history_seconds),
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/indi-allsky/indi_allsky/flask/views.py", line 1597, in getChartData
    col_ma = numpy.ma.masked_array(image_data[:, :, i], mask=numpy_mask)
                                   ~~~~~~~~~~^^^^^^^^^
IndexError: only integers, slices (`:`), ellipsis (`...`), numpy.newaxis (`None`) and integer or boolean arrays are valid indices
```

omeganon, 2025-02-27:
Nevermind, this was me. I somehow renamed `i` to `e` in my codebase here - https://github.com/aaronwmorris/indi-allsky/blob/dev/indi_allsky/flask/views.py#L1595

Not sure how that happened, it's not even a vim keybinding I normally use, but the PR you merged is fine as is.

aaronwmorris, 2025-03-01:
I removed the offset options for now, but I merged #1942 with the new option to maintain the image size.

omeganon, 2025-03-01:
Awesome thanks. I’ll pull main tomorrow.

closed 2025-03-01 by omeganon