aaronwmorris/indi-allsky #1904: New Sensors qwiic

WirthmU, 2025-02-20:
I am setting up a new system and would like to pack as many sensors as possible into it.
For this I use the new Titan Astro  SkyCamOne HAT for RPI5.
The board supports qwiic sensors.

Could you please confirm that indi-allsky would support the following sensors (besides the ASI ZWO camera from which I already get the camera temperature) and which are not on your list of compatible sensors on GitHub:

SCD-30
BME688

aaronwmorris, 2025-02-20:
The BME688 appears to be a variant of the BME680 so that sensor should work, as-is.

I will work on adding SCD-30 support.

aaronwmorris, 2025-02-21:
Merged #1906 with SCD-30 and SCD4x support.  I do not have these sensors, so if you end up using them, please let me know if they work.

pengsloth, 2025-03-08:
I created a quick and crude implementation for two qwiic sensors. The HDC3022 (Temp, Hum, and dew point) and the BMP390 (Temp and Pressure)

[tempSensorHdc302x.txt](https://github.com/user-attachments/files/19141908/tempSensorHdc302x.txt)
[tempSensorBmp3xx.txt](https://github.com/user-attachments/files/19141909/tempSensorBmp3xx.txt)

aaronwmorris, 2025-03-08:
@pengsloth That is really awesome!

I do not want to minimize your effort here, but there are Circuit Python modules for both of these devices.  Would you like me to integrate those?

pengsloth, 2025-03-08:
Oh yes I used those as a basis for these (they were quick to make). I would
be ecstatic if you included those modules for direct support and settings.

WirthmU, 2025-03-08:
I have successfully installed qwiic versions of the BME680 and MLX90640 sensors using the preconfigured sensor parameters in indi-allsky and they perfectly show in the graphs section.

aaronwmorris, 2025-03-09:
> I have successfully installed qwiic versions of the BME680 and MLX90640 sensors using the preconfigured sensor parameters in indi-allsky and they perfectly show in the graphs section.

Great deal!  I have considered creating an image from the MLX90640 thermal data.  You can see my commented code in the sensor source code.

https://github.com/aaronwmorris/indi-allsky/blob/main/indi_allsky/devices/sensors/tempSensorMlx90640.py#L37

aaronwmorris, 2025-03-12:
@pengsloth Merged #1985 with both HDC302x and BMP3xx sensor support.  I do not have these sensors to test, so please let me know how they work.