aaronwmorris/indi-allsky #1909: Docker Build Error

ethanharstad, 2025-02-21:
I'm getting an error on `docker compose build`

`failed to solve: process "/bin/sh -c bash build_indi.sh $INDI_VERSION     && rm -fR /home/allsky/Projects     && sudo apt-get clean && sudo rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*" did not complete successfully: exit code: 1
exit status 1`

Relevant log
```
------
 > [indiserver.indi.allsky 11/11] RUN bash build_indi.sh $INDI_VERSION     && rm -fR /home/allsky/Projects     && sudo apt-get clean && sudo rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*:
0.362 ######################################################
0.363 ### Welcome to the indi-allsky indi compile script ###
0.363 ######################################################
0.371 
0.371 
0.371 Distribution: debian
0.371 Release: 12
0.371 Arch: aarch64
0.371 Bits: 64
0.371 
0.371 CPUs: 4
0.371 Memory: 4144016 kB
0.371 
0.371 Build indi version: ask
0.371 
0.371 Existing INDI: 2.1.1
0.371 
0.371 BUILD_INDI_CORE: false
0.371 BUILD_INDI_3RDPARTY: true
0.371 BUILD_INDI_CAMERA_VENDOR: supported
0.371 
0.371 Running make with 4 processes
0.371 
0.371 Setup proceeding in 10 seconds... (control-c to cancel)
0.371 
10.43 **** Installing packages... ****
10.66 Hit:1 http://deb.debian.org/debian bookworm InRelease
10.71 Hit:2 http://deb.debian.org/debian bookworm-updates InRelease
10.76 Hit:3 http://deb.debian.org/debian-security bookworm-security InRelease
10.85 Reading package lists...
11.51 Reading package lists...
12.14 Building dependency tree...
12.34 Reading state information...
12.77 build-essential is already the newest version (12.9).
12.77 git is already the newest version (1:2.39.5-0+deb12u2).
12.77 ca-certificates is already the newest version (20230311).
12.77 cmake is already the newest version (3.25.1-1).
12.77 whiptail is already the newest version (0.52.23-1+b1).
12.77 fxload is already the newest version (0.0.20081013-2).
12.77 pkgconf is already the newest version (1.8.1-1).
12.77 libavcodec-dev is already the newest version (7:5.1.6-0+deb12u1).
12.77 libavdevice-dev is already the newest version (7:5.1.6-0+deb12u1).
12.77 libboost-dev is already the newest version (1.74.0.3).
12.77 libboost-regex-dev is already the newest version (1.74.0.3).
12.77 libcfitsio-dev is already the newest version (4.2.0-3).
12.77 libcurl4-gnutls-dev is already the newest version (7.88.1-10+deb12u8).
12.77 libdc1394-dev is already the newest version (2.2.6-4).
12.77 libev-dev is already the newest version (1:4.33-1).
12.77 libfftw3-dev is already the newest version (3.3.10-1).
12.77 libftdi1-dev is already the newest version (1.5-6+b2).
12.77 libftdi-dev is already the newest version (0.20-4+b1).
12.77 libgmock-dev is already the newest version (1.12.1-0.2).
12.77 libgphoto2-dev is already the newest version (2.5.30-1).
12.77 libgps-dev is already the newest version (3.22-4.1+b1).
12.77 libgsl-dev is already the newest version (2.7.1+dfsg-5+deb12u1).
12.77 libjpeg-dev is already the newest version (1:2.1.5-2).
12.77 liblimesuite-dev is already the newest version (22.09.1+dfsg-1+b2).
12.77 libnova-dev is already the newest version (0.16-5).
12.77 libraw-dev is already the newest version (0.20.2-2.1).
12.77 librtlsdr-dev is already the newest version (0.6.0-4).
12.77 libtheora-dev is already the newest version (1.1.1+dfsg.1-16.1+b1).
12.77 libtiff-dev is already the newest version (4.5.0-6+deb12u2).
12.77 libusb-1.0-0-dev is already the newest version (2:1.0.26-1).
12.77 libnutclient-dev is already the newest version (2.8.0-7).
12.77 libzmq3-dev is already the newest version (4.3.4-6).
12.77 libahp-gt-dev is already the newest version (1.6.1-2).
12.77 libcamera-dev is already the newest version (0.0.3-6).
12.77 libboost-program-options1.74-dev is already the newest version (1.74.0+ds1-21).
12.77 zlib1g-dev is already the newest version (1:1.2.13.dfsg-1).
12.77 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
12.77 /usr/local/lib
12.77 /usr/local/lib64
------
```

```
 Static hostname: skycam
       Icon name: computer
      Machine ID: 1fdb419fabba4b02ab8e07a0b6457cdd
         Boot ID: 1a254eddc2864760b01d3d81f0e06512
Operating System: Debian GNU/Linux 12 (bookworm)  
          Kernel: Linux 6.6.74+rpt-rpi-2712
    Architecture: arm64
```

aaronwmorris, 2025-02-22:
Thank you for catching this.  Merged #1913 to fix.  There is also an update to the `.env` as I added a new parameter for the build.

ethanharstad, 2025-02-22:
Cool I got halfway there myself. Your PR has resolved this fully.

closed 2025-02-22 by ethanharstad