aaronwmorris/indi-allsky #1918: All sky running but no new image for 15 minutes

abdiastroboy, 2025-02-23:
I am having an issue where images are failing to be generated after a few days of runtime. As a result i notice timelapse not generating (which is how i picked up on it initially ) A reset fixes the issue temporarily but the problem recurs. 

I’ve got loads of storage and done a full system update and indi all sky update etc. i notice the swap file is high almost 100% when it does this but dunno how that affect that. The weather is warmer but rasberry pi works perfectly fine and the temperature are appropriate.

Attached are the config info , in this shot the system had no images generated in the last 30-40 minutes and before that was ok and last nights time lapse was ok. i saw yesterday everything was ok runtime 6 days swap file was up again.

abdiastroboy, 2025-02-23:
/home/ascar5/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 8245504 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux asc 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
Hostname
asc
asc.araboy.local asc.tail1cd7a9.ts.net 

Time
Sun 23 Feb 17:55:40 AEDT 2025

System timezone
Australia/Hobart

Uptime
 17:55:40 up 7 days,  9:30,  2 users,  load average: 0.05, 0.04, 0.00

Memory
               total        used        free      shared  buff/cache   available
Mem:         8245504     2147584      753888      147904     5591136     6097920
Swap:         204784      162320       42464

Filesystems
Filesystem     1K-blocks      Used Available Use% Mounted on
udev             4078704         0   4078704   0% /dev
tmpfs             824560     84784    739776  11% /run
/dev/nvme0n1p2 480233400 182430536 273393644  41% /
tmpfs            4122752      1120   4121632   1% /dev/shm
tmpfs               5120        48      5072   1% /run/lock
/dev/nvme0n1p1    522230     77452    444778  15% /boot/firmware
tmpfs             824544       176    824368   1% /run/user/1000

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
47950

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:58:a7:df brd ff:ff:ff:ff:ff:ff
    inet 10.69.1.123/24 brd 10.69.1.255 scope global dynamic noprefixroute eth0
       valid_lft 37519sec preferred_lft 37519sec
    inet6 fda6:223a:987b:3747:47d4:86b1:6eec:d0f/64 scope global dynamic noprefixroute 
       valid_lft 1763sec preferred_lft 1763sec
    inet6 fe80::c17c:f133:a427:8d03/64 scope link noprefixroute 
[... 1499 lines cut ...]
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": false,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

abdiastroboy, 2025-02-23:
Some info from the log file since no image generation

abdiastroboy, 2025-02-23:
allsky._queueManualTasks() [1228]: Checking for manually submitted tasks
2025-02-23T18:26:15.057817+11:00 asc [INFO] Capture-2-6348/MainThread indi.getTelescopeRaDec() [881]: Telescope Coord: RA 3.49, Dec -42.99
2025-02-23T18:26:14.957602+11:00 asc [INFO] Capture-2-6348/MainThread indi.getCcdTemperature() [896]: Camera temperature: 41.3
2025-02-23T18:26:14.857613+11:00 asc [INFO] Capture-2-6348/MainThread capture.detectNight() [1394]: Moon altitude: -9.3, phase 25.6%
2025-02-23T18:26:14.857582+11:00 asc [INFO] Capture-2-6348/MainThread capture.detectNight() [1387]: Sun altitude: 17.2
2025-02-23T18:26:14.857549+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [235]: Exposure state: ALERT
2025-02-23T18:26:14.857456+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [234]: Camera last ready: 212.8s
2025-02-23T18:26:08.025748+11:00 asc [INFO] Sensor-2-6352/MainThread sensor.saferun() [196]: Sensor SQM: 2.84974
2025-02-23T18:26:08.025704+11:00 asc [INFO] Sensor-2-6352/MainThread sensor.saferun() [192]: Dew Point: 12.5, Frost Point: 6.9, Heat Index: 36.9
2025-02-23T18:26:08.025641+11:00 asc [INFO] Sensor-2-6352/MainThread lightSensorVeml7700.update() [35]: [Light] VEML770 - lux: 7825.7664, light: 16983, white: 46782
2025-02-23T18:26:08.022206+11:00 asc [INFO] Sensor-2-6352/MainThread tempSensorBme280.update() [25]: [THPInternal] BME280 - temp: 37.8c, humidity: 22.2%, pressure: 996.8hPa
2025-02-23T18:26:07.900509+11:00 asc [INFO] Sensor-2-6352/MainThread tempSensorBme280.update() [25]: [THPExternal] BME280 - temp: 22.2c, humidity: 47.5%, pressure: 996.7hPa
2025-02-23T18:26:04.041459+11:00 asc [INFO] MainProcess-2289/MainThread allsky._queueManualTasks() [1228]: Checking for manually submitted tasks
2025-02-23T18:26:03.842205+11:00 asc [INFO] Capture-2-6348/MainThread indi.getTelescopeRaDec() [881]: Telescope Coord: RA 3.48, Dec -42.99
2025-02-23T18:26:03.742020+11:00 asc [INFO] Capture-2-6348/MainThread indi.getCcdTemperature() [896]: Camera temperature: 41.2
2025-02-23T18:26:03.642197+11:00 asc [INFO] Capture-2-6348/MainThread capture.detectNight() [1394]: Moon altitude: -9.3, phase 25.6%
2025-02-23T18:26:03.642155+11:00 asc [INFO] Capture-2-6348/MainThread capture.detectNight() [1387]: Sun altitude: 17.2
2025-02-23T18:26:03.642122+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [235]: Exposure state: ALERT
2025-02-23T18:26:03.642032+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [234]: Camera last ready: 201.6s
2025-02-23T18:25:52.778184+11:00 asc [INFO] Sensor-2-6352/MainThread sensor.saferun() [196]: Sensor SQM: 2.84910
2025-02-23T18:25:52.778140+11:00 asc [INFO] Sensor-2-6352/MainThread sensor.saferun() [192]: Dew Point: 12.4, Frost Point: 6.8, Heat Index: 36.8
2025-02-23T18:25:52.778077+11:00 asc [INFO] Sensor-2-6352/MainThread lightSensorVeml7700.update() [35]: [Light] VEML770 - lux: 7830.3744, light: 16993, white: 46662
2025-02-23T18:25:52.774597+11:00 asc [INFO] Sensor-2-6352/MainThread tempSensorBme280.update() [25]: [THPInternal] BME280 - temp: 37.7c, humidity: 22.1%, pressure: 996.8hPa
2025-02-23T18:25:52.654055+11:00 asc [INFO] Sensor-2-6352/MainThread tempSensorBme280.update() [25]: [THPExternal] BME280 - temp: 22.2c, humidity: 47.2%, pressure: 996.8hPa
2025-02-23T18:25:52.627054+11:00 asc [INFO] Capture-2-6348/MainThread indi.getTelescopeRaDec() [881]: Telescope Coord: RA 3.48, Dec -42.99
2025-02-23T18:25:52.526811+11:00 asc [INFO] Capture-2-6348/MainThread indi.getCcdTemperature() [896]: Camera temperature: 41.1
2025-02-23T18:25:52.426911+11:00 asc [INFO] Capture-2-6348/MainThread capture.detectNight() [1394]: Moon altitude: -9.3, phase 25.6%
2025-02-23T18:25:52.426880+11:00 asc [INFO] Capture-2-6348/MainThread capture.detectNight() [1387]: Sun altitude: 17.2
2025-02-23T18:25:52.426847+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [235]: Exposure state: ALERT
2025-02-23T18:25:52.426759+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [234]: Camera last ready: 190.4s
2025-02-23T18:25:51.039765+11:00 asc [INFO] MainProcess-2289/MainThread allsky._queueManualTasks() [1228]: Checking for manually submitted tasks
2025-02-23T18:25:41.412213+11:00 asc [INFO] Capture-2-6348/MainThread indi.getTelescopeRaDec() [881]: Telescope Coord: RA 3.48, Dec -42.99
2025-02-23T18:25:41.312053+11:00 asc [INFO] Capture-2-6348/MainThread indi.getCcdTemperature() [896]: Camera temperature: 41.1
2025-02-23T18:25:41.212223+11:00 asc [INFO] Capture-2-6348/MainThread capture.detectNight() [1394]: Moon altitude: -9.3, phase 25.6%
2025-02-23T18:25:41.212193+11:00 asc [INFO] Capture-2-6348/MainThread capture.detectNight() [1387]: Sun altitude: 17.3

abdiastroboy, 2025-02-23:
And log data from the time just before exposure was ok till after ( not the exposure line being : ok then changing to ALERT)

abdiastroboy, 2025-02-23:
Capture-2-6348/MainThread capture.saferun() [235]: Exposure state: ALERT
2025-02-23T16:57:48.335964+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [234]: Camera last ready: 47.7s
2025-02-23T16:57:42.494676+11:00 asc [INFO] Sensor-2-6352/MainThread sensor.saferun() [196]: Sensor SQM: 1.91843
2025-02-23T16:57:42.494643+11:00 asc [INFO] Sensor-2-6352/MainThread sensor.saferun() [192]: Dew Point: 14.7, Frost Point: 7.2, Heat Index: 48.1
2025-02-23T16:57:42.494571+11:00 asc [INFO] Sensor-2-6352/MainThread lightSensorVeml7700.update() [35]: [Light] VEML770 - lux: 18452.2752, light: 40044, white: 65535
2025-02-23T16:57:42.491133+11:00 asc [INFO] Sensor-2-6352/MainThread tempSensorBme280.update() [25]: [THPInternal] BME280 - temp: 47.3c, humidity: 15.5%, pressure: 994.9hPa
2025-02-23T16:57:42.370646+11:00 asc [INFO] Sensor-2-6352/MainThread tempSensorBme280.update() [25]: [THPExternal] BME280 - temp: 26.7c, humidity: 35.6%, pressure: 994.9hPa
2025-02-23T16:57:39.239444+11:00 asc [INFO] MainProcess-2289/MainThread allsky._queueManualTasks() [1228]: Checking for manually submitted tasks
2025-02-23T16:57:37.320513+11:00 asc [INFO] Capture-2-6348/MainThread indi.getTelescopeRaDec() [881]: Telescope Coord: RA 2.01, Dec -42.99
2025-02-23T16:57:37.220337+11:00 asc [INFO] Capture-2-6348/MainThread indi.getCcdTemperature() [896]: Camera temperature: 41.8
2025-02-23T16:57:37.120597+11:00 asc [INFO] Capture-2-6348/MainThread capture.detectNight() [1394]: Moon altitude: 1.7, phase 26.2%
2025-02-23T16:57:37.120564+11:00 asc [INFO] Capture-2-6348/MainThread capture.detectNight() [1387]: Sun altitude: 33.0
2025-02-23T16:57:37.120516+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [235]: Exposure state: ALERT
2025-02-23T16:57:37.120363+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [234]: Camera last ready: 36.5s
2025-02-23T16:57:27.249106+11:00 asc [INFO] Sensor-2-6352/MainThread sensor.saferun() [196]: Sensor SQM: 2.14671
2025-02-23T16:57:27.249074+11:00 asc [INFO] Sensor-2-6352/MainThread sensor.saferun() [192]: Dew Point: 14.5, Frost Point: 7.1, Heat Index: 48.0
2025-02-23T16:57:27.249017+11:00 asc [INFO] Sensor-2-6352/MainThread lightSensorVeml7700.update() [35]: [Light] VEML770 - lux: 14953.4208, light: 32451, white: 63165
2025-02-23T16:57:27.245578+11:00 asc [INFO] Sensor-2-6352/MainThread tempSensorBme280.update() [25]: [THPInternal] BME280 - temp: 47.3c, humidity: 15.4%, pressure: 994.8hPa
2025-02-23T16:57:27.125043+11:00 asc [INFO] Sensor-2-6352/MainThread tempSensorBme280.update() [25]: [THPExternal] BME280 - temp: 26.7c, humidity: 35.4%, pressure: 994.8hPa
2025-02-23T16:57:26.237423+11:00 asc [INFO] MainProcess-2289/MainThread allsky._queueManualTasks() [1228]: Checking for manually submitted tasks
2025-02-23T16:57:26.105353+11:00 asc [INFO] Capture-2-6348/MainThread indi.getTelescopeRaDec() [881]: Telescope Coord: RA 2.00, Dec -42.99
2025-02-23T16:57:26.005170+11:00 asc [INFO] Capture-2-6348/MainThread indi.getCcdTemperature() [896]: Camera temperature: 42.0
2025-02-23T16:57:25.905000+11:00 asc [INFO] Capture-2-6348/MainThread capture.detectNight() [1394]: Moon altitude: 1.7, phase 26.2%
2025-02-23T16:57:25.904893+11:00 asc [INFO] Capture-2-6348/MainThread capture.detectNight() [1387]: Sun altitude: 33.1
2025-02-23T16:57:25.904858+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [235]: Exposure state: ALERT
2025-02-23T16:57:25.904761+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [234]: Camera last ready: 25.2s
2025-02-23T16:57:19.013328+11:00 asc [INFO] Capture-2-6348/Dummy-3 indi.newMessage() [353]: new Message 2025-02-23T05:57:19: [ERROR] Exposure failed after 3 attempts. 
2025-02-23T16:57:14.890372+11:00 asc [INFO] Capture-2-6348/MainThread indi.getTelescopeRaDec() [881]: Telescope Coord: RA 2.00, Dec -42.99
2025-02-23T16:57:14.790204+11:00 asc [INFO] Capture-2-6348/MainThread indi.getCcdTemperature() [896]: Camera temperature: 41.6
2025-02-23T16:57:14.690120+11:00 asc [INFO] Capture-2-6348/MainThread capture.detectNight() [1394]: Moon altitude: 1.7, phase 26.2%
2025-02-23T16:57:14.690090+11:00 asc [INFO] Capture-2-6348/MainThread capture.detectNight() [1387]: Sun altitude: 33.1
2025-02-23T16:57:14.690059+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [235]: Exposure state: OK
2025-02-23T16:57:14.689992+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [234]: Camera last ready: 14.0s
2025-02-23T16:57:14.639409+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [584]: Total time since last exposure 15.0328 s
2025-02-23T16:57:14.639376+11:00 asc [INFO] Capture-2-6348/MainThread capture.saferun() [553]: Image queue depth: 0
2025-02-23T16:57:14.639311+11:00 asc [INFO] Capture-2-6348/MainThread indi.set_number() [1363]: Setting Duration (s) = 7.086998812155798e-05
2025-02-23T16:57:14.539073+11:00 asc [INFO] Capture-2-6348/MainThread capture.shoot() [1588]: Taking 0.00007087 s exposure (gain 0)
2025-02-23T16:57:13.235408+11:00 asc [INFO] MainProcess-2289/MainThread allsky._queueManualTasks() [1228]: Checking for manually submitted tasks
2025-02-23T16:57:12.003361+11:00 asc [INFO] Sensor-2-6352/MainThread sensor.saferun() [196]: Sensor SQM: 1.91313
2025-02-23T16:57:12.003326+11:00 asc [INFO] Sensor-2-6352/MainThread sensor.saferun() [192]: Dew Point: 14.7, Frost Point: 7.2, Heat Index: 48.6
2025-02-23T16:57:12.003257+11:00 asc [INFO] Sensor-2-6352/MainThread lightSensorVeml7700.update() [35]: [Light] VEML770 - lux: 18542.5920, light: 40240, white: 65535
2025-02-23T16:57:11.999884+11:00 asc [INFO] Sensor-2-6352/MainThread tempSensorBme280.update() [25]: [THPInternal] BME280 - temp: 47.8c, humidity: 15.2%, pressure: 994.8hPa
2025-02-23T16:57:11.879306+11:00 asc [INFO] Sensor-2-6352/MainThread tempSensorBme280.update() [25]: [THPExternal] BME280 - temp: 26.7c, humidity: 35.4%, pressure: 994.8hPa
2025-02-23T16:57:0

aaronwmorris, 2025-02-23:
For some reason, the camera exposure control has gone into an ALERT status, which is basically an error state.  Hopefully, there is an error earlier in the log that indicates why.

closed 2025-08-24 by aaronwmorris