aaronwmorris/indi-allsky #1924: indi_v4l2_ccd driver no output when multiple cameras connected

promero11, 2025-02-25:
Hello there, my Arducam 1080P Low Light WDR Ultra Wide Angle USB Camera Module for Computer, 2MP CMOS IMX291 160 Degree Fisheye camera works with my windows laptop, but it does not produce an output using the INDI driver on a PC where multiple INDI cameras are present. My goal is to use it with INDI all sky software package. I have tested it with two PCS, with and without other cameras attached, but no output is seen.

aaronwmorris, 2025-02-25:
My guess is that the proper video device needs to be selected in indi.

If you could connect both cameras and post the output of the following command, it could be helpful.

```
source virtualenv/indi-allsky/bin/activate

./misc/camera_properties.py
```

promero11, 2025-03-01:
bash: virtualenv/indi-allsky/bin/activate: No such file or directory
bash: ./misc/camera_properties.py: No such file or directory

Do you think I should cd/ somewhere?

aaronwmorris, 2025-03-01:
You should be in the indi-allsky git checkout folder.

promero11, 2025-03-03:
I found out I can use the SV305 to much better effect. This issue is closed.

closed 2025-03-03 by promero11