aaronwmorris/indi-allsky #1925: INDI Svbony_CCD driver stops transferring Svbony SV305 Pro photos

promero11, 2025-02-25:
I purchased the Svbony SV 305 Pro Camera with IMX 290 chip. The difference between this and the standard camera is a 256 MB buffer. Images successfully display at first, however they halt after two or three, perhaps more. Svbony technical support says it's a driver issue not a camera issue, and original software developer suggests a bandwidth issue. A screeshot of camera config is 

[image]

attached.

Sincerely,

Paul

aaronwmorris, 2025-02-26:
I tested with my SV305 (non-pro) and I did not run into any problems.  I am still testing though

promero11, 2025-02-26:
Is the one you are testing the USB 2.0 camera with the IMX 290 chip? I would like to try that since it's been tested. What are your thoughts about the effect that my pro model has a USB 3.2 connector and cable has on bandwidth?.

aaronwmorris, 2025-02-26:
USB3 vs USB2 is probably the only major difference between these models.  The thought had crossed my mind.  If you have USB 2.0 only ports, you might try using those instead of 3.0 ports.

I looked through the properties on my camera, but I do not see any USB Bandwidth related settings like ZWO cameras have.  You could try posting your Support Info here to see if the USB3 functionality provides any more settings.

The other option I have for you is the SVBony SDK was upgraded in the latest release of indilib.  You could try re-running `./misc/build_indi.sh` and recompiling at version v2.1.2.1

promero11, 2025-02-26:
Hi Aaron,
I will try the standard SV305  amdtry the SV305 Pro on a USB 2 port and using the new SDK this friday and report back.M

Clear Skies,

Paul

closed 2025-03-03 by promero11