aaronwmorris/indi-allsky #1941: Camera not recognized

cdvideo2, 2025-02-28:
Quick question on Indi allsky. I changed my camera to a 220 degree camera view same model as previous camera. Indi would not show the image from the camera. I rebooted with Allsky and it found it immediately . I like some of the features of Indi Allsky but why would it not see the camera. Both cameras are OV5647but different mfg. Allsky loads them both no problem

cdvideo2, 2025-02-28:
OOPS came on but all green

aaronwmorris, 2025-03-01:
Try enabling SCNR Average Neutral to remove the green bias from the image.

cdvideo2, 2025-03-01:
Yes I got it sorted out. Thanks 
I put the cart before the horse in stead of waiting for it to come up with a image. Allsky had the image instantly , my reason of concern of it not working. 
TKS
Dan C