aaronwmorris/indi-allsky #1943: Wrong sensor ID is used

photon1503, 2025-03-01:
Updated to latest version from 2025-03-01 to have the dew points for each sensor, which works.

But now the selected Fan Temp Sensor does not link to the configured sensor.

Configured for FAN: is 

[image]

but in the log we see that another sensor is taken for the delta calculation.

[image]

I guess it's because the NVMe sensor has the same ID (10) as the BME280 sensor

[image]

photon1503, 2025-03-01:
Update: Took NVMe sensor as input for fan, but got instead RH for Sensor B

```
2025-03-01T17:56:07.709515+01:00 amosallsky [INFO] Sensor-5-1189/MainThread sensor.check_fan_thresholds() [563]: Fan threshold current: **42.2,** target: 25.0, delta: 17.2 (85%)
2025-03-01T17:56:07.709479+01:00 amosallsky [INFO] Sensor-5-1189/MainThread sensor.check_dew_heater_thresholds() [524]: Dew Heater threshold current: 13.2, target: -0.8, delta: 14.1 (25%)
2025-03-01T17:56:07.709428+01:00 amosallsky [INFO] Sensor-5-1189/MainThread sensor.saferun() [194]: Dew Point: -0.8, Frost Point: -1.6, Heat Index: 48.6
2025-03-01T17:56:07.709067+01:00 amosallsky [INFO] Sensor-5-1189/MainThread tempSensorBme280.update() [25]: [Temperatur Gehäuse] BME280 - temp: 11.6c, humidity: 42.0%, pressure: 1011.5hPa
2025-03-01T17:56:07.588908+01:00 amosallsky [INFO] Sensor-5-1189/MainThread tempSensorBme280.update() [25]: [Temperatur Kuppel] BME280 - temp: 13.2c, humidity: **42.2**%, pressure: 1011.5hPa
```

photon1503, 2025-03-02:
Guess this is the problematic line in sensor.py ?

[image]

aaronwmorris, 2025-03-02:
Yes, I see the problem now.

aaronwmorris, 2025-03-02:
Merged #1949 to fix.  Let me know how that works.

photon1503, 2025-03-02:
Thank you Aaron, works like a charm now :-D

closed 2025-03-02 by photon1503