aaronwmorris/indi-allsky #1947: pyCurl IP Camera config in Docker

giangitesta, 2025-03-02:
Hi, I would like to use a docker installation of Indi AllSky to receive an image from a Reolink IP Camera. Trying Indi AllSky in a VM without docker works perfectly. In docker I cannot configure it correctly and as soon as I set Camera Interface = pyCurlCamera the Indi server stops working. What should I do?
Thank You

aaronwmorris, 2025-03-02:
When you change the camera interface to pycurl, are you leaving the INDIALLSKY_INDI_CCD_DRIVER `indi_simulator_ccd` in the .env file?

giangitesta, 2025-03-02:
Hi @aaronwmorris, 
yes the .env file is set to indi_simulator_ccd.
I restarted the server and now it works. 
Is a Docker installation reliable enough to put in my domotic docker server or should I think about an independent solution?
Regards
 G

aaronwmorris, 2025-03-02:
It should be fairly reliable.  There are no real special hacks or anything in the docker builds that make it unique or fragile.

closed 2025-04-24 by aaronwmorris