aaronwmorris/indi-allsky #1954: Generate KEOGRAM each X minutes

JeyJeyC, 2025-03-04:
Hi :)

We already spoke about this during the last summer on a forum.

Please, could you add a feature to be able to generate the KEOGRAM continuously?

Like the IRF at Kiruna is doing since at least 10 years:
https://www.irf.se/en/forskning/kago/firmamentkamera/
This keogram is generated each minute in night time.

The need : following, in live, the evolution of the sky (for northern light).
I'm working "around" northern light, and I need to follow the northern light activity (by curiosity, to show to people, to understand the evolution in the sky we have over the head, ...).
Having the keogram at the end of the night is useless for us.

You explained that is using lot of ressources and time:
https://www.cloudynights.com/topic/785514-new-all-sky-camera-management-software-indi-allsky/page-32#entry13476684

Here I gave some ideas https://www.cloudynights.com/topic/785514-new-all-sky-camera-management-software-indi-allsky/page-32#entry13479584

1 - at the moment when each picture is taken, save it at a smaller size copy  (like 600*600), like that it will use less ressources / time to generate the keogram.
2 - better : when each image is taken, save only the central column of pixels (1 x 3000 pixels for example) to stick them all at each generation of the keogram. It will only save and use the necessary data. 

Please :)

aaronwmorris, 2025-03-04:
I recall :-)  I will add it to my backlog.

aaronwmorris, 2025-03-09:
This is my first pass for this functionality.  Merged #1973 .  It is not integrated with file transfers yet, but the keogram is generated with every image.

There are a couple of new options near the keogram section in the config.

JeyJeyC, 2025-03-10:
Wouah! Great!

Thank you !

So, how do you manage it? What did you decide to build it?

aaronwmorris, 2025-03-10:
Basically, I use the same keogram processing as end-of-night and just use it to generate intermediate output.  With every image, it is rotated and the center line extracted.  The data is just kept in memory.  At most, this adds 40-50MB of memory usage.

When you read image data with opencv, the data is held in a numpy array.  If you are not familiar, numpy is just a matrix processing library.  Every 25 images or so, indi-allsky will save the numpy data to disk in the native numpy format.  This way, when the program is restarted (or crashes), it can re-read the numpy data to memory and pick up where it left off.  Once the keogram reaches the maximum size, it will start removing the oldest entries.  You end up with a rolling keogram.

One issue that is a special case... if you change the image size or rotation options it will change the dimensions of the keogram data.  If this happens, indi-allsky will just throw away the current data and start the keogram from scratch.

JeyJeyC, 2025-03-24:
Just a feedback : it's working very well !
Thank you!

(waiting the possibility to upload it on the FTP :D ).

aaronwmorris, 2025-03-26:
Merged #2023 to support filetransfers of the realtime keogram.

I also renamed the data file so it is unique per camera.  If you want to keep your existing data, you can copy the file to the new name.

```
cp -i /var/lib/indi-allsky/realtime_keogram_store.npy /var/lib/indi-allsky/realtime_keogram_store_ccd1.npy
```

JeyJeyC, 2025-03-26:
It's working perfectly !
Thank you !! :)

JeyJeyC, 2025-03-26:
One question...

About "Enable realtime keogram uploads (0 = disable, 1 = every image, 2 = every 2nd image, etc)"
If I choose every 2 images, is it taking ALL image then is only uploaded each 2 image, or is it taking 1 image each 2 (jumping 1)?

To understand if it's still a full keogram, even if it's not uploaded at each image.

aaronwmorris, 2025-03-26:
It is the fulll keogram that is being uploaded.  You may choose to upload it every 1, 2nd, 3rd, etc times it is generated.

JeyJeyC, 2025-04-04:
Could you please add the scale of time under it, like the final timelapse? :)

aaronwmorris, 2025-04-04:
Merged #2047 to support labels for realtime keograms.

This will reset your realtime keogram because I had to create a new data file to store the timestamp data from which the labels are derived.

JeyJeyC, 2025-04-05:
Thanks ! I try now.

closed 2025-04-24 by aaronwmorris