aaronwmorris/indi-allsky #1968: RPi Camera v3. How do I set autofocus?

Billium28, 2025-03-08:
New install, RPi 3b. Camera module 3. Seems to work ok but I can't figure out how to focus the camera. Is it an RPi command using ssh, or is there a way to set it inside Allsky? I saw 1 possible fix but it says I need the Advanced menu. 

Thanks.

aaronwmorris, 2025-03-08:
I just want to make sure you have a camera with an auto-focus mechanism and not the manual focus camera?

Billium28, 2025-03-08:
Yes, I double checked that when you asked haha. It says it's got phase detect autofocus.

aaronwmorris, 2025-03-08:
Okay good.  :-)

Normally, the auto-focus cameras are set for infinity at the 0 (default) position.  However, if you have the camera inside, you might have to adjust the focus for testing.

The autofocus mechanism will probably not work when pointed at the sky, but you can manually adjust focus by putting options like the following in the `Night libcamera extra options` and `Day libcamera extra options` fields on the Camera tab in the config.

```
--autofocus-mode manual --lens-position 7.1
```

The values for the lens-position are a little complex, but this is from the manual...

```
Moves the lens to a fixed focal distance, normally given in dioptres (units of 1 / distance in metres).

    0.0 will move the lens to the "infinity" position

    Any other number: move the lens to the 1 / number position, so the value 2 would focus at approximately 0.5m

    default - move the lens to a default position which corresponds to the hyperfocal position of the lens.
```

Billium28, 2025-03-09:
Thank you, I'll check this out!

closed 2025-04-04 by aaronwmorris