aaronwmorris/indi-allsky #1975: Download image from web page

AstroAndy46, 2025-03-10:
When I go to saved images and press the download button I just get "couldn't download - network issue" please can someone help and also how can I save each night to my pc hard drive. Thank you Andy

TheRealSunJao, 2025-03-10:
Andy, your browser or operating system security might be blocking the download if there is a problem with the web app security certificate, I.E., deemed unsafe.  This can occur when the certificate is self-signed or there is a mismatch between hostname and certificate common name.
An option would be to use Certbot to acquire Let's Encrypt certificate which will allow the browser to trust the source.

aaronwmorris, 2025-03-10:
See if this helps:

https://support.google.com/chrome/thread/10630812/how-to-download-a-file-in-spite-of-error-failed-bad-certificate?hl=en

AstroAndy46, 2025-03-10:
> See if this helps:
> 
> https://support.google.com/chrome/thread/10630812/how-to-download-a-file-in-spite-of-error-failed-bad-certificate?hl=en

Hi
I don't think it is a problem with a certificate as my other camera is fine when I download images from it.

AstroAndy46, 2025-03-10:
Is there a way just to send the data to my HDD in the morning?

aaronwmorris, 2025-03-10:
You could use the file transfer feature to copy files over the network.

AstroAndy46, 2025-03-10:
> You could use the file transfer feature to copy files over the network.

Hi
Thank you for getting back to me, I'm new to the indi-allsky and don't really know how to set it up. Can you guide me through the process.

TheRealSunJao, 2025-03-10:
> > You could use the file transfer feature to copy files over the network.
> 
> Hi Thank you for getting back to me, I'm new to the indi-allsky and don't really know how to set it up. Can you guide me through the process.

I chose the Indi-Allsky PycURL SFTP file transfer option and used Windows OpenSSH built-in server which supports SFTP connections.

Here is the Aaron's Wiki page that I used as a guideline:
https://github.com/aaronwmorris/indi-allsky/wiki/File-transfers#pycurl-sftp

Here is a guide I wrote for using SFTP with Windows...

[Indi-Allsky - using Windows OpenSSH Server (Built-in) for SFTP file transfers.docx](https://github.com/user-attachments/files/19170726/Indi-Allsky.-.using.Windows.OpenSSH.Server.Built-in.for.SFTP.file.transfers.docx)

closed 2025-04-04 by aaronwmorris