aaronwmorris/indi-allsky #1978: Not working. Only error I see is : "No Image for 15 minutes"

babsenfred, 2025-03-10:
"No Image for 15 minutes" Is all I'm getting. Did a clean install with RPI5 and Bookworm 64bit. Using the RPI HQ cam with Libcamera_IMX477. Cam worked fine with the Allsky software. Don't get it to work in this config. Did serveral new installations, all with te same result : "No Image for 15 minutes"

aaronwmorris, 2025-03-10:
Can you post your Support Info?

babsenfred, 2025-03-11:
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 8245504 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux IndiAllsky 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
Hostname
IndiAllsky
IndiAllsky

Time
Tue 11 Mar 09:05:00 CET 2025

System timezone
Europe/Brussels

Uptime
 09:05:00 up 10:31,  3 users,  load average: 0.12, 0.04, 0.01

Memory
               total        used        free      shared  buff/cache   available
Mem:         8245504      933360     6346960       10576     1074960     7312144
Swap:         524272           0      524272

Filesystems
Filesystem     1K-blocks    Used Available Use% Mounted on
udev             4080944       0   4080944   0% /dev
tmpfs             824560    5984    818576   1% /run
/dev/mmcblk0p2 245384396 9901200 222999764   5% /
tmpfs            4122752     704   4122048   1% /dev/shm
tmpfs               5120      48      5072   1% /run/lock
/dev/mmcblk0p1    522230   69392    452838  14% /boot/firmware
tmpfs             824544     176    824368   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
47400

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 2c:cf:67:32:84:e7 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:32:84:e8 brd ff:ff:ff:ff:ff:ff
    inet 192.168.50.38/24 brd 192.168.50.255 scope global dynamic noprefixroute wlan0
       valid_lft 48540sec preferred_lft 48540sec
    inet6 fde3:55ce:ccc8:e694:cfa8:9c8:b29e:89b6/64 scope global dynamic noprefixroute
       valid_lft 1754sec preferred_lft 1754sec
[... 1596 lines cut ...]
2025-03-11T09:02:37.628949+01:00 IndiAllsky [ERROR] Image-1895-16983/MainThread processing.get_astrometric_data() [2023]: HST next pass error: that satellite seems to stay always below your horizon
2025-03-11T09:02:37.673146+01:00 IndiAllsky [INFO] Capture-1-1095/Dummy-1 indi.newMessage() [353]: new Message 2025-03-11T08:02:37: [ERROR] Got no stars, is gsc installed with appropriate environment variables set ??
2025-03-11T09:02:52.805168+01:00 IndiAllsky [ERROR] Image-1895-16983/MainThread processing.get_astrometric_data() [2023]: HST next pass error: that satellite seems to stay always below your horizon
2025-03-11T09:02:52.938566+01:00 IndiAllsky [INFO] Capture-1-1095/Dummy-1 indi.newMessage() [353]: new Message 2025-03-11T08:02:52: [ERROR] Got no stars, is gsc installed with appropriate environment variables set ??
2025-03-11T09:03:08.075024+01:00 IndiAllsky [ERROR] Image-1895-16983/MainThread processing.get_astrometric_data() [2023]: HST next pass error: that satellite seems to stay always below your horizon
2025-03-11T09:03:08.158704+01:00 IndiAllsky [INFO] Capture-1-1095/Dummy-1 indi.newMessage() [353]: new Message 2025-03-11T08:03:08: [ERROR] Got no stars, is gsc installed with appropriate environment variables set ??
2025-03-11T09:03:23.296897+01:00 IndiAllsky [ERROR] Image-1895-16983/MainThread processing.get_astrometric_data() [2023]: HST next pass error: that satellite seems to stay always below your horizon
2025-03-11T09:03:23.389753+01:00 IndiAllsky [INFO] Capture-1-1095/Dummy-1 indi.newMessage() [353]: new Message 2025-03-11T08:03:23: [ERROR] Got no stars, is gsc installed with appropriate environment variables set ??
2025-03-11T09:03:38.520588+01:00 IndiAllsky [ERROR] Image-1895-16983/MainThread processing.get_astrometric_data() [2023]: HST next pass error: that satellite seems to stay always below your horizon
2025-03-11T09:03:38.605725+01:00 IndiAllsky [INFO] Capture-1-1095/Dummy-1 indi.newMessage() [353]: new Message 2025-03-11T08:03:38: [ERROR] Got no stars, is gsc installed with appropriate environment variables set ??
2025-03-11T09:03:53.728664+01:00 IndiAllsky [ERROR] Image-1895-16983/MainThread processing.get_astrometric_data() [2023]: HST next pass error: that satellite seems to stay always below your horizon
2025-03-11T09:03:53.829141+01:00 IndiAllsky [INFO] Capture-1-1095/Dummy-1 indi.newMessage() [353]: new Message 2025-03-11T08:03:53: [ERROR] Got no stars, is gsc installed with appropriate environment variables set ??
2025-03-11T09:04:08.965131+01:00 IndiAllsky [ERROR] Image-1895-16983/MainThread processing.get_astrometric_data() [2023]: HST next pass error: that satellite seems to stay always below your horizon
2025-03-11T09:04:09.043264+01:00 IndiAllsky [INFO] Capture-1-1095/Dummy-1 indi.newMessage() [353]: new Message 2025-03-11T08:04:09: [ERROR] Got no stars, is gsc installed with appropriate environment variables set ??
2025-03-11T09:04:24.172939+01:00 IndiAllsky [ERROR] Image-1895-16983/MainThread processing.get_astrometric_data() [2023]: HST next pass error: that satellite seems to stay always below your horizon
[... cut ...]
2025-03-11T09:04:54.644332+01:00 IndiAllsky [ERROR] Image-1895-16983/MainThread processing.get_astrometric_data() [2023]: HST next pass error: that satellite seems to stay always below your horizon
2025-03-11T09:04:54.737187+01:00 IndiAllsky [INFO] Capture-1-1095/Dummy-1 indi.newMessage() [353]: new Message 2025-03-11T08:04:54: [ERROR] Got no stars, is gsc installed with appropriate environment variables set ??
```

#################################
###     end support info      ###
#################################

baryah, 2025-03-11:
Yesterday I installed Indi-allsky on a new RPi4 with fresh installation of RPi OS. Did it thrice - both the installation of RPi OS as well as Indi-allsky - had the same issue all the three times.
The same message on the webpage - 'No image for 15 minutes'
Logs showed me  a lot of - 
                        "Image Worker Exception"s.
On looking into the error messages one of them caught my attention - 
                        FileNotFoundError: [Errno 2] No such file or directory: '/var/www/html/allsky/images/ccd_dcelfa48-belf-40a1-9d47-7f34012443b1/realtime_keogram.jpg'

Checked the file system - the folder "ccd_dcelfa48-belf-40a1-9d47-7f34012443b1" in the folder /var/www/html/allsky/images/ was not there.

Created it manually and everything worked.

The same happened in all the three installations

aaronwmorris, 2025-03-11:
I think you just need to switch the camera interface to `libcamera IMX477` in the config view and restart.

babsenfred, 2025-03-16:
Complete re-installation did it for me. Works fine now 👍🏻

closed 2025-04-24 by aaronwmorris