aaronwmorris/indi-allsky #1980: 'Image Worker Exception' in fresh installation of Indi-Allsky

baryah, 2025-03-11:
Yesterday I installed Indi-allsky on a new RPi4 with fresh installation of RPi OS. Did it thrice - both the installation of RPi OS as well as Indi-allsky - had the same issue all the three times.
The same message on the webpage - 'No image for 15 minutes'
Logs showed me a lot of -

`"Image Worker Exception"s.`

On looking into the error messages one of them caught my attention -

`FileNotFoundError: [Errno 2] No such file or directory: '/var/www/html/allsky/images/ccd_dcelfa48-belf-40a1-9d47-7f34012443b1/realtime_keogram.jpg'`

Checked the file system - the folder "ccd_dcelfa48-belf-40a1-9d47-7f34012443b1" in the folder /var/www/html/allsky/images/ was not there.

Created it manually and everything worked.

The same happened in all the three installations

aaronwmorris, 2025-03-11:
Can you give me the full error message from those errors?  I need to see the line number that produced the error.

baryah, 2025-03-11:
[indi-allsky.log](https://github.com/user-attachments/files/19184619/indi-allsky.log)

aaronwmorris, 2025-03-11:
I think I understand what is happening.

aaronwmorris, 2025-03-11:
Merged #1983 which should fix this condition.  The base folder for the camera probably did not exist (this is a condition that would only happen on first install).

baryah, 2025-03-19:
Yes fixed.
Checked on a new installation