aaronwmorris/indi-allsky #1982: Constant Exposure Hangups

AirBourn, 2025-03-11:
Been working for months, but now fails often.  Looks like INDI server is losing contact with the camera very often now.  Works for 10-40 mins or so, then I get "Exposure failed after 3 attempts" in the log and "Camera last ready XXXs ago. Camera might be hung. Aborting exposure."

[AllSkySupportInfo.txt](https://github.com/user-attachments/files/19195359/AllSkySupportInfo.txt)

AirBourn, 2025-03-11:
Just did it again.

[AllSkySupportInfo.txt](https://github.com/user-attachments/files/19196840/AllSkySupportInfo.txt)

aaronwmorris, 2025-03-11:
You are not the first to complain about stability of the exposures of the ASI676.

What kind of cable are you using?  Have you tried plugging the camera into the USB2 ports instead of the USB3 ports?

AirBourn, 2025-03-11:
I'm using the cable that came with the camera.  It's pretty beefy.  I've tried multiple USB slots, yes, but not lately I guess and didn't really pay attention to USB2 versus USB3 - I usually figured USB3 would be better for transfer rates.  Do you suggest a USB2 slot?

What has my attention is that this has been working fine for some time - this started happening in the last few days.  One thing I thought about was camera overheating?  Current reported camera temp is 112.3F - and it's 42F outside.  I first installed this camera in the middle of January, and now things are warming up?

AirBourn, 2025-03-11:
Noticed camera was always in one of the USB3 ports.  Tried moving to USB2 and restarted.  Will see how long it lasts now.

TheRealSunJao, 2025-03-11:
Following this as I'm running the ASI676 as well. The 112 F sensor temp with 42 F ambient is much higher than mine.
As a comparison my 676 sensor is reporting 63 F in 40 F ambient right now. My RPI5 CPU is ~35 F.

aaronwmorris, 2025-03-11:
Just to add more context, I would not consider 112F to be "hot".  My cameras commonly hit 160F in the summer.

AirBourn, 2025-03-12:
Still running four hours later, but I did get the "20.0s exposure received in 18.0s. Possible camera problem." dialog sometime in that time span.

[AllSkySupport02.txt](https://github.com/user-attachments/files/19201094/AllSkySupport02.txt)

aaronwmorris, 2025-03-12:
That error can also happen when performing reloads/restarts, but it should be a one time thing.

AirBourn, 2025-03-12:
Ran overnight just fine.  I did notice a handful of .py files in the repo updated, but looks like switching to a USB2 port helped!

aaronwmorris, 2025-03-12:
There are some other adjustments that might help in USB3 mode, mainly the USB bandwidth settings, but a shorter cable is probably recommended (if you want to keep the USB3 connection).  USB2 is almost always the easiest fix.

AirBourn, 2025-03-12:
Shorter cable really isn't an option for this installation.  As long as the transfer rates support image transfer, I see no need for USB3.

omeganon, 2025-03-12:
I was also experiencing this with my 678mc. It was a power issue for me. The PS I was using wasn't sufficient to drive the Pi, camera, and dew heater. Effects were same as yours. I ended up using a 45 watt usb-c power supply for the Pi/camera combo and running the dew heater off a separate power supply for now.

closed 2025-04-24 by aaronwmorris