aaronwmorris/indi-allsky #1989: Gamma Correction Variable bug

omeganon, 2025-03-13:
Hi!

https://github.com/aaronwmorris/indi-allsky/blob/main/indi_allsky/processing.py#L1626

You mean 'GAMMA_CORRECTION_DAY' here.

```
        if self.config.get('USE_NIGHT_COLOR', True):
            GAMMA_CORRECTION = float(self.config.get('GAMMA_CORRECTION', 1.0))
        else:
            if self.night_v.value:
                # night
                GAMMA_CORRECTION = float(self.config.get('GAMMA_CORRECTION', 1.0))
            else:
                # day
                GAMMA_CORRECTION = float(self.config.get('GAMMA_CORRECTIONSATURATION_FACTOR_DAY', 1.0))
```

omeganon, 2025-03-13:
There might be another issue somewhere else, unless it's just me. It won't save any values I put into Gamma Correction for day.

aaronwmorris, 2025-03-13:
Wow... thank you for catching this.  Merged #1990 to fix.

closed 2025-04-04 by aaronwmorris