aaronwmorris/indi-allsky #1997: darks not found

dlemmink, 2025-03-17:
With the latest version I get the following error.  I have my images directory set to another drive on the server.   Do the darks need to be in another directory?

Mar 17 14:24:03 CVG-L-INDI-ALLSKY [ERROR] Image-1-1264/MainThread processing._apply_calibration() [890]: Dark file missing: darks/dark_ccd1_16bit_1s_gain0_bin1_-273c_20250317_124420.fit
Mar 17 14:24:03 CVG-L-INDI-ALLSKY [INFO] Image-1-1264/MainThread sqm.calculate() [41]: Raw SQM: 20044.96, Weighted SQM: 1142540.64

They currently appear in a darks directory in the Image folder and have filenames corresponding to the indicated file.

aaronwmorris, 2025-03-17:
Can you show the output of the following command:

```
sqlite3 -header -table /var/lib/indi-allsky/indi-allsky.sqlite "SELECT id,camera_id,filename FROM darkframe;"
```

dlemmink, 2025-03-17:
Interesting....A second reboot and everything is back to normal.

+----+-----------+------------------------------------------------------------------+
| id | camera_id |                             filename                             |
+----+-----------+------------------------------------------------------------------+
| 1  | 1         | darks/dark_ccd1_16bit_15s_gain0_bin1_-273c_20250207_231739.fit   |
| 2  | 1         | darks/dark_ccd1_16bit_10s_gain0_bin1_-273c_20250207_231953.fit   |
| 3  | 1         | darks/dark_ccd1_16bit_5s_gain0_bin1_-273c_20250207_232110.fit    |
| 4  | 1         | darks/dark_ccd1_16bit_1s_gain0_bin1_-273c_20250207_232147.fit    |
| 5  | 1         | darks/dark_ccd1_16bit_15s_gain275_bin1_-273c_20250207_232448.fit |
| 6  | 1         | darks/dark_ccd1_16bit_10s_gain275_bin1_-273c_20250207_232655.fit |
| 7  | 1         | darks/dark_ccd1_16bit_5s_gain275_bin1_-273c_20250207_232813.fit  |
| 8  | 1         | darks/dark_ccd1_16bit_1s_gain275_bin1_-273c_20250207_232850.fit  |
| 9  | 1         | darks/dark_ccd1_16bit_15s_gain250_bin1_-273c_20250207_233150.fit |
| 10 | 1         | darks/dark_ccd1_16bit_10s_gain250_bin1_-273c_20250207_233358.fit |
| 11 | 1         | darks/dark_ccd1_16bit_5s_gain250_bin1_-273c_20250207_233516.fit  |
| 12 | 1         | darks/dark_ccd1_16bit_1s_gain250_bin1_-273c_20250207_233553.fit  |
| 13 | 1         | darks/dark_ccd1_16bit_10s_gain0_bin1_-273c_20250317_124216.fit   |
| 14 | 1         | darks/dark_ccd1_16bit_5s_gain0_bin1_-273c_20250317_124341.fit    |
| 15 | 1         | darks/dark_ccd1_16bit_1s_gain0_bin1_-273c_20250317_124420.fit    |
| 16 | 1         | darks/dark_ccd1_16bit_10s_gain275_bin1_-273c_20250317_124632.fit |
| 17 | 1         | darks/dark_ccd1_16bit_5s_gain275_bin1_-273c_20250317_124751.fit  |
| 18 | 1         | darks/dark_ccd1_16bit_1s_gain275_bin1_-273c_20250317_124832.fit  |
| 19 | 1         | darks/dark_ccd1_16bit_10s_gain250_bin1_-273c_20250317_125043.fit |
| 20 | 1         | darks/dark_ccd1_16bit_5s_gain250_bin1_-273c_20250317_125202.fit  |
| 21 | 1         | darks/dark_ccd1_16bit_1s_gain250_bin1_-273c_20250317_125241.fit  |
| 22 | 1         | darks/dark_ccd1_16bit_10s_gain0_bin1_-273c_20250317_153400.fit   |
| 23 | 1         | darks/dark_ccd1_16bit_5s_gain0_bin1_-273c_20250317_153521.fit    |
| 24 | 1         | darks/dark_ccd1_16bit_1s_gain0_bin1_-273c_20250317_153558.fit    |
| 25 | 1         | darks/dark_ccd1_16bit_10s_gain275_bin1_-273c_20250317_153810.fit |
| 26 | 1         | darks/dark_ccd1_16bit_5s_gain275_bin1_-273c_20250317_153927.fit  |
| 27 | 1         | darks/dark_ccd1_16bit_1s_gain275_bin1_-273c_20250317_154004.fit  |
| 28 | 1         | darks/dark_ccd1_16bit_10s_gain250_bin1_-273c_20250317_154214.fit |
| 29 | 1         | darks/dark_ccd1_16bit_5s_gain250_bin1_-273c_20250317_154332.fit  |
| 30 | 1         | darks/dark_ccd1_16bit_1s_gain250_bin1_-273c_20250317_154409.fit  |
+----+-----------+------------------------------------------------------------------+

closed 2025-03-17 by dlemmink

aaronwmorris, 2025-03-17:
Did you manually type (or edit) the filename in your first message?  The filename in the original message does NOT exist in the file list.  The file is there, but the original name is missing an `_` (underscore) between bin1 and -273.