aaronwmorris/indi-allsky #2007: Periodic purple/magenta images

WarrenSchultz, 2025-03-20:
Really strange issue. Things are going 99% great with this setup, but every hour or so during daytime, fewer at night, I get an image that looks like the one below. I've tried adjusting the camera bandwidth and the ASI120MC fix (since I switched from that camera, I tested it unintentionally <_< )  

[image]

I'll attach the support info below.

WarrenSchultz, 2025-03-20:
/home/thermal/github/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3882916 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux thermaltest 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
Hostname
thermaltest
thermaltest.gfgnet.home 

Time
Thu 20 Mar 00:39:10 EDT 2025

System timezone
US/Eastern

Uptime
 00:39:10 up 3 days,  7:24,  1 user,  load average: 1.34, 1.40, 1.08

Memory
               total        used        free      shared  buff/cache   available
Mem:         3882916     2693580      941760       31976      357028     1189336
Swap:         524284      455628       68656

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             1670404        0   1670404   0% /dev
tmpfs             388292    38776    349516  10% /run
/dev/mmcblk0p2 122624044 50847552  65527916  44% /
tmpfs            1941456      176   1941280   1% /dev/shm
tmpfs               5120       16      5104   1% /run/lock
/dev/mmcblk0p1    522230    68468    453762  14% /boot/firmware
tmpfs             388288       44    388244   1% /run/user/1000

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
43816

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether d8:3a:dd:6b:b1:0f brd ff:ff:ff:ff:ff:ff
    inet 192.168.125.61/22 brd 192.168.127.255 scope global dynamic noprefixroute eth0
       valid_lft 81448sec preferred_lft 81448sec
    inet6 fd57:af87:3c01:644f:dcd6:a662:58c8:4dc6/64 scope global dynamic noprefixroute 
       valid_lft 1789sec preferred_lft 1789sec
    inet6 fe80::28bf:cee8:a583:afcb/64 scope link noprefixroute 
[... 1501 lines cut ...]
    "SAT_LABEL_TEMPLATE": "{title:s} Alt: {alt:0.1f}° Az: {az:0.1f}° \\n Dir: {dir:s} Mag: {mag:0.1f} Rng: {range:0.1f}km",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,400 (Right)\n# anchor:ra (Right Justified)\n# color:128,128,128\nSatellites"
  }
}

```

indi-allky log errors
```
2025-03-20T00:01:09.031365-04:00 thermaltest [ERROR] Image-5-55915/MainThread processing.stack() [1041]: Registration exceeded the exposure period, cancel alignment
2025-03-20T00:28:56.246280-04:00 thermaltest [ERROR] Video-2-58374/MainThread smoke.download_kml() [205]: URL returned 404
2025-03-20T00:28:56.246460-04:00 thermaltest [ERROR] Video-2-58374/MainThread smoke.update() [60]: No smoke data: No KML data
2025-03-20T00:28:56.391065-04:00 thermaltest [ERROR] Video-2-58374/MainThread smoke.download_kml() [205]: URL returned 404
2025-03-20T00:28:56.391603-04:00 thermaltest [ERROR] Video-2-58374/MainThread smoke.update() [60]: No smoke data: No KML data
2025-03-20T00:33:21.377299-04:00 thermaltest [ERROR] Image-2-58373/MainThread processing.stack() [1041]: Registration exceeded the exposure period, cancel alignment
2025-03-20T00:34:26.182430-04:00 thermaltest [ERROR] Image-2-58373/MainThread processing.stack() [1041]: Registration exceeded the exposure period, cancel alignment
2025-03-20T00:35:21.156215-04:00 thermaltest [ERROR] Image-2-58373/MainThread stack.register() [180]: Rotation exceeded limit of +/- 0.00078927
2025-03-20T00:36:35.706681-04:00 thermaltest [ERROR] Image-2-58373/MainThread processing.stack() [1041]: Registration exceeded the exposure period, cancel alignment
2025-03-20T00:37:29.332004-04:00 thermaltest [ERROR] Image-2-58373/MainThread stack.register() [180]: Rotation exceeded limit of +/- 0.00074651
2025-03-20T00:38:27.076464-04:00 thermaltest [ERROR] Image-2-58373/MainThread stack.register() [180]: Rotation exceeded limit of +/- 0.00070859
```

#################################
###     end support info      ###
#################################

aaronwmorris, 2025-03-20:
Very odd.  I have only ever heard of this issue with the ASI224.  To hear that it is happening with the ASI120 and ASI676 is news to me.

For the ASI224, I have long suspected that the problem was occasionally returning an image where the sensor offset is different.  Most cameras have an "overscan" area of extra pixels that usually do not get used.  If the sensor data is started from a different offset, a different bayer pattern has to be used to get the correct colors.

X and Y "Offset" is a data point that is included with the FITS data returned by INDI.  INDI _might_ be doing the right thing and indicating the data change, but it is a difficult to test/validate since this problem randomly occurs.

omeganon, 2025-03-20:
If there's any debug code you want me to run I'm happy to do so. If there's a way you can tap here to maybe save the FITS file or raw data, that might be clueful.

https://github.com/aaronwmorris/indi-allsky/blob/main/indi_allsky/capture.py#L543-L558

```
2025-03-19T21:26:51.263910-05:00 allsky [ERROR] Capture-17-179345/MainThread capture.saferun() [543]: 30.0000s EXPOSURE RECEIVED IN 20.3831s.  POSSIBLE CAMERA PROBLEM.
2025-03-19T21:29:01.317623-05:00 allsky [ERROR] Capture-17-179345/MainThread capture.saferun() [543]: 30.0000s EXPOSURE RECEIVED IN 27.2484s.  POSSIBLE CAMERA PROBLEM.
```

aaronwmorris, 2025-03-20:
I am considering adding a log message that indicates the bayer pattern and offsets to be able to see the issue.

The colors will do this when the wrong bayer pattern is used to colorize the raw data.  

```
RGRGRG
GBGBGB
RGRGRG
```

This is an abbreviated RGGB bayer matrix.  If you offset Y by one, you have to use GBRG decoding.  If you offset X by one, GRBG is used.  If you offset both, BGGR would need to be used.  The FITS header indicates the overall bayer pattern for the sensor, but the offset indicates a different pattern.  (This assumes you are using INDI and not libcamera).

One person even had a camera where every image needed the wrong bayer pattern for the sensor.

WarrenSchultz, 2025-03-20:
Yeah, glad to test code here as well. 

One quick note, I also realized that I was a bit ambiguous on the ASI120. It didn't have purple image issue that I saw, but I had left the settings in to compensate for its issue, and forgotten to remove them when I switched cameras :)

aaronwmorris, 2025-03-20:
I have merged #2012 which will log the additional bayer information.  You should see log messages like the following:

```
2025-03-20 14:46:00,450 [INFO] Image-1-12919/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN
```

omeganon, 2025-03-20:
Logging confirmed. Now to have an event...

omeganon, 2025-03-20:
I actually did catch an event already. Unfortunately, the bayer values look to be as expected. 

[image]

```
2025-03-20T15:19:25.013240-05:00 allsky [INFO] Capture-1-196565/MainThread capture.saferun() [308]: Exposure state: OK
2025-03-20T15:19:25.013276-05:00 allsky [INFO] Capture-1-196565/MainThread capture.detectNight() [1489]: Sun alt: 40.7, Moon alt: -60.9, phase 66.0%
2025-03-20T15:19:25.113509-05:00 allsky [INFO] Capture-1-196565/MainThread indi.getCcdTemperature() [896]: Camera temperature: 29.4
2025-03-20T15:19:26.916371-05:00 allsky [INFO] Capture-1-196565/MainThread capture.shoot() [1684]: Taking 0.00042627 s exposure (gain 0)
2025-03-20T15:19:27.016485-05:00 allsky [INFO] Capture-1-196565/MainThread indi.set_number() [1363]: Setting Duration (s) = 0.00042626855429261923
2025-03-20T15:19:27.016556-05:00 allsky [INFO] Capture-1-196565/MainThread capture.saferun() [659]: Total time since last exposure 30.0099 s
2025-03-20T15:19:27.972896-05:00 allsky [INFO] Capture-1-196565/MainThread capture.saferun() [540]: Exposure received in 1.0566 s (1.0562)
2025-03-20T15:19:27.989608-05:00 allsky [INFO] Image-1-196567/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-20T15:19:28.037817-05:00 allsky [INFO] Image-1-196567/MainThread processing.detectBitDepth() [3624]: Image max value: 65534
2025-03-20T15:19:28.038253-05:00 allsky [WARNING] Image-1-196567/MainThread processing._calibrate() [749]: Dark frame calibration disabled
2025-03-20T15:19:28.106106-05:00 allsky [INFO] Image-1-196567/MainThread sqm.calculate() [41]: Raw SQM: 25588.53, Weighted SQM: 2354120.10
2025-03-20T15:19:28.106264-05:00 allsky [INFO] Image-1-196567/MainThread image.processImage() [417]: Image: 3552 x 3552
2025-03-20T15:19:28.133048-05:00 allsky [INFO] Image-1-196567/MainThread processing._calculate_8bit_adu() [985]: ADU average: 26596.7 (103)
2025-03-20T15:19:28.397708-05:00 allsky [INFO] Image-1-196567/MainThread mode2_mtf.stretch() [75]: Stretch in 0.2643 s
```

Images around the event --

[image]

```
2025-03-20T15:17:57.951535-05:00 allsky [INFO] Image-1-196567/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-20T15:18:27.962266-05:00 allsky [INFO] Image-1-196567/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-20T15:18:57.986128-05:00 allsky [INFO] Image-1-196567/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-20T15:19:27.989608-05:00 allsky [INFO] Image-1-196567/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-20T15:19:58.011973-05:00 allsky [INFO] Image-1-196567/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-20T15:20:28.037015-05:00 allsky [INFO] Image-1-196567/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-20T15:20:58.084470-05:00 allsky [INFO] Image-1-196567/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-20T15:21:28.090989-05:00 allsky [INFO] Image-1-196567/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-20T15:21:58.101406-05:00 allsky [INFO] Image-1-196567/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-20T15:22:28.111672-05:00 allsky [INFO] Image-1-196567/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN```

aaronwmorris, 2025-03-20:
@omeganon That is unfortunate.  Can you zoom in close on the purple image to see if you see a blocky pattern of dark pixels?

aaronwmorris, 2025-03-20:
JPEG might be compressing the pattern.  If you zoom way in, it does look like there might be a blocky pattern.

aaronwmorris, 2025-03-20:
This is from OPs first picture.  It is mainly visible in areas that are "uniform".  This is leftover pattern from the wrong debayering algorithm being used.

[image]

jimmauck, 2025-03-20:
Excellent!! While you were posting I recombined the channels from RGB to RBG and get the normal back...

[image]

aaronwmorris, 2025-03-20:
@omeganon It is subtle, but you can see the residual pattern.

[image]

aaronwmorris, 2025-03-20:
@jimmauck This is the residual bayer pattern for your image.

[image]

aaronwmorris, 2025-03-20:
@WarrenSchultz @jimmauck and @omeganon You are all using the ASI676.  This is probably enough info to open an issue with the INDI team and maybe engage with ZWO.

jimmauck, 2025-03-20:
I opened a case with ZWO.

jimmauck, 2025-03-20:
Another interesting statistic is that the magenta images are often > 1/3 larger file size than neighboring files.

aaronwmorris, 2025-03-20:
> Another interesting statistic is that the magenta images are often > 1/3 larger file size than neighboring files.

That is probably just the residual bayer pattern left over in the images.  It makes the file less compressible due to containing more "detail".

jimmauck, 2025-03-20:
For your reference, here are the two sample files, normal and magenta.

[image]

[image]

aaronwmorris, 2025-03-20:
@omeganon @jimmauck @WarrenSchultz Would one of you be willing to upgrade INDI to 2.1.2?  There was a ZWO SDK upgrade in that release.  It is probably something that needs to be tested.

https://github.com/indilib/indi-3rdparty/commit/a15affe6071e6abd42f6eb310f1cdc4e4d3246a0

jimmauck, 2025-03-20:
> [@omeganon](https://github.com/omeganon) [@jimmauck](https://github.com/jimmauck) [@WarrenSchultz](https://github.com/WarrenSchultz) Would one of you be willing to upgrade INDI to 2.1.2? There was a ZWO SDK upgrade in that release. It is probably something that needs to be tested.
> 
> [indilib/indi-3rdparty@a15affe](https://github.com/indilib/indi-3rdparty/commit/a15affe6071e6abd42f6eb310f1cdc4e4d3246a0)

Yes, I would like to do that INDI 2.1.2 upgrade, but not sure how.

Do I use the build_indi.sh script?
./misc/build_indi.sh

omeganon, 2025-03-21:
> [@omeganon](https://github.com/omeganon) [@jimmauck](https://github.com/jimmauck) [@WarrenSchultz](https://github.com/WarrenSchultz) Would one of you be willing to upgrade INDI to 2.1.2? There was a [...]

I can but it likely won’t be until tomorrow. (I know that’s fine).

omeganon, 2025-03-21:
> Yes, I would like to do that INDI 2.1.2 upgrade, but not sure how.
>
> Do I use the build_indi.sh script? ./misc/build_indi.sh

Looking at the code, it does look like a selectable version when running that script 

https://github.com/aaronwmorris/indi-allsky/blob/main/misc/build_indi.sh#L574

omeganon, 2025-03-21:
> Looking at the code, it does look like a selectable version when running that script
>
> https://github.com/aaronwmorris/indi-allsky/blob/main/misc/build_indi.sh#L574

Can confirm. It’s a simple as that. Recompiling now from my phone in a restaurant halfway across the country. 🤣

WarrenSchultz, 2025-03-21:
Once you confirm it's compiling and running fine, I can do the same.

omeganon, 2025-03-21:
> Once you confirm it's compiling and running fine, I can do the same.

All good. Recompiled, restarted indi-allsky, getting images.

I think it matters. I chose to compile the third-party drivers too.

jimmauck, 2025-03-21:
> > Once you confirm it's compiling and running fine, I can do the same.
> 
> All good. Recompiled, restarted indi-allsky, getting images.
> 
> I think it matters. I chose to compile the third-party drivers too.

Same here....compile to INDI 2.1.2 complete

aaronwmorris, 2025-03-21:
The indi 3rd party build is necessary to get the updated SDK.

I would also encourage a full reboot afterwards.   At a minimum, the indiserver needs to be restarted to load the new code.

omeganon, 2025-03-21:
Happened again and I do still see the pattern you point out (maybe even more now.)

```
2025-03-21T07:43:35.609514-05:00 allsky [INFO] Capture-1-2273/MainThread capture.saferun() [308]: Exposure state: OK
2025-03-21T07:43:35.609548-05:00 allsky [INFO] Capture-1-2273/MainThread capture.detectNight() [1489]: Sun alt: 10.1, Moon alt: 20.6, phase 59.6%
2025-03-21T07:43:35.709797-05:00 allsky [INFO] Capture-1-2273/MainThread indi.getCcdTemperature() [896]: Camera temperature: 27.3
2025-03-21T07:43:37.562644-05:00 allsky [INFO] Capture-1-2273/MainThread capture.shoot() [1684]: Taking 0.00226977 s exposure (gain 0)
2025-03-21T07:43:37.662746-05:00 allsky [INFO] Capture-1-2273/MainThread indi.set_number() [1363]: Setting Duration (s) = 0.002269770484417677
2025-03-21T07:43:37.662825-05:00 allsky [INFO] Capture-1-2273/MainThread capture.saferun() [659]: Total time since last exposure 30.0084 s
2025-03-21T07:43:38.448087-05:00 allsky [INFO] Sensor-1-2280/MainThread sensor.saferun() [194]: Dew Point: 21.2, Frost Point: 22.0, Heat Index: 185.9
2025-03-21T07:43:38.614983-05:00 allsky [INFO] Capture-1-2273/MainThread capture.saferun() [540]: Exposure received in 1.0525 s (1.0502)
2025-03-21T07:43:38.631845-05:00 allsky [INFO] Image-2-2332/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-21T07:43:38.670091-05:00 allsky [INFO] Image-2-2332/MainThread processing.detectBitDepth() [3624]: Image max value: 65534
2025-03-21T07:43:38.670419-05:00 allsky [WARNING] Image-2-2332/MainThread processing._calibrate() [749]: Dark frame calibration disabled
2025-03-21T07:43:38.776450-05:00 allsky [INFO] Image-2-2332/MainThread sqm.calculate() [41]: Raw SQM: 27407.86, Weighted SQM: 2521380.27
2025-03-21T07:43:38.776635-05:00 allsky [INFO] Image-2-2332/MainThread image.processImage() [417]: Image: 3552 x 3552
2025-03-21T07:43:38.813145-05:00 allsky [INFO] Image-2-2332/MainThread processing._calculate_8bit_adu() [985]: ADU average: 27648.1 (108)
2025-03-21T07:43:39.070541-05:00 allsky [INFO] Image-2-2332/MainThread mode2_mtf.stretch() [75]: Stretch in 0.2571 s
2025-03-21T07:43:39.823095-05:00 allsky [INFO] Image-2-2332/MainThread processing._saturation_adjust() [1542]: Applying saturation settings
2025-03-21T07:43:41.296517-05:00 allsky [INFO] Image-2-2332/MainThread processing.get_image_label() [2340]: Adding aircraft text
2025-03-21T07:43:41.502106-05:00 allsky [INFO] Image-2-2332/MainThread image.processImage() [676]: Image processed in 2.8972 s```

![Image](https://github.com/user-attachments/assets/c10f932f-ba55-4ef6-83c4-a576ece09041)

<img width="787" alt="Image" src="https://github.com/user-attachments/assets/a94508a4-07d9-48e7-b1b8-88c4c770bea0" />

omeganon, 2025-03-21:
```
/home/admin/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 8245504 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux allsky 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
Hostname
allsky
allsky.home.local 

Time
Fri 21 Mar 08:04:48 CDT 2025

System timezone
America/Chicago

Uptime
 08:04:48 up  8:28,  3 users,  load average: 0.26, 0.28, 0.34

Memory
               total        used        free      shared  buff/cache   available
Mem:         8245504     2311328      564528       96672     5565536     5934176
Swap:         524272       39936      484336

Filesystems
Filesystem      1K-blocks      Used  Available Use% Mounted on
udev              4080944         0    4080944   0% /dev
tmpfs              824560     86288     738272  11% /run
/dev/mmcblk0p2 1968204352 364191160 1504030424  20% /
tmpfs             4122752      1008    4121744   1% /dev/shm
tmpfs                5120        48       5072   1% /run/lock
/dev/mmcblk0p1     522230     69388     452842  14% /boot/firmware
tmpfs              824544       192     824352   1% /run/user/1000

systemctl runlevel
graphical.target

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
45750

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:4e:fe:9c brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.10/24 brd 192.168.1.255 scope global dynamic noprefixroute eth0
[... 1602 lines cut ...]
2025-03-21T05:04:23.320500-05:00 allsky [ERROR] Sensor-1-2280/MainThread tempApiAmbientWeather.update() [231]: Dew Point calculation error - ValueError: math domain error
2025-03-21T05:09:23.727689-05:00 allsky [ERROR] Sensor-1-2280/MainThread tempApiAmbientWeather.update() [231]: Dew Point calculation error - ValueError: math domain error
2025-03-21T05:14:24.130710-05:00 allsky [ERROR] Sensor-1-2280/MainThread tempApiAmbientWeather.update() [231]: Dew Point calculation error - ValueError: math domain error
2025-03-21T05:19:24.486311-05:00 allsky [ERROR] Sensor-1-2280/MainThread tempApiAmbientWeather.update() [231]: Dew Point calculation error - ValueError: math domain error
2025-03-21T05:24:24.851480-05:00 allsky [ERROR] Sensor-1-2280/MainThread tempApiAmbientWeather.update() [231]: Dew Point calculation error - ValueError: math domain error
2025-03-21T05:29:25.230939-05:00 allsky [ERROR] Sensor-1-2280/MainThread tempApiAmbientWeather.update() [231]: Dew Point calculation error - ValueError: math domain error
2025-03-21T05:34:25.589951-05:00 allsky [ERROR] Sensor-1-2280/MainThread tempApiAmbientWeather.update() [231]: Dew Point calculation error - ValueError: math domain error
2025-03-21T05:39:25.942344-05:00 allsky [ERROR] Sensor-1-2280/MainThread tempApiAmbientWeather.update() [231]: Dew Point calculation error - ValueError: math domain error
2025-03-21T05:44:27.727600-05:00 allsky [ERROR] Sensor-1-2280/MainThread tempApiAmbientWeather.update() [231]: Dew Point calculation error - ValueError: math domain error
2025-03-21T05:47:40.651297-05:00 allsky [ERROR] Video-1-2279/MainThread smoke.download_kml() [205]: URL returned 404
2025-03-21T05:47:40.651591-05:00 allsky [ERROR] Video-1-2279/MainThread smoke.update() [60]: No smoke data: No KML data
2025-03-21T05:49:28.113200-05:00 allsky [ERROR] Sensor-1-2280/MainThread tempApiAmbientWeather.update() [231]: Dew Point calculation error - ValueError: math domain error
2025-03-21T05:54:28.430544-05:00 allsky [ERROR] Sensor-1-2280/MainThread tempApiAmbientWeather.update() [231]: Dew Point calculation error - ValueError: math domain error
2025-03-21T05:59:28.841073-05:00 allsky [ERROR] Sensor-1-2280/MainThread tempApiAmbientWeather.update() [231]: Dew Point calculation error - ValueError: math domain error
2025-03-21T06:04:29.209728-05:00 allsky [ERROR] Sensor-1-2280/MainThread tempApiAmbientWeather.update() [231]: Dew Point calculation error - ValueError: math domain error
2025-03-21T06:47:34.671127-05:00 allsky [INFO] Video-1-2279/MainThread timelapse.generate() [142]: FFmpeg command: ffmpeg -y -loglevel level+error -r 25.00 -f image2 -i /tmp/tmpwc9ed0e6_timelapse/%05d.jpg -vcodec libx264 -b:v 35000k -pix_fmt yuv420p -movflags +faststart -level 3.1 /var/www/html/allsky/images/ccd_c6d29829-b452-4e90-a032-da65e710e3eb/timelapse/20250320/allsky-startrail_timelapse_ccd1_20250320_night_1742556265.mp4
2025-03-21T06:53:59.474235-05:00 allsky [INFO] Video-1-2279/MainThread timelapse.generate() [142]: FFmpeg command: ffmpeg -y -loglevel level+error -r 25.00 -f image2 -i /tmp/tmp65725q39_timelapse/%05d.jpg -vcodec libx264 -b:v 35000k -pix_fmt yuv420p -movflags +faststart -level 3.1 /var/www/html/allsky/images/ccd_c6d29829-b452-4e90-a032-da65e710e3eb/timelapse/20250320/allsky-timelapse_ccd1_20250320_night_1742558015.mp4
[... cut ...]  libx264 -b:v 35000k -pix_fmt yuv420p -movflags +faststart -level 3.1 /var/www/html/allsky/images/ccd_c6d29829-b452-4e90-a032-da65e710e3eb/timelapse/20250320/allsky-panorama_timelapse_ccd1_20250320_night_1742558581.mp4
2025-03-21T08:04:40.395405-05:00 allsky [ERROR] Sensor-1-2280/MainThread tempApiAmbientWeather.update() [231]: Dew Point calculation error - ValueError: math domain error
```

#################################
###     end support info      ###
#################################
```

aaronwmorris, 2025-03-21:
If someone wants to try to run their camera in RGB24 mode instead of RAW16, there is a way to do that.  This will invalidate your dark frames since the camera is debayering the images internally and is returning 8-bit RGB data.  You can just take new dark frames and keep your old ones.  indi-allsky knows not to use the 16-bit data since the data types do not match.

Just to note: This is the mode TJ's allsky uses to get data from ZWO cameras.

```json
{
   "SWITCHES" : {
      "CCD_VIDEO_FORMAT" : {
         "on" : [
            "ASI_IMG_RGB24"
         ],
         "off" : [
            "ASI_IMG_RAW8",
            "ASI_IMG_RAW16"
         ]
      }
   }
}
```

omeganon, 2025-03-21:
>   aaronwmorris  left a comment   (aaronwmorris/indi-allsky#2007) (https://github.com/aaronwmorris/indi-allsky/issues/2007#issuecomment-2743612853)

Done
  

  
  
  

  
  
  
  
  
—  
Marc

WarrenSchultz, 2025-03-21:
I'll give this a shot as well. I haven't rebuilt with the latest indi-allsky, so I can be a control group test.

WarrenSchultz, 2025-03-21:
@omeganon was that text from the log viewer or are you querying elsewhere?

omeganon, 2025-03-21:
> [@omeganon](https://github.com/omeganon) was that text from the log viewer or are you querying elsewhere?

Support info dump after reload. Still on my phone so copy/paste is meh.

WarrenSchultz, 2025-03-21:
Gotcha, thanks. One thing that's odd, it also lists this variable: ZWO CCD ASI676MC.CCD_CAPTURE_FORMAT.ASI_IMG_RGB24=Off
Does INDI use the Video format instead of Capture for long exposures?

jimmauck, 2025-03-21:
> Last evening I compiled to INDI 2.1.2, rebooted this morning and captured two purple images since.
> 
> 15 minutes ago, I updated camera config to RGB24 and just captured the first purple image.
> 
> Edit to note I submitted the RGB24 config change in the "NDI Camera Config (Day)" field, and then only saved and reloaded the config change.

Here is the image capture log to confirm 8-bit capture:
2025-03-21T13:40:08.734648-05:00 allsky-robin-obs [INFO] Image-2-16822/MainThread processing.add() [578]: Image data - bits: 8, cfa: None, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-21T13:40:08.738946-05:00 allsky-robin-obs [INFO] Image-2-16822/MainThread processing.detectBitDepth() [3624]: Image max value: 255
2025-03-21T13:40:08.739129-05:00 allsky-robin-obs [INFO] Image-2-16822/MainThread processing._apply_calibration() [835]: Searching for dark frame: gain 0, exposure >= 0.0, temp >= 32.2c
2025-03-21T13:40:08.740902-05:00 allsky-robin-obs [WARNING] Image-2-16822/MainThread processing._apply_calibration() [874]: Dark not found: ccd3 8bit 0.0350893s gain 0 bin 1 32.20c

WarrenSchultz, 2025-03-21:
For fundamental camera changes, I'd restart the Pi completely. Completely restarting the service vs. reload config should also work, but for debugging, I'm always more thorough.

jimmauck, 2025-03-21:
> For fundamental camera changes, I'd restart the Pi completely. Completely restarting the service vs. reload config should also work, but for debugging, I'm always more thorough.

Got it, rebooting now...thanks!

jimmauck, 2025-03-21:
> > For fundamental camera changes, I'd restart the Pi completely. Completely restarting the service vs. reload config should also work, but for debugging, I'm always more thorough.
> 
> Got it, rebooting now...thanks!

Just captured a fresh purple after reboot.

omeganon, 2025-03-21:
Same. Full reboot just performed.

omeganon, 2025-03-21:
> Gotcha, thanks. One thing that's odd, it also lists this variable: ZWO CCD ASI676MC.CCD_CAPTURE_FORMAT.ASI_IMG_RGB24=Off Does INDI use the Video format instead of Capture for long exposures?

Interesting observation that I don’t know the answer to. It’s does seem to just be setting the video flag. 

```
ZWO CCD ASI676MC.CCD_CAPTURE_FORMAT.ASI_IMG_RAW8=Off
ZWO CCD ASI676MC.CCD_CAPTURE_FORMAT.ASI_IMG_RGB24=Off
ZWO CCD ASI676MC.CCD_CAPTURE_FORMAT.ASI_IMG_Y8=Off
ZWO CCD ASI676MC.CCD_CAPTURE_FORMAT.ASI_IMG_RAW16=On
ZWO CCD ASI676MC.CCD_TRANSFER_FORMAT.FORMAT_FITS=On
ZWO CCD ASI676MC.CCD_TRANSFER_FORMAT.FORMAT_NATIVE=Off
ZWO CCD ASI676MC.CCD_TRANSFER_FORMAT.=Off
…
ZWO CCD ASI676MC.CCD_VIDEO_FORMAT.ASI_IMG_RAW8=Off
ZWO CCD ASI676MC.CCD_VIDEO_FORMAT.ASI_IMG_RGB24=On
ZWO CCD ASI676MC.CCD_VIDEO_FORMAT.ASI_IMG_Y8=Off
ZWO CCD ASI676MC.CCD_VIDEO_FORMAT.ASI_IMG_RAW16=Off
```

Using

```
{
    "SWITCHES": {
        "CCD_VIDEO_FORMAT": {
            "on": [
                "ASI_IMG_RGB24"
            ],
            "off": [
                "ASI_IMG_RAW8",
                "ASI_IMG_RAW16"
            ]
        }
    },
    "PROPERTIES": {
        "CCD_CONTROLS": {
            "BandWidth": 90
        }
    },
    "TEXT": {}
}
```

WarrenSchultz, 2025-03-21:
Interestingly, on the current codebase, I haven't had a purple frame (yet) since applying the change. But it's random enough that it's no guarantee. 

Understood and agreed about the flag. I know that it's a different variable, but as you said, I'm curious about the difference between functions/flows.

Heading into nighttime now though, so I'll have to wait until tomorrow for more frames to test.

jimmauck, 2025-03-22:
> Interestingly, on the current codebase, I haven't had a purple frame (yet) since applying the change. But it's random enough that it's no guarantee.

I set the Exposure Period (Day) to minimum recommended value of 5.0, which increased the "random" occurance of purple.

aaronwmorris, 2025-03-22:
> I set the Exposure Period (Day) to minimum recommended value of 5.0, which increased the "random" occurance of purple.

@jimmauck the 5s recommendation was just due to some cameras taking 4+ seconds to download the image from the camera.  Check the Lag view and just make sure the period is longer than the download time (for the daytime).

jimmauck, 2025-03-22:
> [@jimmauck](https://github.com/jimmauck) the 5s recommendation was just due to some cameras taking 4+ seconds to download the image from the camera. Check the Lag view and just make sure the period is longer than the download time (for the daytime).
> 
> Edit: You can set it lower. The queue backlog detection should kick in to add additional delays. You may get a warning in the web interface.

Sounds good, Aaron. I've set it to 1.0 and hopefully drive some meaningful statistics.

jimmauck, 2025-03-22:
Not sure if helpful, but the purple frequency appears to increase when exposure < 1.0 s.

WarrenSchultz, 2025-03-22:
Yeah, that's something I've been wondering about. Since we haven't seen these at all at night AFIAK (?), fast daytime exposures may be exposing an edge case that doesn't get tested a lot with most astronomy cameras.

WarrenSchultz, 2025-03-22:
I did finally get a purple frame at 9am this morning, but not since. Just for testing purposes, I've got the latest code (including logging) and am also grabbing images at 3s intervals, but back at RAW16.

WarrenSchultz, 2025-03-22:
One last thing for now, I did get another purple frame after switching back to 16-bit mode with short interval image taking. 

Here's the full log for the frame (14:42) from capture to exported jpg.

```
2025-03-22T16:04:42.577146-04:00 thermaltest [INFO] Capture-3-1506/MainThread capture.saferun() [540]: Exposure received in 1.8742 s (1.8739)
2025-03-22T16:04:42.656103-04:00 thermaltest [INFO] Image-3-1508/MainThread miscDb.addImage() [242]: Adding image ccd_26eb2df4-3653-4f85-9b7d-44285891d06b/exposures/20250322/day/22_16/ccd2_20250322_160433.jpg to DB
2025-03-22T16:04:42.664770-04:00 thermaltest [INFO] Image-3-1508/MainThread miscDb.addThumbnail() [1170]: Adding thumbnail to DB: /var/www/html/allsky/images/ccd_26eb2df4-3653-4f85-9b7d-44285891d06b/exposures/20250322/day/22_16/thumbnails/1c2aee69-5334-4c44-bb09-3fe0369136ee.jpg
2025-03-22T16:04:42.911505-04:00 thermaltest [INFO] MainProcess-1036/Upload-6 generic.put() [80]: Uploading /var/www/html/allsky/images/ccd_26eb2df4-3653-4f85-9b7d-44285891d06b/exposures/20250322/day/22_16/ccd2_20250322_160433.jpg
2025-03-22T16:04:42.986386-04:00 thermaltest [INFO] MainProcess-1036/Upload-6 paho_mqtt.put() [134]: File transferred in 0.0728 s (14749.98 kB/s)
2025-03-22T16:04:42.987055-04:00 thermaltest [INFO] MainProcess-1036/Upload-6 uploader.processUpload() [544]: Upload transaction completed in 0.0760 s
2025-03-22T16:04:42.991702-04:00 thermaltest [INFO] Image-3-1508/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-22T16:04:43.067852-04:00 thermaltest [INFO] Image-3-1508/MainThread processing.detectBitDepth() [3630]: Image max value: 65534
2025-03-22T16:04:43.068289-04:00 thermaltest [INFO] Image-3-1508/MainThread processing._apply_calibration() [835]: Searching for dark frame: gain 0, exposure >= 0.0, temp >= 47.8c
2025-03-22T16:04:43.073201-04:00 thermaltest [WARNING] Image-3-1508/MainThread processing._apply_calibration() [874]: Dark not found: ccd2 16bit 0.0003400s gain 0 bin 1 47.80c
2025-03-22T16:04:43.207235-04:00 thermaltest [INFO] Image-3-1508/MainThread sqm.calculate() [41]: Raw SQM: 22217.25, Weighted SQM: 1599628.40
2025-03-22T16:04:43.207405-04:00 thermaltest [INFO] Image-3-1508/MainThread image.processImage() [417]: Image: 3552 x 3552
2025-03-22T16:04:43.263567-04:00 thermaltest [INFO] Image-3-1508/MainThread processing._calculate_8bit_adu() [985]: ADU average: 22502.7 (87)
2025-03-22T16:04:43.743599-04:00 thermaltest [INFO] Capture-3-1506/MainThread capture.shoot() [1684]: Taking 0.00033996 s exposure (gain 0)
2025-03-22T16:04:43.744091-04:00 thermaltest [INFO] Image-3-1508/MainThread processing._saturation_adjust() [1542]: Applying saturation settings
2025-03-22T16:04:43.843773-04:00 thermaltest [INFO] Capture-3-1506/MainThread indi.set_number() [1363]: Setting Duration (s) = 0.0003399647830519825
2025-03-22T16:04:43.844085-04:00 thermaltest [WARNING] Capture-3-1506/MainThread capture.saferun() [627]: Image queue depth: 2
2025-03-22T16:04:43.844219-04:00 thermaltest [INFO] Capture-3-1506/MainThread capture.saferun() [659]: Total time since last exposure 3.0405 s
2025-03-22T16:04:45.135477-04:00 thermaltest [INFO] Image-3-1508/MainThread processing.get_satellite_tracking_text() [2812]: Satellite tracking in 0.0104 s
2025-03-22T16:04:45.135663-04:00 thermaltest [INFO] Image-3-1508/MainThread processing.get_image_label() [2349]: Adding satellite text
2025-03-22T16:04:45.626800-04:00 thermaltest [INFO] Image-3-1508/MainThread image.processImage() [676]: Image processed in 2.7191 s
2025-03-22T16:04:45.630134-04:00 thermaltest [INFO] Capture-3-1506/MainThread capture.saferun() [540]: Exposure received in 1.8863 s (1.8859)
2025-03-22T16:04:45.630938-04:00 thermaltest [WARNING] Capture-3-1506/MainThread capture._periodic_tasks() [1164]: Periodic tasks triggered
2025-03-22T16:04:45.999962-04:00 thermaltest [INFO] Image-3-1508/MainThread miscDb.addImage() [242]: Adding image ccd_26eb2df4-3653-4f85-9b7d-44285891d06b/exposures/20250322/day/22_16/ccd2_20250322_160436.jpg to DB
2025-03-22T16:04:46.008833-04:00 thermaltest [INFO] Image-3-1508/MainThread miscDb.addThumbnail() [1170]: Adding thumbnail to DB: /var/www/html/allsky/images/ccd_26eb2df4-3653-4f85-9b7d-44285891d06b/exposures/20250322/day/22_16/thumbnails/ea1e077a-ac77-42d2-9676-dc45ef446075.jpg
2025-03-22T16:04:46.607808-04:00 thermaltest [INFO] MainProcess-1036/Upload-5 generic.put() [80]: Uploading /var/www/html/allsky/images/ccd_26eb2df4-3653-4f85-9b7d-44285891d06b/exposures/20250322/day/22_16/ccd2_20250322_160436.jpg
2025-03-22T16:04:46.679246-04:00 thermaltest [INFO] MainProcess-1036/Upload-5 paho_mqtt.put() [134]: File transferred in 0.0707 s (15183.35 kB/s)
2025-03-22T16:04:46.679420-04:00 thermaltest [INFO] MainProcess-1036/Upload-5 uploader.processUpload() [544]: Upload transaction completed in 0.0728 s
2025-03-22T16:04:46.687489-04:00 thermaltest [INFO] Image-3-1508/MainThread processing.add() [578]: Image data - bits: 16, cfa: RGGB, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-22T16:04:46.759007-04:00 thermaltest [INFO] Image-3-1508/MainThread processing.detectBitDepth() [3630]: Image max value: 65534
2025-03-22T16:04:46.759662-04:00 thermaltest [INFO] Image-3-1508/MainThread processing._apply_calibration() [835]: Searching for dark frame: gain 0, exposure >= 0.0, temp >= 47.8c
[... cut ...]
2025-03-22T16:04:53.978061-04:00 thermaltest [INFO] Image-3-1508/MainThread miscDb.addThumbnail() [1170]: Adding thumbnail to DB: /var/www/html/allsky/images/ccd_26eb2df4-3653-4f85-9b7d-44285891d06b/exposures/20250322/day/22_16/thumbnails/127211bb-262b-4d49-9da3-f7c42bec891b.jpg
2025-03-22T16:04:54.223272-04:00 thermaltest [INFO] MainProcess-1036/Upload-6 generic.put() [80]: Uploading /var/www/html/allsky/images/ccd_26eb2df4-3653-4f85-9b7d-44285891d06b/exposures/20250322/day/22_16/ccd2_20250322_160442.j
```

omeganon, 2025-03-23:
I had one incident today :/

```
2025-03-22T17:56:50.641374-05:00 allsky [INFO] Capture-1-2310/MainThread capture.saferun() [308]: Exposure state: OK
2025-03-22T17:56:50.641410-05:00 allsky [INFO] Capture-1-2310/MainThread capture.detectNight() [1489]: Sun alt: 12.2, Moon alt: -70.7, phase 45.4%
2025-03-22T17:56:50.741658-05:00 allsky [INFO] Capture-1-2310/MainThread indi.getCcdTemperature() [896]: Camera temperature: 36.8
2025-03-22T17:56:51.993639-05:00 allsky [INFO] Capture-1-2310/MainThread capture.shoot() [1684]: Taking 0.00129548 s exposure (gain 0)
2025-03-22T17:56:52.093712-05:00 allsky [INFO] Capture-1-2310/MainThread indi.set_number() [1363]: Setting Duration (s) = 0.0012954834382981062
2025-03-22T17:56:52.093958-05:00 allsky [INFO] Capture-1-2310/MainThread capture.saferun() [659]: Total time since last exposure 30.0122 s
2025-03-22T17:56:53.258524-05:00 allsky [INFO] Capture-1-2310/MainThread capture.saferun() [540]: Exposure received in 1.2649 s (1.2636)
2025-03-22T17:56:53.301253-05:00 allsky [INFO] Image-3-20600/MainThread processing.add() [578]: Image data - bits: 8, cfa: None, offsets: X 0, Y 0, row order: TOP-DOWN
2025-03-22T17:56:53.305378-05:00 allsky [INFO] Image-3-20600/MainThread processing.detectBitDepth() [3624]: Image max value: 255
2025-03-22T17:56:53.305534-05:00 allsky [WARNING] Image-3-20600/MainThread processing._calibrate() [749]: Dark frame calibration disabled
2025-03-22T17:56:53.484940-05:00 allsky [INFO] Image-3-20600/MainThread sqm.calculate() [41]: Raw SQM: 86.31, Weighted SQM: 7940.30
2025-03-22T17:56:53.485005-05:00 allsky [INFO] Image-3-20600/MainThread image.processImage() [417]: Image: 3552 x 3552
2025-03-22T17:56:53.507215-05:00 allsky [INFO] Image-3-20600/MainThread processing._calculate_8bit_adu() [985]: ADU average: 86.9 (86)
```

[image]

No clear bayer pattern —

[image]

omeganon, 2025-03-23:
No events today. It definitely feels improved.

omeganon, 2025-03-30:
Just a follow up. For me, the number of incidents has definitely reduced since switching to RGB24, but normally eliminated it. I still get 1 frame a day that’s purple. It would be great to be able to eliminate them, but something I can live with as is.

aaronwmorris, 2025-08-24:
Closing this issue for now.  Please re-open if there are any updates.

I have created a wiki page that documents this issue for the ASI676:
https://github.com/aaronwmorris/indi-allsky/wiki/ASI676MC-Known-Issues

closed 2025-08-24 by aaronwmorris

WarrenSchultz, 2026-04-03:
I have seen them still, but haven't updated in a while. I'll try to do that soon, been busy!

omeganon, 2026-04-03:
Running 2.2.0 now. 🕐

omeganon, 2026-04-07:
Still happening. meh.

[image]
[image]

omeganon, 2026-04-10:
I think I've switched to RAW16 after reboot. `file` command shows 16-bit image data v.s. 8-bit now. Pretty sure the switch to RGB24 was an attempt to resolve this issue anyway.

$ file ccd1_20260410_132648.fit
ccd1_20260410_132648.fit: FITS image data, 16-bit, two's complement binary integer

```
{
    "SWITCHES": {
        "CCD_VIDEO_FORMAT": {
            "on": [
                "ASI_IMG_RAW16"
            ],
            "off": [
                "ASI_IMG_RAW8",
                "ASI_IMG_RGB24"
            ]
        }
    },
    "PROPERTIES": {
        "CCD_CONTROLS": {
            "BandWidth": 90
        }
    },
    "TEXT": {}
}
```